Koenkk/zigbee2mqtt #2417: Solving ZigBee Light-Link and 3.0 losing connection

Boostflow, 2019-11-26:
# Bug Report
## What happened
Out of 3 connectors 2 disconnect as expected and 1 remains fully functioning
## What did you expect to happen
All 3 connectors to lose connection after ~10 hours
## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.7.1
CC1352P-2 firmware version: 3.x.0 (20191106)

Following this issue https://github.com/Koenkk/zigbee2mqtt/issues/1602 I find myself in the same situation. It seems 2 out of my 3 connectors suffer from this issue and disconnect after a couple of hours.

Reading on about the issue here: https://translate.google.com/translate?sl=de&tl=en&u=https%3A%2F%2Fforum.fhem.de%2Findex.php%2Ftopic%2C100772.msg976520.html%23msg976520 it seems a possible solution would be to "creates the DIRECT pairing with the de gateway - so in case of doubt all zigBee 3.0 components (eg Tradfri bulbs) disable and pair". How do I identify all Zigbee 3.0 devices in my network? I hope there is a workaround for this issue as I have 3 switches installed and working great before.

Koenkk, 2019-11-26:
Besides checking the manufacturer site I don't think there is an easy way to check if a device is 3.0 or 1.2. The `database.db` will contain various version numbers for each device (e.g. `stackVersion` and `zclVersion`) but these don't seem to indicate the Zigbee version.

felixstorm, 2019-12-03:
@Boostflow If Jung really delivers a firmware update (I was asked by their technical hotline to check with them again in a week or so), there might be a chance that it will also work on the Gira devices as they both seem to report identically on the ZigBee side (I have both of them here, neither one mentions Gira or Jung, both only identify as "Insta"). So both might actually be running identical firmware.

felixstorm, 2020-02-03:
Just a quick update: I am still in contact with Jung - the update has been delayed a bit, but is still scheduled to be released within the next few weeks. I will try it out on my Jung device and also check if it works on a Gira device as well and let you now.

felixstorm, 2020-03-13:
### Some bad news for the Gira owners, some good news for Jung owners
Jung as requested Insta (the OEM who originally developed the switches both for Gira and Jung) to produce a firmware update for their devices to solve the issues. I have an updated Jung ZLLLS5004M working for a week now without any issues so far (Z2M with CC2652R coordinator, around 20-30 devices, most of them 3.0 capable, bindings from the switch only to the coordinator).  
According to Jung the primary downside of the update is a reduced battery lifetime of around 6 months, which is totally fine for me but lead Jung to discontinue sales of their switches.  
There is no public download link available for the update, but it can be requested directly from Jung by contacting their technical support - my communication with them was very, very friendly and helpful.

The bad news: Apparently Gira has discontinued their switches a lot earlier and has therefore not even requested a firmware update from Insta at all. Also, just brute force changing of any hardware ids in the download file and trying to force it onto a Gira switch might not be a good idea either as some of the maintenance button combinations include the use of the two lowermost buttons which are missing on the Gira devices...

### Installation of the update
I used [deCONZ](https://www.dresden-elektronik.de/funk/software/deconz.html) to install the update as I had a ConBee stick lying around and could just fire it up as an HA addon. But the update file is just a normal ZigBee OTA file, so if Z2M should get an OTA module for updating arbitrary devices using local files, this should work as well.  
For the switch to ask for/accept the update, it needs to be in "programming mode" which can be entered be pushing the top right and the bottom left buttons simultaneously for a few seconds:
[image]
Thereafter I was able to update the switch using a VNC connection to the deCONZ Linux UI, but it might also be possible using their web interface.

kaeferfreund, 2020-03-13:
Thank you GIRA

tetzlav, 2020-03-23:
@felixstorm Do you have an update file for us? I write to Jung support, but no answer...

felixstorm, 2020-03-23:
@tetzlav Here you go: https://github.com/Koenkk/zigbee-OTA/raw/master/images/Insta/ZLL_WS_4f_J_Release_10.03.32.02.zigbee

As of now, there are still some [fixes/enhancements](https://github.com/Koenkk/zigbee-herdsman-converters/pull/1082) that need to be applied to Z2M OTA, but I am pretty confident that this could happen soon. Then Z2M will be able to update the Jung switch using the normal [OTA update procedures](https://www.zigbee2mqtt.io/information/ota_updates.html).

pvermeij, 2020-05-13:
@felixstorm Thanks for the good explanation of the way to perform the OTA update.
Had disconnecting issue's with my Jung ZLLA5004M switches and just updated them without any problem using Zigbee2mqttAssistant and a C2531 stick.
(putting the switch in 'programming mode' was a great tip)
Now let's wait and see if the switches will stay connected to the Zigbee network.

@Koenkk 
OTA worked flawlessly (took around an hour per switch) with nice status updates in the zigbee2mqtt log. Thanks!

tetzlav, 2020-05-13:
Do i understand that right, that the update also works with Gira devices? The Gira 2430-100 is mentioned in the OTA Readme...
I own a Gira hand remote which detected as 

`{"id":7,"type":"EndDevice","ieeeAddr":"0x000f17124101023c","nwkAddr":3441,"manufId":4474,"manufName":" Insta ","powerSource":"Battery","modelId":" Remote","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":2048,"inClusterList":[0,4096],"outClusterList":[4096,3,6,8,4,5,768,0,65280,25],"clusters":{"genBasic":{"attributes":{"modelId":" Remote","manufacturerName":" Insta ","powerSource":3,"zclVersion":1,"appVersion":10,"stackVersion":32,"hwVersion":1,"dateCode":" 20160530","swBuildId":" 10023202"}}},"binds":[]}},"appVersion":10,"stackVersion":32,"hwVersion":1,"dateCode":" 20160530","swBuildId":" 10023202","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1589403133633}`

But update check results in 

`debug 2020-05-13 22:51:45: Check if update available for '0x000f17124101023c' ( Remote)
error 2020-05-13 22:51:45: Failed to call 'OTAUpdate' 'onZigbeeEvent' (AssertionError [ERR_ASSERTION]: No image available for imageType '53249'
    at getImageMeta (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/ota/zigbeeOTA.js:13:5)
    at process._tickCallback (internal/process/next_tick.js:68:7))`

Any chance to update this device?

felixstorm, 2020-05-14:
@tetzlav No, the update does unfortunately not work with Gira devices, please see my [comment](https://github.com/Koenkk/zigbee2mqtt/issues/2417#issuecomment-598650782) from above:
> The bad news: Apparently Gira has discontinued their switches a lot earlier and has therefore not even requested a firmware update from Insta at all. Also, just brute force changing of any hardware [...]

Although they're probably very similar from their hardware, they request two different OTA upgrade image types and as long as Gira does not publish an image for their image type id, there is nothing zigbee2mqtt could do.

Could you point me to the location where the Gira 2430-100 is mentioned for OTA upgrades? I will be happy to correct this then to avoid further confusion.

tetzlav, 2020-05-14:
Thank you again for your clarification. I assume you allready requested Gira for help in updates?

The thing that make me hope is that there is also a hand-held remote from Jung with exactly the same specification. This update should work with the Jung version, why not with the Gira-Version?
https://zigbeealliance.org/zigbee_products/zigbee-radio-hand-held-transmitter-gira/
https://zigbeealliance.org/zigbee_products/zigbee-radio-hand-held-transmitter-jung/
https://zigbeealliance.org/zigbee_products/zigbee-light-link-radio-hand-held-transmitter/

Its exactly the same Insta OEM hardware and the lowermost buttons are there too. Maybe i should give it a try to change the hardware ids in image file and try to update the Gira remote with Jung file. At the actual buggy firmware state of the Gira remote makes it useless  for me anyway...

> Could you point me to the location where the Gira 2430-100 is mentioned for OTA upgrades? I will be happy to correct this then to avoid further confusion.

https://www.zigbee2mqtt.io/information/ota_updates.html

felixstorm, 2020-05-14:
@tetzlav No, I did not ask directly at Gira as I personally use the Jung devices only (I only have one Gira switch here that I acquired for testing purposes about 2 or 3 years ago). The information I have is from somebody working at Jung who was kind enough to ask one of his contacts at Gira for me about this, but the Gira contact reportedly said that Gira had not at all requested the firmware update from Insta (the OEM for both).

And whether the Gira hand-held remotes will work with the update or not is hard to say without actually trying it out as it only depends on the image type that gets requested by the device. Also, the firmware update file for the Jung wall switch has the letter "J" in it's name which probably stands for "Jung": `ZLL_WS_4f_J_Release_10.03.32.02.zigbee`. But as the hand helds only seem to differ in the outer marking, they _might_ also request the same firmware image type.  
Also, the firmware update file for the Jung hand-held remote has not yet been added to the zigbee2mqtt OTA repository (I do not have it either), so somebody would need to request it from Jung.

And trying to change the firmware image type in the OTA file might unfortunately not be as easy as it looks since the firmware itself seems to be encrypted in the file.

If you really want to dive deeper into this, I see two more options:
* Use a [Gira 2003 100](https://katalog.gira.de/en/datenblatt.html?id=653059) instead and connect it to some other ZigBee sensor.
* Completely dump Gira's or Jung's firmware and build a new one yourself based on Microchip (ex Atmel) ZigBee sample code and directly flash it onto the device (it's based on an ATSAMR21G18A chip).

And thanks for the reference to the docs - I just created a [PR](https://github.com/Koenkk/zigbee2mqtt.io/pull/287) to correct this.

pvermeij, 2020-05-15:
After the upgrade of two Jung ZLLA5004M switches I've had no disconnects of these devices anymore, so it looks like the upgrade is doing it's job. 
I do notice something strange with some other Jung ZLLA5004M switches I just connected via Z2M.
These devices don't show the update as available allthough they haven't been updated yet.
(these have been laying around in their boxes for more than a year)

Is there a way to force the switches to update?

felixstorm, 2020-05-19:
> Is there a way to force the switches to update?

Are there any log entries for the non-updating switches?

What helped for me in cases when the switches seemed to be stuck somehow was to remove the battery for a few seconds and then push the [magic button combination](https://github.com/Koenkk/zigbee2mqtt/issues/2417#issuecomment-598650782) again. Thereafter I could usually force the update by publishing to `zigbee2mqtt/bridge/ota_update/update` (as explained [here](zigbee2mqtt/bridge/ota_update/update)).  
In case removing the battery does not help either, you could even completely reset and re-pair the switch:
[image]

pvermeij, 2020-05-19:
There were no log entries except that there was no update available. 

But resetting the switch, re-pairing it, pushing the magic button combination and then forcing the update with  `zigbee2mqtt/bridge/ota_update/update` with payload `friendly name` did the trick.

Thank you for the instructions!

I do think there's something wrong with the firmware's version number as the switch is still reporting the old version number and the OTA success message doesn't contain the new version number:
`zigbee2mqtt:info  2020-05-20 00:00:36: Finished update of 'Keuken - Schakelaar 1'
zigbee2mqtt:info  2020-05-20 00:00:36: MQTT publish: topic 'zigbee2mqtt/Keuken - Schakelaar 1', payload '{"linkquality":42,"update_available":false}'
zigbee2mqtt:info  2020-05-20 00:00:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"ota_update","message":"Finished update of 'Keuken - Schakelaar 1'","meta":{"status":"update_succeeded","device":"Keuken - Schakelaar 1","from":{"softwareBuildID":" 10023202","dateCode":" 20160530"},"to":null}}'`

After the update the log shows the following information about the switch:
`{"ieeeAddr":"0x000f17124102004a","type":"EndDevice","networkAddress":5796,"model":"InstaRemote","vendor":"Insta","description":"ZigBee Light Link wall transmitter","friendly_name":"Keuken - Schakelaar 1","manufacturerID":4474,"manufacturerName":" Insta ","powerSource":"Battery","modelID":" Remote","hardwareVersion":1,"softwareBuildID":" 10023202","dateCode":" 20160530","lastSeen":1589925495057},`

closed 2020-07-26 by stale[bot]

tetzlav, 2020-09-08:
There also exits an update file for the hand remote from Jung: ZLL_HS_4f_GJ_Release_10.03.32.02.zigbee
https://nextcloud.jungit.net/index.php/s/9ice7FCiLnNXRSA

I own such identical remote from Gira, an i hope the G in update file name stands for "Gira". What code need to be changed to test the update file with my "Gira Zigbee Light Link Funk Handsender" 2435 10" (https://zigbee.blakadder.com/Gira_0527-00.html)

felixstorm, 2020-09-14:
It might very well work with the Gira model as well - I would also suspect that the G stands for "Gira". I have created a [PR](https://github.com/Koenkk/zigbee-OTA/pull/10) with the update file in the respective repository.

@tetzlav Is the device (apart from OTA updates) already being supported by zigebee2mqtt? And if yes - does it identify itself as "ZigBee Light Link wall transmitter"? Then the OTA should probably just work as soon as the above PR hast been merged.  
Otherwise an update to https://github.com/Koenkk/zigbee-herdsman-converters will also need to be made to a) support the device in general and/or b) make it updateble using `ota.zigbeeOTA`.

tetzlav, 2020-09-14:
Thank you!

Yes, the device ist supported by z2m:
`
database.db:{"id":7,"type":"EndDevice","ieeeAddr":"0x000f17124101023c","nwkAddr":25636,"manufId":4474,"manufName":" Insta ","powerSource":"Battery","modelId":" Remote","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":2048,"inClusterList":[0,4096],"outClusterList":[4096,3,6,8,4,5,768,0,65280,25],"clusters":{"genBasic":{"attributes":{"modelId":" Remote","manufacturerName":" Insta ","powerSource":3,"zclVersion":1,"appVersion":10,"stackVersion":32,"hwVersion":1,"dateCode":" 20160530","swBuildId":" 10023202"}}},"binds":[]}},"appVersion":10,"stackVersion":32,"hwVersion":1,"dateCode":" 20160530","swBuildId":" 10023202","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":null}`

I'll wait until your PR is merged...

felixstorm, 2020-09-14:
Koen has already merged the PR, so the OTA update for your handheld remote should already work now. Please remember that the remote needs to be set to passive link mode for OTA updates to work (press top right and bottom left buttons simultaneously according to the Jung documentation).

My Jung wall remotes have been working well for about half a year now, with only two (rather small) annoyances:
* Battery lifetime has dropped down tremendously - I need to replace batteries every few months.
* Sometimes a button only reacts after it has been pressed for a second time (but I have not yet checked if this is really caused by the remote itself or by z2m or HA - although my Ikea remotes do not suffer from this).

Would be interesting to know if you experience the same with your Gira remote.

tetzlav, 2020-09-17:
Jippie! Thank you all very much!
```
Sep 16 23:44:04 phoscon npm[8036]: Zigbee2MQTT:debug 2020-09-16 23:44:04: Received Zigbee message from '0x000f17124101023c', type 'commandQueryNextImageRequest', clu
Sep 16 23:44:09 phoscon npm[8036]: Zigbee2MQTT:debug 2020-09-16 23:44:09: Received MQTT message on 'zigbee2mqtt/bridge/ota_update/check' with data '0x000f17124101023
Sep 16 23:44:09 phoscon npm[8036]: Zigbee2MQTT:info  2020-09-16 23:44:09: Checking if update available for '0x000f17124101023c'
Sep 16 23:44:09 phoscon npm[8036]: Zigbee2MQTT:info  2020-09-16 23:44:09: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Checking if update avail
Sep 16 23:44:09 phoscon npm[8036]: Zigbee2MQTT:debug 2020-09-16 23:44:09: Check if update available for '0x000f17124101023c' ( Remote)
Sep 16 23:44:09 phoscon npm[8036]: Zigbee2MQTT:debug 2020-09-16 23:44:09: Using endpoint '1'
Sep 16 23:44:15 phoscon npm[8036]: Zigbee2MQTT:debug 2020-09-16 23:44:15: Received Zigbee message from '0x000f17124101023c', type 'commandQueryNextImageRequest', clu
Sep 16 23:44:15 phoscon npm[8036]: Zigbee2MQTT:debug 2020-09-16 23:44:15: Got OTA request '{"fieldControl":0,"manufacturerCode":4474,"imageType":53249,"fileVersion":
Sep 16 23:44:15 phoscon npm[8036]: Zigbee2MQTT:debug 2020-09-16 23:44:15: Is new image available for '0x000f17124101023c', current '{"fieldControl":0,"manufacturerCo
Sep 16 23:44:15 phoscon npm[8036]: Zigbee2MQTT:debug 2020-09-16 23:44:15: Update available for '0x000f17124101023c': YES
Sep 16 23:44:15 phoscon npm[8036]: Zigbee2MQTT:info  2020-09-16 23:44:15: Update available for '0x000f17124101023c'
Sep 16 23:44:15 phoscon npm[8036]: Zigbee2MQTT:info  2020-09-16 23:44:15: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update available for '0x
Sep 16 23:44:15 phoscon npm[8036]: Zigbee2MQTT:info  2020-09-16 23:44:15: MQTT publish: topic 'zigbee2mqtt/0x000f17124101023c', payload '{"linkquality":70,"update_av
Sep 16 23:44:20 phoscon npm[8036]: Zigbee2MQTT:debug 2020-09-16 23:44:20: Received Zigbee message from '0x000f17124101023c', type 'commandQueryNextImageRequest', clu
Sep 16 23:44:25 phoscon npm[8036]: Zigbee2MQTT:debug 2020-09-16 23:44:25: Received MQTT message on 'zigbee2mqtt/bridge/ota_update/update' with data '0x000f1712410102
Sep 16 23:44:25 phoscon npm[8036]: Zigbee2MQTT:info  2020-09-16 23:44:25: Updating '0x000f17124101023c' to latest firmware
Sep 16 23:44:25 phoscon npm[8036]: Zigbee2MQTT:info  2020-09-16 23:44:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"meta":{"device":"0x000f17124101023
Sep 16 23:44:25 phoscon npm[8036]: Zigbee2MQTT:debug 2020-09-16 23:44:25: Received Zigbee message from '0x000f17124101023c', type 'readResponse', cluster 'genBasic',
Sep 16 23:44:25 phoscon npm[8036]: Zigbee2MQTT:debug 2020-09-16 23:44:25: No converter available for 'InstaRemote' with cluster 'genBasic' and type 'readResponse' an
Sep 16 23:44:25 phoscon npm[8036]: Zigbee2MQTT:debug 2020-09-16 23:44:25: Updating to latest '0x000f17124101023c' ( Remote)
Sep 16 23:44:25 phoscon npm[8036]: Zigbee2MQTT:debug 2020-09-16 23:44:25: Using endpoint '1'
Sep 16 23:44:33 phoscon npm[8036]: Zigbee2MQTT:debug 2020-09-16 23:44:33: Received Zigbee message from '0x000f17124101023c', type 'commandQueryNextImageRequest', clu
Sep 16 23:44:33 phoscon npm[8036]: Zigbee2MQTT:debug 2020-09-16 23:44:33: Got OTA request '{"fieldControl":0,"manufacturerCode":4474,"imageType":53249,"fileVersion":
Sep 16 23:44:33 phoscon npm[8036]: Zigbee2MQTT:debug 2020-09-16 23:44:33: getNewImage for '0x000f17124101023c', meta {"fileVersion":36832016,"fileSize":133287,"url":
Sep 16 23:44:34 phoscon npm[8036]: Zigbee2MQTT:debug 2020-09-16 23:44:34: getNewImage for '0x000f17124101023c', image header {"otaUpgradeFileIdentifier":{"type":"Buf
Sep 16 23:44:34 phoscon npm[8036]: Zigbee2MQTT:debug 2020-09-16 23:44:34: Got new image for '0x000f17124101023c'
Sep 16 23:44:34 phoscon npm[8036]: Zigbee2MQTT:debug 2020-09-16 23:44:34: Starting upgrade
Sep 16 23:44:44 phoscon npm[8036]: Zigbee2MQTT:debug 2020-09-16 23:44:44: Received Zigbee message from '0x000f17124101023c', type 'commandQueryNextImageRequest', clu
Sep 16 23:44:46 phoscon npm[8036]: Zigbee2MQTT:debug 2020-09-16 23:44:46: OTA update at 0.05%, remaining 25458.867741935486 seconds
Sep 16 23:44:46 phoscon npm[8036]: Zigbee2MQTT:info  2020-09-16 23:44:46: Update of '0x000f17124101023c' at 0.05%, +- 424 minutes remaining
Sep 16 23:44:46 phoscon npm[8036]: Zigbee2MQTT:info  2020-09-16 23:44:46: MQTT publish: topic 'zigbee2mqtt/0x000f17124101023c', payload '{"linkquality":70,"update_av
Sep 16 23:44:46 phoscon npm[8036]: Zigbee2MQTT:info  2020-09-16 23:44:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of '0x000f1712410
Sep 16 23:45:16 phoscon npm[8036]: Zigbee2MQTT:debug 2020-09-16 23:45:16: OTA update at 0.82%, remaining 5072.634497257769 seconds
Sep 16 23:45:16 phoscon npm[8036]: Zigbee2MQTT:info  2020-09-16 23:45:16: Update of '0x000f17124101023c' at 0.82%, +- 85 minutes remaining
Sep 16 23:45:16 phoscon npm[8036]: Zigbee2MQTT:info  2020-09-16 23:45:16: MQTT publish: topic 'zigbee2mqtt/0x000f17124101023c', payload '{"linkquality":70,"update_av
Sep 16 23:45:16 phoscon npm[8036]: Zigbee2MQTT:info  2020-09-16 23:45:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of '0x000f1712410
[... cut ...]
Sep 17 00:55:18 phoscon npm[8036]: Zigbee2MQTT:info  2020-09-17 00:55:18: MQTT publish: topic 'zigbee2mqtt/0x000f17124101023c', payload '{"linkquality":70,"update_available":false}'
Sep 17 00:55:18 phoscon npm[8036]: Zigbee2MQTT:info  2020-09-17 00:55:18: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"0x000f17124101023c","meta":{"device":"0x000f17124101023c","from":{"date_code":" 20160530","software_build_id":" 10023202"},"status":"update_succeeded","to":null},"type":"ota_update"}'
```

tetzlav, 2020-09-17:
> 
> Would be interesting to know if you experience the same with your Gira remote.

I updated allready a Jung ZLL5004M (https://zigbee.blakadder.com/JUNG_ZLL5004M.html). The connection lost issues are gone and we use it since Juni 2-3 times a day to switch lights in our bedroom: No significant battery life drop rocognized yet, whould say, i use the first battery until now. Very rarely a button show no reaction at first push. For me everything is fine now... ;)

tetzlav, 2020-09-17:
Maybe you should update https://www.zigbee2mqtt.io/information/ota_updates.html with info for success with "Model 2435-10" by Gira...