Koenkk/zigbee2mqtt #2418: Trying to pair HEIMAN smart remote controller: HS1RC-M

VenkateshGSRao, 2019-11-26:
# Bug Report
## What happened
Trying to pair HEIMAN smart remote controller: HS1RC-M
Gets paired successfully without any arror. But nothing happens when i press any of the buttons on the device.

Here is the log:

zigbee2mqtt:info 2019-11-26 17:15:04: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-11-26.17-15-03'
zigbee2mqtt:debug 2019-11-26 17:15:04: Removing old log directory '/opt/zigbee2mqtt/data/log/2019-11-23.18-23-32'
zigbee2mqtt:debug 2019-11-26 17:15:04: Loaded state from file /opt/zigbee2mqtt/data/state.json
zigbee2mqtt:info 2019-11-26 17:15:04: Starting zigbee2mqtt version 1.7.1+dev (commit #546a72f)
zigbee2mqtt:info 2019-11-26 17:15:04: Starting zigbee-herdsman...
zigbee2mqtt:debug 2019-11-26 17:15:04: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
zigbee2mqtt:info 2019-11-26 17:15:06: zigbee-herdsman started
zigbee2mqtt:info 2019-11-26 17:15:06: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
zigbee2mqtt:debug 2019-11-26 17:15:06: Zigbee network parameters: {"panID":6754,"extendedPanID":"0xdddddddddddddddd","channel":11}
zigbee2mqtt:info 2019-11-26 17:15:06: Currently 5 devices are joined:
zigbee2mqtt:info 2019-11-26 17:15:06: 0x00158d00033e8fdb (0x00158d00033e8fdb): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
zigbee2mqtt:info 2019-11-26 17:15:06: 0x00158d000303f8aa (0x00158d000303f8aa): RTCGQ01LM - Xiaomi MiJia human body movement sensor (EndDevice)
zigbee2mqtt:info 2019-11-26 17:15:06: 0x00158d00033ad918 (0x00158d00033ad918): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
zigbee2mqtt:info 2019-11-26 17:15:06: 0x00124b001d436c34 (0x00124b001d436c34): SA-003-Zigbee - eWeLink Zigbee smart plug (Router)
zigbee2mqtt:info 2019-11-26 17:15:06: 0x000d6f00104f28bd (0x000d6f00104f28bd): HS1RC-E - HEIMAN Smart remote controller (EndDevice)
zigbee2mqtt:warn 2019-11-26 17:15:06: permit_join set to true in configuration.yaml.
zigbee2mqtt:warn 2019-11-26 17:15:06: Allowing new devices to join.
zigbee2mqtt:warn 2019-11-26 17:15:06: Set permit_join to false once you joined all devices.
zigbee2mqtt:info 2019-11-26 17:15:06: Zigbee: allowing new devices to join.
zigbee2mqtt:info 2019-11-26 17:15:06: Connecting to MQTT server at mqtt://localhost
zigbee2mqtt:info 2019-11-26 17:15:07: Connected to MQTT server
zigbee2mqtt:info 2019-11-26 17:15:07: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt:info 2019-11-26 17:15:07: MQTT publish: topic 'zigbee2mqtt/0x00158d00033e8fdb', payload '{"battery":100,"voltage":3092,"linkquality":120}'
zigbee2mqtt:info 2019-11-26 17:15:07: MQTT publish: topic 'zigbee2mqtt/0x00158d000303f8aa', payload '{"occupancy":false,"linkquality":131,"battery":100,"voltage":3015}'
zigbee2mqtt:info 2019-11-26 17:15:07: MQTT publish: topic 'zigbee2mqtt/0x00158d00033ad918', payload '{"battery":100,"voltage":3065,"linkquality":120,"contact":true}'
zigbee2mqtt:info 2019-11-26 17:15:07: MQTT publish: topic 'zigbee2mqtt/0x00124b001d436c34', payload '{"state":"OFF","linkquality":141}'
zigbee2mqtt:info 2019-11-26 17:15:07: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.7.1+dev","commit":"546a72f","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}'
zigbee2mqtt:info 2019-11-26 17:16:33: Device '0x005043c9a32f701d' joined
zigbee2mqtt:info 2019-11-26 17:16:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x005043c9a32f701d"}}'
zigbee2mqtt:info 2019-11-26 17:16:33: Starting interview of '0x005043c9a32f701d'
zigbee2mqtt:info 2019-11-26 17:16:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x005043c9a32f701d"}}'
zigbee2mqtt:debug 2019-11-26 17:16:33: Device '0x005043c9a32f701d' announced itself
zigbee2mqtt:debug 2019-11-26 17:16:38: Received Zigbee message from '0x005043c9a32f701d', type 'readResponse', cluster 'genBasic', data '{"modelId":"RC_V14","manufacturerName":"Heiman","powerSource":3}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2019-11-26 17:16:38: No converter available for 'HS1RC-M' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"RC_V14","manufacturerName":"Heiman","powerSource":3}'
zigbee2mqtt:debug 2019-11-26 17:16:39: Received Zigbee message from '0x005043c9a32f701d', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1,"appVersion":3,"stackVersion":2}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2019-11-26 17:16:39: No converter available for 'HS1RC-M' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":1,"appVersion":3,"stackVersion":2}'
zigbee2mqtt:debug 2019-11-26 17:16:42: Received Zigbee message from '0x005043c9a32f701d', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1,"dateCode":"20150427"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2019-11-26 17:16:42: No converter available for 'HS1RC-M' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1,"dateCode":"20150427"}'
zigbee2mqtt:info 2019-11-26 17:16:42: Successfully interviewed '0x005043c9a32f701d', device has successfully been paired
zigbee2mqtt:info 2019-11-26 17:16:42: Device '0x005043c9a32f701d' is supported, identified as: HEIMAN Smart remote controller (HS1RC-M)
zigbee2mqtt:info 2019-11-26 17:16:42: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x005043c9a32f701d","model":"HS1RC-M","vendor":"HEIMAN","description":"Smart remote controller","supported":true}}'
## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

bambouzel, 2019-11-26:
Can you try with the firmware 20190425 ?  Then you are using the same as me.

info 2019-11-18T17:40:58: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":**20190425**}}'

Can you close #2409 which is basically the same issue?

VenkateshGSRao, 2019-11-26:
Where do i get this 20190425 version firmware?

#2409 : I am not sure the issue is same or not as the models are different and how Z2M responds is also different.

bambouzel, 2019-11-26:
@VenkateshGSRao [link](https://github.com/Koenkk/zigbee2mqtt/issues/1957#issuecomment-556217527)

VenkateshGSRao, 2019-11-26:
Tried with firmware 20190425. No luck. Still no response for button presses...

Anything else i can try?...

Thanks.

bambouzel, 2019-11-26:
When I look at your log file on top I still see that a device still is recognized as HS1RC-E

zigbee2mqtt:info 2019-11-26 17:15:06: 0x000d6f00104f28bd (0x000d6f00104f28bd): HS1RC-E - HEIMAN Smart remote controller (EndDevice)

Then, a bit later 0x005043c9a32f701d joins which is your remote I presume, with model "RC_V14".

So maybe first try to get the remote leave the network so that no Heiman remote is available and then pair again.  This looks a lot like issue #1957.

VenkateshGSRao, 2019-11-29:
@bambouzel,
Thanks for your support. I will try and update soon...

closed 2020-02-04 by stale[bot]