Koenkk/zigbee2mqtt #2419: Get offline availability message for removed device + trace

sjorge, 2019-11-26:
# Bug Report
## What happened

While testing a GLEDOPTO GL-B-001Z and being very unsatisfied I removed it from the network, all was fine until the availability timeout was reached. Then I got greated with the error below.

```
zigbee2mqtt:info  2019-11-26 15:52:59: MQTT publish: topic 'zigbee2mqtt/0x00124b001edd7777/availability', payload 'offline'
(node:25679) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'name' of null
    at DeviceAvailability.handleInterval (/opt/zigbee2mqtt/lib/extension/deviceAvailability.js:81:53)
(node:25679) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:25679) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
zigbee2mqtt:info  2019-11-26 15:57:12: MQTT publish: topic 'zigbee2mqtt/masterbedroom/repeater', payload '{"linkquality":60}'
```

This error repeats itself every timeout interval until zigbee2mqtt is restarted, I think something is not being cleaned up when removing a device.

```
(node:25679) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'name' of null
    at DeviceAvailability.handleInterval (/opt/zigbee2mqtt/lib/extension/deviceAvailability.js:81:53)
(node:25679) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 3)
zigbee2mqtt:info  2019-11-26 15:57:12: MQTT publish: topic 'zigbee2mqtt/masterbedroom/repeater', payload '{"linkquality":60}'
```

The repeater being right after is a give away as that linkquality message is triggered by the ping.

## What did you expect to happen
Once a device is removed it should no longer be pinged for by the availability extension.

## How to reproduce it (minimal and precise)
- pair a new mains powered device
- use it for a few minutes
- remove device (in my case the leave was clean as it was no longer available in the device list)
- notice errors in log

## Debug Info
zigbee2mqtt version: 1.7.0+dev
CC253X firmware version:

sjorge, 2019-11-26:
Repeatable with an Innr bulb too, it also cleanly left the network

Koenkk, 2019-11-27:
Should be fixed in latest dev.

sjorge, 2019-11-27:
👍

closed 2019-11-27 by sjorge