Koenkk/zigbee2mqtt #2422: Gledopto GL-S-007ZS GU10 Bulb Support

milobt, 2019-11-27:
# Bug Report
Add support for Gledopto GL-S-007ZS (Plus) Bulb
## What happened
I have purchase a number of plus versions of the Gledopto GU10 bulbs.
The current GL-S-007Z is supported, but not the S variant.

I have tried to add support myself, by creating a devices.js file in the Hassio Zigbee2Mqtt directory and then adding  "zigbee_shepherd_devices": true to the configuration file, but get an error upon restarting.

I thought the process of adding new devices would be alot simpler - maybe I've complicated it somewhere.

I'm running the release version of Zigbee2Mqtt, not the edge version, does this make a difference?

Here's what I believe should be the new device:

{
    zigbeeModel: ['GL-S-007ZS'],
    model: 'GL-S-007ZS',
    vendor: 'Gledopto',
    description: 'Smart RGB+CCT GU10',
    extend: gledopto.light,
    supports: 'on/off, brightness, color, white',
}
## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.7.1
CC253X firmware version:

Can anyone help with adding support please?

Koenkk, 2019-11-27:
Thanks! Will be supported in the dev branch in a few hours.

closed 2019-11-27 by Koenkk

milobt, 2019-11-27:
Thanks @Koenkk 

I did mange to get it paired using the following for the device:

```
{
        zigbeeModel: ['GL-S-007ZS'],
        model: 'GL-S-007ZS',
        vendor: 'Gledopto',
        description: 'Smart RGB + CCT GU10',
        extend: gledopto.light,
        meta: {options: {disableDefaultResponse: true}},
        supports: 'on/off, brightness, color temperature, color',
},
```
Is this the same/similar to your entry?

Also, can you help with an issue: when a failed connection happens, for instance an unsupported device or a failed pair, when it does pair successfully, it no longer auto integrates into Hass and so does not show as a device in the MQTT integration page etc.

Is there a way to resolve this? or force it to look again for new devices?

Koenkk, 2019-11-28:
This is the entry: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L2950

milobt, 2019-11-30:
The bulbs will connect, but do not show up in the MQTT integrations page.
I have removed and re-added; restarted mqtt and zigbee2mqtt plugins and still not showing up.

If I send a get command:
`zigbee2mqtt:info  2019-11-30T18:24:41: MQTT publish: topic 'zigbee2mqtt/0x00124b001ed009c4', payload '{"state":"ON","brightness":0,"linkquality":21}'`

If I try a set command, including a brightness, I get the following: 
`zigbee2mqtt:error 2019-11-30T18:24:41: No converter available for 'Brightness' (50)`

I have also paired a LED strip controller, Gleopto GL-C-007, which paired, showed up in the integrations and was controllable, but has since stopped responding to commands.

milobt, 2020-01-16:
I am still having issues with the Gledopto GL-S-007ZS bulbs and think it may be something to do with the xy values when converted to mqtt.

I have a number of bulbs that have been grouped and tend to work ~OK, but still do not always turn ON when stating a brightness, etc. If just turning ON or OFF, they generally work fine.

I have two bulbs (either side of our bed) that I've left as individuals and control separately, these hardly ever turn on when sent with a brightness, but will work fine when just ON or OFF.

I have noticed that the x and y values sent to these bulbs are generally 15 decimal places long, rather than the general 3 decimal places; some times these values are 4 places, but sometimes work, e.g.:

`zigbee2mqtt:info  2020-01-16 08:23:30: MQTT publish: topic 'zigbee2mqtt/Miles Spot', payload '{"state":"OFF","brightness":190,"color_temp":386,"color":{"x":0.469092603126886,"y":0.412452218918178}}'
zigbee2mqtt:info  2020-01-16 08:23:30: MQTT publish: topic 'zigbee2mqtt/Nichola Spot', payload '{"state":"OFF","color_temp":401,"color":{"x":0.477890571505866,"y":0.413792775585208},"brightness":190}'
zigbee2mqtt:info  2020-01-16 08:23:30: MQTT publish: topic 'zigbee2mqtt/kitchen_spots', payload '{"state":"OFF","brightness":130,"color_temp":374,"color":{"x":0.4534,"y":0.432}}'`

The top two devices (Miles Spot and Nichola Spot) are the individual bulbs, the third is a group (kitchen_spots).

Is there a converter missing for these bulbs?
I noticed that there are no converters for the GL-S bulbs or RGB strip controllers GL-C-008S / 9 (there is one for GL-C-008), so may just needs the S variant adding.
Although, I am unsure whether the converter would limit the xy value and whether the current 15 decimal places is normal or not?

I also don't get a Link Quality for my GL-S-007ZS bulbs, so can't tell if their problem is related to a weak signal.

I notice that in the devices.js files for the GL-S-007Z that you have the additional line:
`meta: {options: {disableDefaultResponse: true}},`

But not for the GL-S-007ZS, is this intentional?

Finally, I tend to get a lot of Mac and timeout errors, are these caused by weak signals/interference or other reason?
`zigbee2mqtt:error 2020-01-16 08:51:30: Publish 'set' 'state' to 'Headboard' failed: 'Error: Data request failed with error: 'MAC no ack' (233)'
zigbee2mqtt:error 2020-01-16 08:51:15: Publish 'set' 'state' to 'Headboard' failed: 'Error: Timeout - 17817 - 15 - 131 - 6 - 11 after 15000ms'`