Koenkk/zigbee2mqtt #2424: Paulmann Licht sends every 3 seconds

pepp86, 2019-11-27:
Hi, I have a "Paulmann SmartHome Zigbee LED-stripe (50045)" connected to my coordinator. Everything is working well, but this devices sends every 3 seconds its status (without reporting enabled).

Has anyone else the same issue? Is there a way to configure the device to not spam the network?

Thx
Patrick

Koenkk, 2019-11-27:
I'm not sure where this spam comes from, this bulb is not configured to do so.

monotonus, 2019-11-27:
I have the same issue with a Müller Licht Tint E27 bulb: https://www.zigbee2mqtt.io/devices/404006_404008_404004.html

After updating to version 1.7.1 it started to send every 1-3 seconds an "update":

Nov 27 23:59:29 hassbian npm[466]: #033[32mzigbee2mqtt:info #033[39m 2019-11-27T22:59:29: MQTT publish: topic 'zigbee2mqtt/Schlafzimmerlampe', payload '{"state":"OFF","linkquality":21,"brightness":14,"color_temp":370,"color_mode":2,"color":{"x":0.4522,"y":0.4085}}'
Nov 27 23:59:32 hassbian npm[466]: #033[32mzigbee2mqtt:info #033[39m 2019-11-27T22:59:32: MQTT publish: topic 'zigbee2mqtt/Schlafzimmerlampe', payload '{"state":"OFF","linkquality":21,"brightness":14,"color_temp":370,"color_mode":2,"color":{"x":0.4522,"y":0.4085}}'
Nov 27 23:59:33 hassbian npm[466]: #033[32mzigbee2mqtt:info #033[39m 2019-11-27T22:59:33: MQTT publish: topic 'zigbee2mqtt/Schlafzimmerlampe', payload '{"state":"OFF","linkquality":21,"brightness":14,"color_temp":370,"color_mode":2,"color":{"x":0.4522,"y":0.4085}}'
Nov 27 23:59:35 hassbian npm[466]: #033[32mzigbee2mqtt:info #033[39m 2019-11-27T22:59:35: MQTT publish: topic 'zigbee2mqtt/Schlafzimmerlampe', payload '{"state":"OFF","linkquality":21,"brightness":14,"color_temp":370,"color_mode":2,"color":{"x":0.4522,"y":0.4085}}'
Nov 27 23:59:36 hassbian npm[466]: #033[32mzigbee2mqtt:info #033[39m 2019-11-27T22:59:36: MQTT publish: topic 'zigbee2mqtt/Schlafzimmerlampe', payload '{"state":"OFF","linkquality":21,"brightness":14,"color_temp":370,"color_mode":2,"color":{"x":0.4522,"y":0.4085}}'
Nov 27 23:59:38 hassbian npm[466]: #033[32mzigbee2mqtt:info #033[39m 2019-11-27T22:59:38: MQTT publish: topic 'zigbee2mqtt/Schlafzimmerlampe', payload '{"state":"OFF","linkquality":18,"brightness":14,"color_temp":370,"color_mode":2,"color":{"x":0.4522,"y":0.4085}}'
Nov 27 23:59:39 hassbian npm[466]: #033[32mzigbee2mqtt:info #033[39m 2019-11-27T22:59:39: MQTT publish: topic 'zigbee2mqtt/Schlafzimmerlampe', payload '{"state":"OFF","linkquality":21,"brightness":14,"color_temp":370,"color_mode":2,"color":{"x":0.4522,"y":0.4085}}'
Nov 27 23:59:41 hassbian npm[466]: #033[32mzigbee2mqtt:info #033[39m 2019-11-27T22:59:41: MQTT publish: topic 'zigbee2mqtt/Schlafzimmerlampe', payload '{"state":"OFF","linkquality":21,"brightness":14,"color_temp":370,"color_mode":2,"color":{"x":0.4522,"y":0.4085}}'

There are nearly no updates. From time to time the linkquality differs a little bit, but the other values stay the same.

Maybe we also need to update the coordinator firmware itself? I'm waiting for my CC debugger atm.

Koenkk, 2019-11-28:
While reporting is disabled now, did you enable it once? If so, please re-pair the device, otherwise the reporting configuration is still present on the device.

monotonus, 2019-11-28:
Yes I also had reporting enabled in the configuration.yaml, but it never had an effect. After removing reporting and repairing the device all is fine now again for me.

pepp86, 2019-11-29:
Disabled reporting, restarted and repaired the device and now the device is no longer spamming the network.

Koenkk, 2019-11-29:
To fix this issue when reporting is enabled, I need to have the debug logging of the spam.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

pepp86, 2019-11-29:
Here you go:

https://pastebin.com/LSTMktsA

Had debugging enabled before

Koenkk, 2019-11-30:
Can you:

- Remove the device from the network
- Stop zigbee2mqtt
- Change https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/deviceReport.js#L10 to `minimumReportInterval: 3, maximumReportInterval: 300, reportableChange: 1,` 
- Start zigbee2mqtt with reporting enabled.
- Pair the device
- Does it still report every 3 seconds now?

pepp86, 2019-12-02:
Will try it on the weekend and let you know the result.

pepp86, 2019-12-06:
Seems to work - but would this affect any other functions?

Koenkk, 2019-12-07:
Nope, it means that they will only report when the value has changed (by a minimal value of 1). Closing this, thanks!

closed 2019-12-07 by Koenkk