Koenkk/zigbee2mqtt #2425: node.js binding

Spyky, 2019-11-27:
Error: The module '/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/Release/bindings.node'
was compiled against a different Node.js version using
Nov 27 17:52:25 raspberrypi npm[23771]: NODE_MODULE_VERSION 79. This version of Node.js requires
Nov 27 17:52:25 raspberrypi npm[23771]: NODE_MODULE_VERSION 64. Please try re-compiling or re-installing

I tried reinstalling and some other this, but still the same issue.
Couldn't  find a solution. Can you help?

Koenkk, 2019-11-27:
What is your setup, do you use virtualenv?

Did you try `rm -rf node_modules && npm install`?

Spyky, 2019-11-27:
Hello

Yes I did that. I'm using a raspberry pi with domoticz. Use allready mqtt
for sonoff stuff.

Greetings,

Spyky, 2019-11-28:
> What is your setup, do you use virtualenv?
> 
> Did you try `rm -rf node_modules && npm install`?

Tried it again, but without result. Added some info. 
[zigbee2mqtt.txt](https://github.com/Koenkk/zigbee2mqtt/files/3902424/zigbee2mqtt.txt)

Koenkk, 2019-11-28:
Can you try `npm rebuild @serialport/bindings --update-binary`.

Spyky, 2019-11-28:
Hello Koen,

Still the same problem.

Koenkk, 2019-11-29:
Can you try https://github.com/serialport/node-serialport/issues/1910#issuecomment-524949720

Spyky, 2019-11-29:
> Can you try serialport/node-serialport#1910 (comment)

Didn't found the folders

   - Remove from the node-modules folder the serialport and @serialport
    folders.
   - Remove the file packages-lock.json

   How do i run this  ./node_modules/.bin/electron-rebuild

   - Can't find this folder

Spyky, 2019-11-29:
Found the rebuild and did
npm install --save-dev electron-rebuild

Still the some in the log:
-- Logs begin at Fri 2019-11-29 18:05:38 CET. --
Nov 29 18:06:29 raspberrypi npm[1444]: npm ERR! Failed at the
zigbee2mqtt@1.7.1 start script.
Nov 29 18:06:29 raspberrypi npm[1444]: npm ERR! This is probably not a
problem with npm. There is likely additional logging output above.
Nov 29 18:06:29 raspberrypi npm[1444]: npm ERR! A complete log of this run
can be found in:
Nov 29 18:06:29 raspberrypi npm[1444]: npm ERR!
/home/pi/.npm/_logs/2019-11-29T17_06_29_733Z-debug.log
Nov 29 18:06:29 raspberrypi systemd[1]: zigbee2mqtt.service: Main process
exited, code=exited, status=1/FAILURE
Nov 29 18:06:29 raspberrypi systemd[1]: zigbee2mqtt.service: Unit entered
failed state.
Nov 29 18:06:29 raspberrypi systemd[1]: zigbee2mqtt.service: Failed with
result 'exit-code'.
Nov 29 18:06:30 raspberrypi systemd[1]: zigbee2mqtt.service: Service
hold-off time over, scheduling restart.
Nov 29 18:06:30 raspberrypi systemd[1]: Stopped zigbee2mqtt.service.
Nov 29 18:06:30 raspberrypi systemd[1]: Started zigbee2mqtt.service.
Nov 29 18:06:31 raspberrypi npm[1512]: > zigbee2mqtt@1.7.1 start
/opt/zigbee2mqtt
Nov 29 18:06:31 raspberrypi npm[1512]: > node index.js
Nov 29 18:06:33 raspberrypi npm[1512]: zigbee2mqtt:info
 2019-11-29T17:06:33: Logging to directory:
'/opt/zigbee2mqtt/data/log/2019-11-29.18-06-33'
Nov 29 18:06:34 raspberrypi npm[1512]: zigbee2mqtt:info
 2019-11-29T17:06:34: Starting zigbee2mqtt version 1.7.1 (commit #b459c35)
Nov 29 18:06:34 raspberrypi npm[1512]: zigbee2mqtt:info
 2019-11-29T17:06:34: Starting zigbee-herdsman...
Nov 29 18:06:34 raspberrypi npm[1512]: zigbee2mqtt:error
2019-11-29T17:06:34: Error while starting zigbee-herdsman
Nov 29 18:06:34 raspberrypi npm[1512]: zigbee2mqtt:error
2019-11-29T17:06:34: Failed to start zigbee
Nov 29 18:06:34 raspberrypi npm[1512]: zigbee2mqtt:error
2019-11-29T17:06:34: Exiting...
Nov 29 18:06:34 raspberrypi npm[1512]: zigbee2mqtt:error
2019-11-29T17:06:34: Error: The module
'/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/Release/bindings.node'
Nov 29 18:06:34 raspberrypi npm[1512]: was compiled against a different
Node.js version using
Nov 29 18:06:34 raspberrypi npm[1512]: NODE_MODULE_VERSION 79. This version
of Node.js requires
Nov 29 18:06:34 raspberrypi npm[1512]: NODE_MODULE_VERSION 64. Please try
re-compiling or re-installing
Nov 29 18:06:34 raspberrypi npm[1512]: the module (for instance, using `npm
rebuild` or `npm install`).
Nov 29 18:06:34 raspberrypi npm[1512]:     at
Object.Module._extensions..node (internal/modules/cjs/loader.js:807:18)
Nov 29 18:06:34 raspberrypi npm[1512]:     at Module.load
(internal/modules/cjs/loader.js:653:32)
Nov 29 18:06:34 raspberrypi npm[1512]:     at tryModuleLoad
(internal/modules/cjs/loader.js:593:12)
Nov 29 18:06:34 raspberrypi npm[1512]:     at Function.Module._load
(internal/modules/cjs/loader.js:585:3)
Nov 29 18:06:34 raspberrypi npm[1512]:     at Module.require
(internal/modules/cjs/loader.js:692:17)
Nov 29 18:06:34 raspberrypi npm[1512]:     at require
(internal/modules/cjs/helpers.js:25:18)
Nov 29 18:06:34 raspberrypi npm[1512]:     at bindings
(/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/bindings/bindings.js:112:48)
Nov 29 18:06:34 raspberrypi npm[1512]:     at Object.<anonymous>
(/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/lib/linux.js:2:36)
Nov 29 18:06:34 raspberrypi npm[1512]:     at Module._compile
(internal/modules/cjs/loader.js:778:30)
Nov 29 18:06:34 raspberrypi npm[1512]:     at Object.Module._extensions..js
(internal/modules/cjs/loader.js:789:10)
[... 286 lines cut ...]
Nov 29 18:06:51 raspberrypi npm[1733]: npm ERR! zigbee2mqtt@1.7.1 start:
`node index.js`
Nov 29 18:06:51 raspberrypi npm[1733]: npm ERR! Exit status 1
Nov 29 18:06:51 raspberrypi npm[1733]: npm ERR!
Nov 29 18:06:51 raspberrypi npm[1733]: npm ERR! Failed at the
zigbee2mqtt@1.7.1 start script.
Nov 29 18:06:51 raspberrypi npm[1733]: npm ERR! This is probably not a
problem with npm. There is likely additional logging output above.
Nov 29 18:06:51 raspberrypi npm[1733]: npm ERR! A complete log of this run
can be found in:
Nov 29 18:06:51 raspberrypi npm[1733]: npm ERR!
/home/pi/.npm/_logs/2019-11-29T17_06_51_448Z-debug.log
Nov 29 18:06:51 raspberrypi systemd[1]: zigbee2mqtt.service: Main process
exited, code=exited, status=1/FAILURE
Nov 29 18:06:51 raspberrypi systemd[1]: zigbee2mqtt.service: Unit entered
failed state.
Nov 29 18:06:51 raspberrypi systemd[1]: zigbee2mqtt.service: Failed with
result 'exit-code'.
Nov 29 18:06:51 raspberrypi systemd[1]: zigbee2mqtt.service: Service
hold-off time over, scheduling restart.
Nov 29 18:06:51 raspberrypi systemd[1]: Stopped zigbee2mqtt.service.
Nov 29 18:06:51 raspberrypi systemd[1]: Started zigbee2mqtt.service.
Nov 29 18:06:53 raspberrypi npm[1784]: > zigbee2mqtt@1.7.1 start
/opt/zigbee2mqtt
Nov 29 18:06:53 raspberrypi npm[1784]: > node index.js

Spyky, 2019-12-01:
Koen,

It is working now.
We installed it again in de zigbee folder.
Next we going to try to get the Ikea sonos remote working.

Thank you for you're help

Koenkk, 2019-12-01:
Great! Then I will close this one.

closed 2019-12-01 by Koenkk

Spyky, 2019-12-24:
Hello Koen,

After working now i again have a problem. After we turned the electric power of for a few hours and started domoticz again the Xiaomi Zigbee motion sensor was gone. In hardware Zigbee2mqtt is still there. Did a lot of things to  check if everything is ok and can't find the problem.
When i try to add the sensor it is not in the device list. 
In domoticz setting i allow to add devices.
Hold the button on the sensor in for 5 seconds and it blinks. 
Have you any idea what i'm doing wrong and why it is gone af the electric was turned off?

This looks ok
 zigbee2mqtt.service
   Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2019-12-24 18:47:59 CET; 3s ago
 Main PID: 1839 (npm)
   CGroup: /system.slice/zigbee2mqtt.service
           ├─1839 npm
           ├─1861 sh -c node index.js
           └─1862 node index.js

Dec 24 18:47:59 raspberrypi systemd[1]: Started zigbee2mqtt.service.
Dec 24 18:48:00 raspberrypi npm[1839]: > zigbee2mqtt@1.7.1 start /opt/zigbee2mqtt
Dec 24 18:48:00 raspberrypi npm[1839]: > node index.js

USB:
P: /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/tty/ttyACM0
N: ttyACM0
S: serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B00194A1B36-if00
S: serial/by-path/platform-3f980000.usb-usb-0:1.3:1.0
E: DEVLINKS=/dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B00194A1B36-if00 /dev/serial/by-path/platform-3f980000.usb-usb-0:1.3:1.0
E: DEVNAME=/dev/ttyACM0
E: DEVPATH=/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/tty/ttyACM0
E: ID_BUS=usb
E: ID_MODEL=TI_CC2531_USB_CDC
E: ID_MODEL_ENC=TI\x20CC2531\x20USB\x20CDC
E: ID_MODEL_ID=16a8
E: ID_PATH=platform-3f980000.usb-usb-0:1.3:1.0
E: ID_PATH_TAG=platform-3f980000_usb-usb-0_1_3_1_0
E: ID_REVISION=0009
E: ID_SERIAL=Texas_Instruments_TI_CC2531_USB_CDC___0X00124B00194A1B36
E: ID_SERIAL_SHORT=__0X00124B00194A1B36
E: ID_TYPE=generic

This is in the configurationfile
# allow new devices to join
permit_join: true
# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: /dev/ttyACM0