Koenkk/zigbee2mqtt #2426: Heiman HS2SK - Device needs to paired again after restart of zigbee2mqtt

TheSkater, 2019-11-27:
Since my HS2SK device is now working fine (see:  #2277) I have a new problem. 
When I stop the zigbee2mqtt service and start it it again I have to pair the device manally again!
All devices are present, only the HS2SK is missing.

Koenkk, 2019-11-28:
What do you see in the log when controlling the device after a restart?

TheSkater, 2019-11-29:
The device is present at the start but will then be removed due to a `force_remove` message which is still in the system. I issued this message some days before.
Can some one tell me how I can undo or disable the `force_remove` when starting zigbee2mqtt?

`info  2019-11-29 16:14:03: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-11-29.16-14-02'
debug 2019-11-29 16:14:03: Removing old log directory '/opt/zigbee2mqtt/data/log/2019-11-23.03-02-37'
debug 2019-11-29 16:14:03: Loaded state from file /opt/zigbee2mqtt/data/state.json
info  2019-11-29 16:14:05: Starting zigbee2mqtt version 1.7.1+dev (commit #d37d9ee)
info  2019-11-29 16:14:05: Starting zigbee-herdsman...
debug 2019-11-29 16:14:05: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
info  2019-11-29 16:14:07: zigbee-herdsman started
info  2019-11-29 16:14:07: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
debug 2019-11-29 16:14:07: Zigbee network parameters: {"panID":6754,"extendedPanID":"0xdddddddddddddddd","channel":11}
info  2019-11-29 16:14:07: Currently 11 devices are joined:

info  2019-11-29 16:14:07: Keller/Wasserwerk/Zigbee/Schalter (0x000d6f000b7a7240): HS2SK - HEIMAN Smart metering plug (Router)
warn  2019-11-29 16:14:07: `permit_join` set to  `true` in configuration.yaml.
warn  2019-11-29 16:14:07: Allowing new devices to join.
warn  2019-11-29 16:14:07: Set `permit_join` to `false` once you joined all devices.
info  2019-11-29 16:14:07: Zigbee: allowing new devices to join.
info  2019-11-29 16:14:07: Connecting to MQTT server at mqtt://localhost
info  2019-11-29 16:14:07: Connected to MQTT server
info  2019-11-29 16:14:07: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'

info  2019-11-29 16:14:07: MQTT publish: topic 'zigbee2mqtt/Keller/Wasserwerk/Zigbee/Schalter', payload '{"voltage":235.25,"linkquality":84,"current":3.28,"power":706,"state":"ON"}'

debug 2019-11-29 16:14:07: Received MQTT message on 'zigbee2mqtt/bridge/config/force_remove' with data 'Keller/Wasserwerk/Zigbee/Schalter'
info  2019-11-29 16:14:07: Force removing 'Keller/Wasserwerk/Zigbee/Schalter'
info  2019-11-29 16:14:07: Successfully force_removed Keller/Wasserwerk/Zigbee/Schalter`

TheSkater, 2019-12-02:
I delete the device with the ` force_remove` message before. This message was obviously send with the `retained` flag. So it had to delete the message out of the mqtt server with the followng description:  
`https://community.openhab.org/t/clearing-mqtt-retained-messages/58221`

closed 2019-12-02 by TheSkater