Koenkk/zigbee2mqtt #2427: Monitoring zigbee2mqtt when started as a service

regissg, 2019-11-27:
# Bug Report
## What happened
zigbee2mqtt runs well as a service on my windows PC but when I stop the service and try to run npm start in a cmd window (in order to monitor it) I have the following error message:

zigbee2mqtt:info  2019-11-27T20:52:51: Logging to directory: 'C:\Program Files\zigbee2mqtt\data\log\2019-11-27.21-52-50'
(node:4052) UnhandledPromiseRejectionWarning: Error: EPERM: operation not permitted, unlink 'C:\Program Files\zigbee2mqtt\data\log\2019-11-27.18-15-53\log.txt'
    at Object.unlinkSync (fs.js:1035:3)
    at rimrafSync (C:\Program Files\zigbee2mqtt\node_modules\rimraf\rimraf.js:309:17)
    at C:\Program Files\zigbee2mqtt\node_modules\rimraf\rimraf.js:344:39
    at Array.forEach (<anonymous>)
    at rmkidsSync (C:\Program Files\zigbee2mqtt\node_modules\rimraf\rimraf.js:344:26)
    at rmdirSync (C:\Program Files\zigbee2mqtt\node_modules\rimraf\rimraf.js:337:7)
    at Function.rimrafSync [as sync] (C:\Program Files\zigbee2mqtt\node_modules\rimraf\rimraf.js:307:9)
    at C:\Program Files\zigbee2mqtt\lib\util\logger.js:77:20
    at Array.forEach (<anonymous>)
    at DerivedLogger.cleanup (C:\Program Files\zigbee2mqtt\lib\util\logger.js:75:21)
(node:4052) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:4052) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
  
## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:  
C253X firmware version:

Koenkk, 2019-11-28:
Looks like a permission issue, try running zigbee2mqtt as admin.

regissg, 2019-11-28:
You are right it was that. I should have think about it...

closed 2019-11-28 by regissg