Koenkk/zigbee2mqtt #2428: Status of the group not updating

gaggio, 2019-11-28:
# Bug Report
## What happened
Defined a group of lights, configured the group in home assistant, verified it is correctly working as expected: light can be switched on and off, brightess/temperature tuning. 
Availability feature is active and correctly working for all the devices, included the ones part of the group. 
When switching on and off the light by means of the light switch, the status of the devices (lights) part of the group is correctly changing accordingly. The status of the group is not changing.

## What did you expect to happen
Status of the group to be updated according to the status of the devices in it.

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.7.1
CC253X firmware version: {"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}

Koenkk, 2019-11-28:
Can you provide your configuration? Are you sure you didn't put `optimistic: false` for the group? (https://www.zigbee2mqtt.io/information/groups.html#state-changes)

gaggio, 2019-11-28:
Thanks for your reply, this is the configuration of my group (removed the devices names, but there are three devices in the actual group):

```# ID, each group should have a different numerical ID
'1':
  # Required: Name which will be used to control the group
  friendly_name: 'tavolo_cucina'
  # Required: Retain messages (true/false) (default: false)
  retain: false
  # Optional: Default transition to be used when e.g. changing brightness (in seconds) (default: 0)
  transition: 2
  # Optional: Change group state when one of the devices in it changes state, see 'State changes' below (default: true)
  optimistic: true
  # Optional: Devices of this group,
  # Note: this has to be the ieeeAddr of the device, not the friendly_name! (default: empty)
  devices:
```

Let me further clarify, main issue is that when I switch off the lamps cutting power by means of the wall switch, the lights are shown as "unavailable" while the group appears still on.

Koenkk, 2019-11-29:
> Let me further clarify, main issue is that when I switch off the lamps cutting power by means of the wall switch, the lights are shown as "unavailable" while the group appears still on.

The behaviour of this indeed is not defined yet, the question is what should it be? Should the group become unavailable if one of the device becomes unavailable (seems unhandy to me as other bulbs could still be controlled)? @gabe565 Do you have an opinion about this?

sjorge, 2019-11-29:
Tossing in my two cents, maybe have a additional property to indicate the group is consistent (aka all devices report the same state) or inconsistent (some device have a different state, e.g. 3 bulbs online/on, one offline/unknown (assume off).

Then you still update the group when a change happens but we also indicate if it is consistent or not?

gaggio, 2019-11-30:
> Should the group become unavailable if one of the device becomes unavailable (seems unhandy to me as other bulbs could still be controlled)?

I expected the group to become unavailable if ALL the the devices become unavailable.

closed 2020-02-06 by stale[bot]