Koenkk/zigbee2mqtt #2429: Can't get npm started with zigbee2mqtt (New to this)

experiencingiot, 2019-11-28:
Hello,

I'm new to zigbee2mqtt and also raspberry even. What i want to do is to use zigbee2mqtt with iobroker mqtt server. Application is to get / send data to/from zigbee philips devices. 
My coordinator cc2531 is flashed with new firmware (just got it purchased).
However, when i run npm.. it never starts and displays some errors like can't open serial port.
I'm wondering that it happened 2 to 3 times that zigbee herdsmen got started (without any changing in the setup), however, never established connection with mqtt server.
But most of the times, herdsmen never starts. 
Please help me cover up this problem as a beginner. 

The link to npm log is given below:

https://pastebin.com/UgmfprxS

Looking for friendly responses.

Koenkk, 2019-11-28:
Hi!

Did you check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start ?

experiencingiot, 2019-12-01:
Yes, I have surely gone through this link. I'm having correct port in configuration.yaml file. Also i tested the command "test -w [PORT] && echo success || echo failure" which gives success to me every time. Any more hints?

davet2001, 2019-12-23:
It looks more like npm is not starting properly rather than a port error. 

Did you run ‘npm install’ before ‘npm start’?

closed 2020-02-21 by Koenkk