Koenkk/zigbee2mqtt #2430: Brightness not updating correctly (IKEA LED1537R6 / E1743)

polskafan, 2019-11-28:
# Bug Report
## What happened
I put four IKEA LED1537R6 spots in a group and bound an IKEA E1743 on/off dimmer to the group. When I use the dimmer to change the state, only the ON/OFF state gets updated on the /get endpoint, but the brightness stays at the old value.

## How to reproduce it
1. Bind a dimmer to a group
2. Change the brightness of the group with the dimmer
3. Use the /get endpoint to retrieve the current brightness level

Bulbs were dimmed down via MQTT brightness: 0 and then dimmed up with the bound dimmer remote:
> {"state": ""} -> zigbee2mqtt/office-spot-1/get
> zigbee2mqtt/office-spot-1 -> {"state":"ON","brightness":0,"linkquality":89,"color_temp":450}

## Debug Info
zigbee2mqtt version: 1.7.1 / 1.7.1-dev (commit #99fd93f)
CC2531 firmware version: 20190608

Would it also be possible to get a MQTT event, after the bound dimmer switch has changed the state? It seems that zigbee2mqtt gets at least the information, when the bulbs turn on again, but no MQTT message is sent, when that happens.

Koenkk, 2019-11-29:
For this you should use the report feature: https://www.zigbee2mqtt.io/information/report.html

polskafan, 2019-11-29:
I somehow missed this feature in the documentation. This fixes the wrong brightness values and the missing MQTT on/off events. Thank you.

The Philips Hue Bloom seems to be generating additional state updates, when it gets switched from the coordinator with this setting. It first sends the old and then immediately afterwards the new state.

Switch to red, and later to blue:
> Nov 29 13:08:22 nodered npm[4583]: zigbee2mqtt:info  2019-11-29 13:08:22: MQTT publish: topic 'zigbee2mqtt/office-bloom', payload '{"state":"ON","linkquality":105,"brightness":255,"color":{"x":0.5061,"y":0.456}}'
> Nov 29 13:08:22 nodered npm[4583]: zigbee2mqtt:info  2019-11-29 13:08:22: MQTT publish: topic 'zigbee2mqtt/office-bloom', payload '{"state":"ON","linkquality":105,"brightness":255,"color":{"x":0.6803,"y":0.3034}}'
> Nov 29 13:08:29 nodered npm[4583]: zigbee2mqtt:info  2019-11-29 13:08:29: MQTT publish: topic 'zigbee2mqtt/office-bloom', payload '{"state":"ON","linkquality":105,"brightness":255,"color":{"x":0.6803,"y":0.3034}}'
> Nov 29 13:08:29 nodered npm[4583]: zigbee2mqtt:info  2019-11-29 13:08:29: MQTT publish: topic 'zigbee2mqtt/office-bloom', payload '{"state":"ON","linkquality":105,"brightness":255,"color":{"x":0.1434,"y":0.0591}}'

And I think, I'll need a more powerful coordinator now. The CC2531 seems to be at its limit with 7 lamps and 3 remotes.

Thank you very much though. I think this can be closed.

closed 2019-11-29 by polskafan