Koenkk/zigbee2mqtt #2431: color_temp

zhapwey, 2019-11-28:
hello:

Is there a maximum and a minimum for the color_temp value of the bulbs?
And another thing I don't know is the hue?

thanks

sjorge, 2019-11-29:
The zcl specifications have this to day about color_temp
```
Color temperature = 1,000,000 / ColorTemperature in the range 1 to 65279
inclusive, giving a color temperature range from 1,000,000 Kelvins to 15.32
Kelvins).

The value ColorTemperature = 0 indicates an undefined value. The value
ColorTemperature = 65535 indicates an invalid value.
If this attribute is implemented then the ColorMode attrib
```

There seems to be no limit set on it in the zigbee-herdsman-converters, https://github.com/Koenkk/zigbee-herdsman-converters/blob/51a14aaa270df6b9ea7a6f2689675aabb024d9d7/converters/toZigbee.js#L335

gdave321, 2019-12-04:
For my hue lights I use  minimum 153 maximum 500 which seems to give the full range.
According to Philips: "Mired color temperature: 153 (blue, 6500K) - 500 (yellow, 2000K)"

closed 2020-02-09 by stale[bot]