Koenkk/zigbee2mqtt #2433: Aqara Opple Switch failing interview

liftylol, 2019-11-28:
# Bug Report
## What happened
I tried pairing a new Aqara Opple Switch, managed to do so once but didnt get it to pair again after creating a device in device.js and trying to re-pair for troubleshooting.

got 'device has not successfully been paired' after every single of 50+ tries. tried everything from deleting it from the db/config and im literally holding it 1cm away from the dongle. other devices can still pair.

The interview seems to timeout if i can read the output of the zigbee-herdsman debug correctly. the output of the debug is here:  https://pastebin.com/pb9hmdZA

## What did you expect to happen
device should pair

## How to reproduce it (minimal and precise)
try to pair the aqara opple switch

## Debug Info
zigbee2mqtt version: 1.7.1
CC253X firmware version: revision 20190608

liftylol, 2019-11-28:
literally 2minutes after typing this it worked. sorry for the hassle lol

closed 2019-11-28 by liftylol

kirovilya, 2019-12-02:
It is very difficult to pair new Aqara Opple devices lumi.remote.b286opcn01, lumi.remote.b486opcn01, lumi.remote.b686opcn01...
Interview crashes when receiving a description of the endpoints. Sometimes it goes fine, sometimes not.
But, I found an interesting changes in zigpy project related to inactive endpoints https://github.com/zigpy/zigpy/pull/266