Koenkk/zigbee2mqtt #2434: E1744 with duplicated commands

soloam, 2019-11-29:
I paired for the first time the Ikea E1744 (sonos dimmer). And it's sending duplicated commands on the click, double click and triple click. If I click once I receive 2 inputs on the mqtt with the same json. Also the dim sends a lot of duplicate inputs to a micro rotation! If I make a very small rotation I receive around 22 messages. 

Thank you

Koenkk, 2019-11-30:
@ssiegel could you help with this?

ssiegel, 2019-12-01:
I think Koenkk/zigbee-herdsman-converters#792 should fix this.

The move messages are are synthesized by the ictcg1 function (https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js#L94) which should, as I understand it, generate one message every 200ms. I still see more messages than that, and I still see two rotate_stop at the end of the movement. I currently can't figure out what might cause the duplication. In the debug log there is exactly one message “Received Zigbee message from 'Ikea Symfonisk E1744', type 'commandMove', cluster 'genLevelCtrl', data '{"movemode":0,"rate":195}' from endpoint 1 with groupID 0” when starting the rotation and one message “Received Zigbee message from 'Ikea Symfonisk E1744', type 'commandStop', cluster 'genLevelCtrl', data '{}' from endpoint 1 with groupID 0” when stopping it.

Koenkk, 2019-12-03:
@ssiegel did you check against the latest zigbee2mqtt dev?

soloam, 2019-12-04:
I can try the latest DEV, is this submitted to the dev?

Koenkk, 2019-12-04:
yes

soloam, 2019-12-06:
I tried to update to the lates dev, now I can't start zigbee2mqtt

coordeinator CC1352P-2

The detailed log 
`
DEBUG=* npm start

> zigbee2mqtt@1.7.1 start /opt/zigbee2mqtt
> node index.js

winston:create-logger Define prototype method for "error"
winston:create-logger Define prototype method for "warn"
winston:create-logger Define prototype method for "info"
winston:create-logger Define prototype method for "http"
winston:create-logger Define prototype method for "verbose"
winston:create-logger Define prototype method for "debug"
winston:create-logger Define prototype method for "silly"
winston:create-logger Define prototype method for "error"
winston:create-logger Define prototype method for "warn"
winston:create-logger Define prototype method for "info"
winston:create-logger Define prototype method for "http"
winston:create-logger Define prototype method for "verbose"
winston:create-logger Define prototype method for "debug"
winston:create-logger Define prototype method for "silly"
zigbee2mqtt:info  2019-12-06 20:12:19: Logging to console and directory: 'data/log'
winston:file written true false
zigbee2mqtt:debug 2019-12-06 20:12:22: Loaded state from file /opt/zigbee2mqtt/data/state.json
winston:file written true false
winston:file logged 72 info  2019-12-06 20:12:19: Logging to console and directory: 'data/log'

winston:file logged 155 debug 2019-12-06 20:12:22: Loaded state from file /opt/zigbee2mqtt/data/state.json

winston:file ENOENT ok data/log/log.txt
winston:file stat done: log.txt { size: 0 }
winston:file create stream start data/log/log.txt { flags: 'a' }
winston:file create stream ok data/log/log.txt
winston:file file open ok data/log/log.txt
zigbee2mqtt:info  2019-12-06 20:12:22: Starting zigbee2mqtt version 1.7.1+dev (commit #3320136)
winston:file written true false
zigbee2mqtt:info  2019-12-06 20:12:22: Starting zigbee-herdsman...
winston:file written true false
zigbee2mqtt:debug 2019-12-06 20:12:22: Using zigbee-herdsman with settings: '{"network":{"panID":6758,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
winston:file written true false
  serialport/bindings loading LinuxBinding +0ms
  serialport/stream .list +0ms
winston:file logged 84 info  2019-12-06 20:12:22: Starting zigbee2mqtt version 1.7.1+dev (commit #3320136)

winston:file logged 139 info  2019-12-06 20:12:22: Starting zigbee-herdsman...

winston:file logged 568 debug 2019-12-06 20:12:22: Using zigbee-herdsman with settings: '{"network":{"panID":6758,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'

  zigbee-herdsman:adapter Path '/dev/ttyACM0' is valid for 'ZStackAdapter' +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6758,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json"}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyACM0 and {"baudRate":115200,"rtscts":true,"autoOpen":false} +0ms
  serialport/stream opening path: /dev/ttyACM0 +1s
  serialport/binding-abstract open +0ms
  serialport/stream _read queueing _read for after open +20ms
  serialport/bindings/poller Creating poller +0ms
  serialport/stream opened path: /dev/ttyACM0 +21ms
  serialport/stream _read reading { start: 0, toRead: 65536 } +7ms
  serialport/binding-abstract read +56ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +96ms
  zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +6ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  serialport/stream _write 1 bytes of data +46ms
  serialport/binding-abstract write 1 bytes +50ms
  serialport/bindings/unixWrite Starting write 1 bytes offset 0 bytesToWrite 1 +0ms
  serialport/bindings/unixRead Starting read +0ms
  serialport/bindings/unixWrite write returned: wrote 1 bytes +29ms
[... 242 lines cut ...]
zigbee2mqtt:error 2019-12-06 20:13:27: Error while starting zigbee-herdsman
winston:file written true false
zigbee2mqtt:error 2019-12-06 20:13:27: Failed to start zigbee
winston:file written true false
zigbee2mqtt:error 2019-12-06 20:13:27: Exiting...
winston:file written true false
zigbee2mqtt:error 2019-12-06 20:13:27: Error: AREQ - ZDO - stateChangeInd after 60000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
winston:file written true false
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.7.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.7.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-12-06T20_13_27_712Z-debug.log

`

soloam, 2019-12-06:
I rolled back to master branch and all is back online!

soloam, 2019-12-07:
Ok I tested out the new "3320136" commit! The problem now is different.

MQTT from 90º rotation none stop, I think that it should be on rotate_left and one rotate_stop at the end. A none stopping rotation produced several rotate_stop.

```
{"action":"rotate_left","linkquality":0}
{"action":"rotate_left","rate":195,"linkquality":0}
{"action":"rotate_left","linkquality":0}
{"action":"rotate_left","rate":195,"linkquality":0}
{"action":"rotate_left","linkquality":0,"brightness":127}
{"action":"rotate_left","linkquality":0,"brightness":107}
{"action":"rotate_stop","brightness":92,"linkquality":0}
{"action":"rotate_stop","brightness":92,"linkquality":0}
{"action":"rotate_stop","brightness":92,"linkquality":0}
{"action":"rotate_stop","brightness":92,"linkquality":0}
```

A single click in the button generates 2 messages
```
{"action":"play_pause","linkquality":57}
{"action":"play_pause","linkquality":57}
```

Double click in the button generates 2 messages
```
{"action":"skip_forward","step_size":1,"transition_time":0,"linkquality":39}
{"action":"skip_forward","step_size":1,"transition_time":0,"linkquality":39}
```

Thank you

Koenkk, 2019-12-07:
There are also some additional configuration recommendation, see https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/devices/E1744.md#recommendation (but for this you need to be on the dev branch).

soloam, 2019-12-08:
Tested the config, and that solves the problem of duplicated outputs! A lot better, thank you! For me, I only have one problem with this, don't know if it's a limitation from the hardware part! The button, on rotate sends rotation followed by a stop, even if I keep spinning the wheel without stopping! This is a problem because I can't detect the speed that I'm spinning, if a I spin very fast I get only one rotation and one stop! I could use the brightness property to detect, but I would like the rotation to be relative to the actual state of the lamp, the bigness on the wheel is relative to the button! If my lamp is at 50% and I spin it right, I what to go up from 50%, and not get the brightness from the button to send it to the lamp! I could detect the difference from the bugging of the bigness of the wheel and the end, and calculate the delta, to increase or decrease that value from the lamp, but if the wheel is at 1 and I go down, or if it's 255 and I go up, the value is always the min or max!

Thank you for your hard work!

elektrikal80, 2019-12-27:
Hello all, 
I am new here and a beginner,
Mij problem is : I have 2 traverdi ikea led's , one LED1836G9 and a LED1842G3, my system runs on a rasperry pi and cc2531 sniffer  with mqtt and zigbeemqtt with domoticz  all installe with the tutorial from [](https://github.com/Koenkk/zigbee2mqtt), now i can add the led 1842G3 in domoticz but the other led wil not connect and in domoticz  in the log file can i see strange things:

Received available devices list from bridge
Device **LED1842G3** 0x14b457fffe44d4ad (0x14b457fffe44d4ad)
Received available devices list from bridge
Device **LED1842G3** 0x14b457fffe44d4ad (0x14b457fffe44d4ad)
Device 0x14b457fffe2e953a (0x14b457fffe2e953a) doesn't have "model" attribute, skipped

the led that alreddy runs in domoticz is LED 1842G3, 
the led wich i want to add is 1836G9. nd not 1842G3
Hope that anyone can help me

closed 2020-03-05 by stale[bot]