Koenkk/zigbee2mqtt #2436: devices in networkmap detached

hajo62, 2019-11-29:
I'm using networkmap and there is one device which I can see the values off, detached.
First open [this issue](https://github.com/azuwis/zigbee2mqtt-networkmap/issues/14), but was pointed to zigbee2mqtt:
---
**Home Assistant version**
v0.101.3

**Zigbee2mqtt version**
v1.7.1

**OS and browser version**
MacOS Firefox v70.0.1 or Chrome v78.0.3904.108

**Error message on browser**
No error

**State attributes of `sensor.zigbee2mqtt_networkmap`**
```
nodes:
  - ieeeAddr: '0x00124b00193648ca'
    friendlyName: Coordinator
    type: Coordinator
    networkAddress: 0
    failed: []
    lastSeen: null
  - ieeeAddr: '0x00158d0002b5196f'
    friendlyName: XiaomiAqara-1
    type: EndDevice
    networkAddress: 26058
    manufacturerName: LUMI
    modelID: lumi.weather
    lastSeen: 1575015908044
  - ieeeAddr: '0x00158d0002e23355'
    friendlyName: XiaomiSmart-1
    type: EndDevice
    networkAddress: 11126
    manufacturerName: LUMI
    modelID: lumi.sens
    lastSeen: null
  - ieeeAddr: '0x00158d00025ee3a6'
    friendlyName: XiaomiAqara-2
    type: EndDevice
    networkAddress: 36594
    manufacturerName: LUMI
    modelID: lumi.weather
    lastSeen: 1575016030142
  - ieeeAddr: '0x000b57fffe271050'
    friendlyName: '0x000b57fffe271050'
    type: EndDevice
    networkAddress: 19118
    manufacturerName: IKEA of Sweden
    modelID: TRADFRI wireless dimmer
    lastSeen: null
links:
  - source:
      ieeeAddr: '0x000b57fffe271050'
      networkAddress: 19118
    target:
      ieeeAddr: '0x00124b00193648ca'
      networkAddress: 0
    linkquality: 170
    depth: 1
    routes: []
    sourceIeeeAddr: '0x000b57fffe271050'
    targetIeeeAddr: '0x00124b00193648ca'
    sourceNwkAddr: 19118
    lqi: 170
    relationship: 1
  - source:
      ieeeAddr: '0x00158d0002b5196f'
      networkAddress: 26058
    target:
      ieeeAddr: '0x00124b00193648ca'
      networkAddress: 0
    linkquality: 5
    depth: 1
    routes: []
    sourceIeeeAddr: '0x00158d0002b5196f'
    targetIeeeAddr: '0x00124b00193648ca'
    sourceNwkAddr: 26058
    lqi: 5
    relationship: 1
  - source:
      ieeeAddr: '0x00158d00025ee3a6'
      networkAddress: 36594
    target:
      ieeeAddr: '0x00124b00193648ca'
      networkAddress: 0
    linkquality: 26
    depth: 1
    routes: []
    sourceIeeeAddr: '0x00158d00025ee3a6'
    targetIeeeAddr: '0x00124b00193648ca'
    sourceNwkAddr: 36594
    lqi: 26
    relationship: 1
  - source:
      ieeeAddr: '0x001788010479f6d9'
      networkAddress: 45798
    target:
      ieeeAddr: '0x00124b00193648ca'
      networkAddress: 0
    linkquality: 0
    depth: 1
    routes: []
    sourceIeeeAddr: '0x001788010479f6d9'
    targetIeeeAddr: '0x00124b00193648ca'
    sourceNwkAddr: 45798
    lqi: 0
    relationship: 1
  - source:
      ieeeAddr: '0x0000000000000000'
      networkAddress: 11126
    target:
      ieeeAddr: '0x00124b00193648ca'
      networkAddress: 0
    linkquality: 0
    depth: 255
    routes: []
    sourceIeeeAddr: '0x0000000000000000'
    targetIeeeAddr: '0x00124b00193648ca'
    sourceNwkAddr: 11126
    lqi: 0
    relationship: 3
friendly_name: Zigbee2mqtt Networkmap
```

**Home Assistant log**
nothing relevant

Koenkk, 2019-11-30:
If only the end devices are detached, this is not really strange, they can sleep for a long time causing them to drop from the routers routing tables.

hajo62, 2019-12-03:
It's only one device, always the same and it sends values...
It has been connected for a while, but since some weeks, it isn't anymore.

[image]

The one on the left isn't connected in the map, the one on the right is...

Stooovie, 2019-12-03:
I have the same issue after swapping one CC2531 adapter for another. All end devices are detached, even when they all work just fine. Bridge v 1.7.1., firmware rev 20190608.

The only thing that could be different is some kind of UUID but I can't find one for the bridge (adapter), only for the end devices. Any tips?

pcb1962, 2019-12-04:
This is not an issue at all, I have a load of end devices showing as not connected in my map, it's a cosmetic issue only, makes absolutely no difference to operation.
[image]

hajo62, 2019-12-04:
Sorry, but we are not discussing if the devices are working. 
Here we discuss, that the networkmap does not show them as connected... It's cosmetic, but that the thing the network map does...

Stooovie, 2019-12-04:
It's an entire point of the zigbee map :)

hajo62, 2020-02-02:
@Koenkk : Do you need some additional information/logs etc.?

Koenkk, 2020-02-02:
@hajo62 What kind of devices show disconnected?

hajo62, 2020-02-02:
I have three xiaomi sensors (plus my neighbors ikea).
Two Xiaomi [Aqara](https://xiaomi-mi.de/mi-smart-home/aqara-temperature-and-humidity-sensor) - which are _connected_
One Xiaomi [Smart](https://xiaomi-mi.de/mi-smart-home/aqara-temperature-and-humidity-sensor/) - not connected, but has been some month ago...

definitio, 2020-02-02:
@Koenkk I have the same problem. [Xiaomi MiJia switches](https://www.zigbee2mqtt.io/devices/WXKG01LM.html), [Xiaomi Aqara switches](https://www.zigbee2mqtt.io/devices/WXKG11LM.html), [Xiaomi MiJia door & window contact sensors](https://www.zigbee2mqtt.io/devices/MCCGQ01LM.html) and [Xiaomi Aqara door & window contact sensors](https://www.zigbee2mqtt.io/devices/MCCGQ11LM.html) are disonnected.

Koenkk, 2020-02-02:
For Xiaomi devices this is common, my assumption is that the devices sleep a long time causing them to be removed from the router table. Therefore they will not be included in the link quality queries. Functional wise it doesn't matter, but they just don't show up in the network map.

hajo62, 2020-02-02:
And there is no (easy) way to re-include those in the router table?

Koenkk, 2020-02-03:
I don't think so, note that this is not the routing table of the coordinator but of the e.g. bulbs (ac powered devices) in your network. Zigbee2mqtt doesn't have influence on that.

SargonofAssyria, 2020-02-24:
I have the same issue. BUT! I have 2 identical devices: https://www.zigbee2mqtt.io/devices/WXKG02LM.html
They are both close to an AC powered device: https://www.zigbee2mqtt.io/devices/ZNLDP12LM.html
The one that has a little bit more distance to the bulb is ALWAYS connected to the bulb and the other one is  never connected to the bulb or any other device in the networkmap.
Is n't that strang?

lolorc, 2020-03-26:
after upgrading my cc2652r to firmware 20200312, a lot of devices are working but shown as disconnected.
(don't if it comes from the firmware upgrade or from the cc2652r being power cycled)
re-pairing a devices makes it "connected" on the map again.

[image]

hajo62, 2020-05-25:
> this is not the routing table of the coordinato

I have only the coordinator and Xiaomi 3 edge devices. No other device.

closed 2020-07-26 by stale[bot]

Maurotb, 2020-07-26:
SAme problem here, after replacing my cc2531 i have some disconnected device in my map but device works correctly

Smitkas, 2020-07-29:
[image]
same problem here, devices not sleep a long time (PIR and switches in use)

tbnobody, 2020-08-03:
Have exactly the same problem as @Maurotb. I replaced my cc2531 with an CC2652RB. the network works correctly but 2 devices where shown detached. I repaired one of this devices again and now it shows connected. 
But one thing is strange (except for the one not connected device). 
The coordinator is in room 1, and a router is in room 2. I have several battery powered devices in both rooms. But all end devices are shown as connected to the router. No device seems to be connected directly to the coordinator. 
[image]

Maurotb, 2020-08-08:
I have replaced cc2531 with another cc2531 (same firmware) and have a problem...
This issue is now closed,  @Koenkk   no solution for this issue?

Koenkk, 2020-08-08:
@Maurotb end devices can shown as deattached in the networkmap when they are sleeping for too long, the routers mark them as inactive (there is nothing z2m can do about this). This **does not** mean that they are not in the network anymore!

Maurotb, 2020-09-01:
I have try with latest zigbee2mqtt , the problem is present for active sensor (i have open a door with door sensor)
In network map is disconnected
I have try with webgraphviz and  zigbee2mqtt-networkmap plugin

pcb1962, 2020-09-01:
> The problem for me begin after a cc2531 replacement. I have try other cc3531, same problem

In what sense is it a problem?

Maurotb, 2020-09-02:
Before changing cc 2531 the network map was correct. 
After changing the cc2531, while the zigbee network is working, the network map from several disconnected devices (almost all)
Is not a sleep problem, i have try to trigger some sensors but in network map stay disconnected (the trigger is correctly detected by zigbee2mqtt and passet to mosquitto)

pcb1962, 2020-09-02:
So your zigbee network is working correctly, the only 'problem' is that you see disconnected devices in the network map. 
The reason for that has been explained, it is not a 'problem', it is a cosmetic issue only.

Maurotb, 2020-09-02:
@pcb1962 yes is a cosmetic issue,
I wanted to report that it doesn't look like a sleep problem but other problem if you want to solve it

Maurotb, 2020-10-27:
fo info,
i have change my cc2531 with a new zig-a-zig-ah! CC2652R , firmware  CC26X2R1_20200925.
Rejoin all devices.
Same problem, some device are disconnected, all bticino switch (with router cappabilities) and two aquara device.
 maybe the problem is that you can't see the routers?

tbnobody, 2020-11-09:
I've upgraded zigbee2mqtt to 1.16 and the firmware to CC2652RB_20201026.hex
For me the issue seems to be resolved. No orphan devices anymore.

Maurotb, 2020-11-10:
Hi, for me no luck, i have upgrade firmware and zigbee2mqtt to version 1.16.1, now i have all devices disconnected in my map