Koenkk/zigbee2mqtt #2437: Publish 'set' 'state' to '0x7cb03eaa0a01b20d' failed: 'Error: Data request failed with error: 'MAC channel access failure' (225)'

wojciechczyz, 2019-11-29:
# Bug Report
## What happened

This problem repeated twice by now, reinstalling Home Assistant from scratch.
When trying to change state on any switch/ligth I am getting this error:

Publish 'set' 'state' to '0x7cb03eaa0a01b20d' failed: 'Error: Data request failed with error: 'MAC channel access failure' (225)'

Seems like it is affecting all router devices, all battery operated devices keep operating.

## What did you expect to happen

Switch should work, light should shine
## How to reproduce it (minimal and precise)

I have setup hassio over Open Media Valult with installation script.
I have paired my devices - xiaomi sensors (battery operated) such as temperature, door opening, movement.
I have paired lights  - Gledopto controllers.
I have paired Osram Smart + plug 
I have paired Philips plug

Waited 2 days (few restarts in middle)

Now only battery operating devices are alived.

THis happened 2 times already.
I have moved stick from one port to another in raspberry - would be strange if related (Xiaomi devices work happilly)

## Debug Info
zigbee2mqtt version:
1.7.1
CC253X firmware version:
20190608, here is initializing line:
zigbee2mqtt:info  2019-11-29T12:42:42: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.7.1","commit":"unknown","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}'

Here is my startup log:

zigbee2mqtt:warn  2019-11-29T12:42:41: Allowing new devices to join.
zigbee2mqtt:warn  2019-11-29T12:42:41: Set `permit_join` to `false` once you joined all devices.
zigbee2mqtt:info  2019-11-29T12:42:41: Zigbee: allowing new devices to join.
zigbee2mqtt:info  2019-11-29T12:42:41: Connecting to MQTT server at mqtt://core-mosquitto
zigbee2mqtt:info  2019-11-29T12:42:42: Connected to MQTT server
zigbee2mqtt:info  2019-11-29T12:42:42: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt:info  2019-11-29T12:42:42: MQTT publish: topic 'zigbee2mqtt/0x00158d0003cb1567', payload '{"battery":91,"voltage":2985,"temperature":22.87,"humidity":58.7,"linkquality":110}'
zigbee2mqtt:info  2019-11-29T12:42:42: MQTT publish: topic 'zigbee2mqtt/0x00158d0003d49fa5', payload '{"battery":80,"voltage":2965,"linkquality":47,"contact":false}'
zigbee2mqtt:info  2019-11-29T12:42:42: MQTT publish: topic 'zigbee2mqtt/0x001788010804158f', payload '{"state":"ON","linkquality":78}'
zigbee2mqtt:info  2019-11-29T12:42:42: MQTT publish: topic 'zigbee2mqtt/0x7cb03eaa0a01b20d', payload '{"state":"ON","linkquality":68}'
zigbee2mqtt:info  2019-11-29T12:42:42: MQTT publish: topic 'zigbee2mqtt/0x00124b001ead90e9', payload '{"state":"OFF"}'
zigbee2mqtt:info  2019-11-29T12:42:42: MQTT publish: topic 'zigbee2mqtt/0x00158d00044d5e6b', payload '{"battery":100,"voltage":3015,"linkquality":76,"contact":false}'
zigbee2mqtt:info  2019-11-29T12:42:42: MQTT publish: topic 'zigbee2mqtt/0x00124b001d43cbed', payload '{"state":"OFF","brightness":67,"color":{"x":0.581,"y":0.388},"color_temp":162}'
zigbee2mqtt:info  2019-11-29T12:42:42: MQTT publish: topic 'zigbee2mqtt/0x00158d00044db511', payload '{"battery":100,"voltage":3035,"temperature":23.45,"humidity":55.02,"pressure":1014,"linkquality":68}'
zigbee2mqtt:info  2019-11-29T12:42:42: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.7.1","commit":"unknown","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}'
zigbee2mqtt:info  2019-11-29T12:42:42: MQTT publish: topic 'homeassistant/sensor/0x00158d0003cb1567/temperature/config', payload '{"unit_of_measurement":"°C","device_class":"temperature","value_template":"{{ value_json.temperature }}","state_topic":"zigbee2mqtt/0x00158d0003cb1567","json_attributes_topic":"zigbee2mqtt/0x00158d0003cb1567","name":"0x00158d0003cb1567_temperature","unique_id":"0x00158d0003cb1567_temperature_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d0003cb1567"],"name":"0x00158d0003cb1567","sw_version":"Zigbee2mqtt 1.7.1","model":"MiJia temperature & humidity sensor (WSDCGQ01LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2019-11-29T12:42:42: MQTT publish: topic 'homeassistant/sensor/0x00158d0003cb1567/humidity/config', payload '{"unit_of_measurement":"%","device_class":"humidity","value_template":"{{ value_json.humidity }}","state_topic":"zigbee2mqtt/0x00158d0003cb1567","json_attributes_topic":"zigbee2mqtt/0x00158d0003cb1567","name":"0x00158d0003cb1567_humidity","unique_id":"0x00158d0003cb1567_humidity_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d0003cb1567"],"name":"0x00158d0003cb1567","sw_version":"Zigbee2mqtt 1.7.1","model":"MiJia temperature & humidity sensor (WSDCGQ01LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2019-11-29T12:42:42: MQTT publish: topic 'homeassistant/sensor/0x00158d0003cb1567/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x00158d0003cb1567","json_attributes_topic":"zigbee2mqtt/0x00158d0003cb1567","name":"0x00158d0003cb1567_battery","unique_id":"0x00158d0003cb1567_battery_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d0003cb1567"],"name":"0x00158d0003cb1567","sw_version":"Zigbee2mqtt 1.7.1","model":"MiJia temperature & humidity sensor (WSDCGQ01LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2019-11-29T12:42:42: MQTT publish: topic 'homeassistant/sensor/0x00158d0003cb1567/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x00158d0003cb1567","json_attributes_topic":"zigbee2mqtt/0x00158d0003cb1567","name":"0x00158d0003cb1567_linkquality","unique_id":"0x00158d0003cb1567_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d0003cb1567"],"name":"0x00158d0003cb1567","sw_version":"Zigbee2mqtt 1.7.1","model":"MiJia temperature & humidity sensor (WSDCGQ01LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2019-11-29T12:42:42: MQTT publish: topic 'homeassistant/binary_sensor/0x00158d0003d49fa5/contact/config', payload '{"payload_on":false,"payload_off":true,"value_template":"{{ value_json.contact }}","device_class":"door","state_topic":"zigbee2mqtt/0x00158d0003d49fa5","json_attributes_topic":"zigbee2mqtt/0x00158d0003d49fa5","name":"0x00158d0003d49fa5_contact","unique_id":"0x00158d0003d49fa5_contact_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d0003d49fa5"],"name":"0x00158d0003d49fa5","sw_version":"Zigbee2mqtt 1.7.1","model":"MiJia door & window contact sensor (MCCGQ01LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2019-11-29T12:42:42: MQTT publish: topic 'homeassistant/sensor/0x00158d0003d49fa5/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x00158d0003d49fa5","json_attributes_topic":"zigbee2mqtt/0x00158d0003d49fa5","name":"0x00158d0003d49fa5_battery","unique_id":"0x00158d0003d49fa5_battery_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d0003d49fa5"],"name":"0x00158d0003d49fa5","sw_version":"Zigbee2mqtt 1.7.1","model":"MiJia door & window contact sensor (MCCGQ01LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2019-11-29T12:42:42: MQTT publish: topic 'homeassistant/sensor/0x00158d0003d49fa5/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x00158d0003d49fa5","json_attributes_topic":"zigbee2mqtt/0x00158d0003d49fa5","name":"0x00158d0003d49fa5_linkquality","unique_id":"0x00158d0003d49fa5_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d0003d49fa5"],"name":"0x00158d0003d49fa5","sw_version":"Zigbee2mqtt 1.7.1","model":"MiJia door & window contact sensor (MCCGQ01LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2019-11-29T12:42:42: MQTT publish: topic 'homeassistant/switch/0x001788010804158f/switch/config', payload '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state }}","comman [... cut ...] 88010804158f"}}'
zigbee2mqtt:error 2019-11-29T12:44:20: Publish 'set' 'state' to '0x001788010804158f' failed: 'Error: Data request failed with error: 'MAC channel access failure' (225)'
zigbee2mqtt:info  2019-11-29T12:44:20: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x001788010804158f' failed: 'Error: Data request failed with error: 'MAC channel access failure' (225)'","meta":{"friendly_name":"0x001788010804158f"}}'

xader, 2019-12-08:
encountered same issue:
zigbee2mqtt:error 2019-12-08T17:52:40: Publish 'set' 'state' to 'salon_komputer' failed: 'Error: Data request failed with error: 'MAC channel access failure' (225)'

0.7.1 @ 20190608 @ RPI4 powered by Raspberry usb-c supply
It only fails if external SSD is plugged in, literally - If I unplug  SSD - everything works, as soon as I plug SSD back - I cannot control anything, just receive updates.

Will try to find powered hub to check if it's some kind of undervoltage issue.

Koenkk, 2019-12-09:
In case you are still on the `20190608` firmware, you might want to try the source routing firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

xader, 2019-12-10:
@Koenkk 
Tried source routing-

Basically same issue with different symptoms:
- either zigbee2mqtt fails to initialize:
zigbee2mqtt:error 2019-12-10T18:11:00: Error while starting zigbee-herdsman
zigbee2mqtt:error 2019-12-10T18:11:00: Failed to start zigbee
zigbee2mqtt:error 2019-12-10T18:11:00: Exiting...
zigbee2mqtt:error 2019-12-10T18:11:00: Error: AREQ - ZDO - stateChangeInd after 60000ms

or 225 error:
zigbee2mqtt:debug 2019-12-10T18:31:17: Received MQTT message on 'zigbee2mqtt/salon_komputer/set' with data '{"state": "ON"}'
zigbee2mqtt:debug 2019-12-10T18:31:17: Publishing 'set' 'state' to 'salon_komputer'
zigbee2mqtt:error 2019-12-10T18:31:17: Publish 'set' 'state' to 'salon_komputer' failed: 'Error: Data request failed with error: 'MAC channel access failure' (225)'
zigbee2mqtt:info  2019-12-10T18:31:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'salon_komputer' failed: 'Error: Data request failed with error: 'MAC channel access failure' (225)'","meta":{"friendly_name":"salon_komputer"}}'

or 205 error:

zigbee2mqtt:debug 2019-12-10T18:54:24: Received MQTT message on 'zigbee2mqtt/salon_komputer/set' with data '{"state": "OFF"}'
zigbee2mqtt:debug 2019-12-10T18:54:24: Publishing 'set' 'state' to 'salon_komputer'
zigbee2mqtt:error 2019-12-10T18:54:24: Publish 'set' 'state' to 'salon_komputer' failed: 'Error: Data request failed with error: 'No network route' (205)'
zigbee2mqtt:info  2019-12-10T18:54:24: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'salon_komputer' failed: 'Error: Data request failed with error: 'No network route' (205)'","meta":{"friendly_name":"salon_komputer"}}'

If USB SSD (bx200 connected through JMS567 SATA->USB bridge) is disconnected everything works.

I've tried powered hub but either system doesn't start or cannot enumerate cc2531.
In a meantime I've ordered USB meter to check voltages and current,  if that's going nowhere I've spare CC2531 to test.

Koenkk, 2019-12-11:
> If USB SSD (bx200 connected through JMS567 SATA->USB bridge) is disconnected everything works.

This sound like interference is going on, try to connect the CC2531 through an USB extension cable to and move it away from the pi.

justin8, 2019-12-20:
I had the same issue; just incase someone else gets to this. My setup had been working for about 2 months with zero issues; I moved the server running it (and it's external hard drive) to a different part of my house, and 4 days later all zigbee devices were unresponsive, reporting `MAC channel access failure`. The CC2531 was plugged in to the slot right beneath an external HDD cable as well. Restarting the service and the server had no result.

I put it on a USB extension cable, about 30cm away, restarted zigbee2mqtt and everything came back on perfectly.

shafr, 2020-01-07:
I have same exact error, nothing had changed on my host configuration - I've just paired extra xiaomi temperature sensor. Restarting service does not work. Unplugging / plugging does not work.

xescure, 2020-01-19:
@xader Save yourself a few bucks, I don't think it's a power issue. I've just cut the power wires of my USB -> SATA adapter and connected it directly to an external PSU (so it is literally drawing 0A from the USB port) and the interference is still present. I'll try to do justin8's solution and connect it via an extender cable, hopefully that works.

coolguymatt, 2020-02-04:
@xescure did the extender cable fix your issue? I have started to have this with my Philips Hue bulbs.

xescure, 2020-02-27:
@coolguymatt Yes, the extender fixed everything. It needs to be more than 30cm away from my Pi for it to be reliable.

marrold, 2020-05-01:
FWIW I had the same issue with a Raspberry Pi 4 + SSD + CC26X2R1, and extending the cable did the trick. Thanks

odx, 2020-05-15:
Just wanted to confirm that USB drives may cause this sort of interference. I added a USB HDD and the network stopped working. Moving the CC1352P-2 away from the Pi / HDD it reverts back to normal.

wojciechczyz, 2020-05-18:
This is old case, I fixed it by long cable indeed. thank you for help!

closed 2020-05-18 by wojciechczyz

arturdd, 2020-06-01:
A week of life wasted on flashing cc2531/ re-configuring / pairing devices. I was about to order a second CC2531 adapter because I thought my device is broken.
It appears all my issues were caused by some interferences generated by the Raspberry Pi  or an SSD disk connected to USB.
My CC2531 was plugged directly into rpi4 USB port.  An extension USB cable solved all my issues.
@Koenkk  May I suggest changing status of USB cable from "Optional" to "Highly Recommended" here https://www.zigbee2mqtt.io/getting_started/what_do_i_need.html ?

Koenkk, 2020-06-01:
@arturdd done, thanks!

etique57, 2021-12-11:
> A week of life wasted on flashing cc2531/ re-configuring / pairing devices. I was about to order a second CC2531 adapter because I thought my device is broken. It appears all my issues were caused [...]

Just wanted to say thanks because I had the exact same problem just today and looking for the error led me directly here.

danir-de, 2023-01-05:
Having the same problem and nothing helping (USB extension cable, interference check, using USB Hubs, etc) on a raspberry pi 4b, the problem seems to be solved by just plugging the Zigbee Stick into an USB 2 port instead of using the USB 3 port.
Maybe this helps someone in the future.

Koenkk, 2023-01-06:
@danir-de thanks for letting us know, added this to the FAQ.

gamberr0, 2023-06-29:
Tenía el mismo problema desde que instalé HomeAssistant en un SSD en una raspi 4. El coordinador zigbee no funcionaba.

Puse un cable extensor USB con el coordinador y todo volvió a funcionar.

Inexplicable, 

muchas gracias por la ayuda