Koenkk/zigbee2mqtt #2438: philips hue dimmer not working

pcfreak1, 2019-11-29:
i have an issue with my philips hue dimmer (324131092621)
i paired it successfully but it does not generate any mqtt messages
am i doing something wrong or is it not fully supported yet because it is in the supported devices i tought it would work but it clearly does not.

sjorge, 2019-11-29:
Did it successfully join and complete all the configuration steps? Mine work and generates events for all buttons. Although when joining it did take a few tries before it succesfully completed the whole interview.

pcfreak1, 2019-11-29:
which steps should i perform then?

sjorge, 2019-11-29:
I'd probbaly start with:
- removing the device
- enabling debug logging
- enable permit_join
- add the device (click some of the buttons every second or so until you see the 'Succesfully interviewd' message)

If it still doesn't work after that, you can post the debug log.

Garfonso, 2019-12-10:
I have a similar behaviour here. I use iobroker ( https://github.com/ioBroker/ioBroker.zigbee ) and upgraded to herdsman a few days ago. Everything worked fine, including the hue dimmer switches.

Then, today, I switched to a new coordinator ( CC26X2R1 ) and ZStack3 and had to repair all my devices. Since that the dimmer switches won't work anymore. Behaviour is quite strange. The adapter (sorry, don't know if I can get to log output from herdsman in iobroker somewhere) seems happy about the join process:
https://pastebin.com/9AX7BqVn

But after that I do not receive any button presses. Also the LED in front of the dimmer switches won't light up on button presses anymore (before it would emit green light for successful transmitted command and red if there were connection issues). 

So there seems to be something wrong somewhere... I'd be glad if I can be of any more help.

Garfonso, 2019-12-11:
This might be a duplicate of #2224 

Question: why is configuration not attempted directly after pairing? Or is it? I could not see debug output for that.

emilklindt, 2020-01-05:
Based on the information provided in this issue, I am not certain that it is related to my experience with the Philips Hue Dimmer Switch. However, I also experience similar (but probably not completely identical) problems with the switch.

Typically the switch works properly for a while. Then every once in a while (between a few minutes and multiple hours) the switch locks up completely. Each time a button is pressed in this state, the light indicator on the switch blinks red. I've posted a log of such an occurrence [here](https://gist.github.com/emilklindt/f2bf0c7ff3c27b2200c9365960fe80f2#file-bedroom_switch_unresponsive-log).

Other times the MQTT messages are published correctly, but often with a few seconds delay. It seems that it might be caused by the device being unreachable and then reconnecting multiple times, but that is only a guess based on the log messages. The log from this kind of event can be found [here](https://gist.github.com/emilklindt/f2bf0c7ff3c27b2200c9365960fe80f2#file-bedroom_switch_delayed-log).

I'm on the newest `master` branch, version 1.8.0, with the following configuration:
```
{"version":"1.8.0","commit":"da4d26a","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":false}
```

Switching to the `dev` branch did not resolve this issue. Turning on/off the report feature does not seem to make a difference. My switches are able to interview and join successfully. It does not occur when the dimmer switches are connected to the Philips Hue bridge.

Connectivity does not seem to be the problem, as I live in a small apartment and the problem persists when moving the switch right next to the coordinator. If I can be of any help debugging or similar, please let me know!

james-fry, 2020-01-11:
> Each time a button is pressed in this state, the light indicator on the switch blinks red.

I have exactly the same issue.
I have 4 hue dimmer switches and they all do this from time to time.
Now 2 of them dropped out of my network and are not at all responsive.
I will try unpairing and repairing them, but I guess this would only be a temporary fix.

Koenkk, 2020-01-12:
Dropping of is a different issue, see #2693

ZzetT, 2020-01-30:
I see a similar issue. With the Hue bridge, I never had any problems with the dimmer switches. They work very reliable. But when I try to switch them to zigbee2mqtt, I sometimes see a red light after I pressed a button. I can't really say when that happens. If you press a button multiple times in a row, it's very likely that one of the signals shows a red indication.

Btw. is it normal that when I press a button that the action is sometimes empty?
```
zigbee2mqtt:info  2020-01-30 23:16:08: MQTT publish: topic 'zigbee2mqtt/bed_side_remote_alex', payload '{"counter":1,"linkquality":36,"brightness":223,"battery":93.5,"action":"on-press","duration":0}'
zigbee2mqtt:info  2020-01-30 23:16:08: MQTT publish: topic 'zigbee2mqtt/bed_side_remote_alex', payload '{"counter":1,"linkquality":36,"brightness":223,"battery":93.5,"action":""}'
```
Here are some logs from Wireshark. One for the communication between the dimmer switch and zigbee2mqtt. The first button press was "green", the one after the mark showed red:
[image]

And another one for the communication between the dimmer switch and the Hue Bridge:
[image]

I don't see anything useful here. But I think it's weird that the dump for the communication with the Hue Bridge doesn't show any respone to the dimmer (address 0x0006). However it should be, otherwise the dimmer couldn't acknowledge it with a green light? (or I did something wrong here)

Koenkk, 2020-01-31:
@ZzetT interesting finding:

- What coordinator/firmware are you using?
- Besides the red light, does everything else work? (no clicks missed etc.)
- Empty action is expected (when `homeassistant: true` is set in `configuration.yaml`), it's required by home assistant to create automations.

ZzetT, 2020-02-03:
>What coordinator/firmware are you using?

The firmware is zStack12 rev 20190608 and I'm using a CC2531.

>Besides the red light, does everything else work? (no clicks missed etc.)

What I can say is that even if the red light flashes, I can see the clicks in the logs so they don't get lost. If I go to the next room, all signals from the remote are lost and the red light is always shown. Maybe it only works close to a light bulb which acts as a router, because for now I still have a few bulb connected to the Hue Bridge and not to the CC2531.

XanderTenBoden, 2020-04-04:
Does anyone have a solution for this? I've been trying to get my remote back to work again for weeks but so far I had no success, even after switching from a CC2531 to a CC1352P-2. The remote is connected successfully, but button presses are not being received as it seems.

ZzetT, 2020-04-04:
It is working for me, but I agree that it's always a little pain to re-pair Hue devices like the remote or the motioin sensor. I have to try it a few times beforeI see a successful pairing. Is your remote successfully paired?

XanderTenBoden, 2020-04-04:
It seems that it is successfully paired, yes.

ZzetT, 2020-04-04:
Could you try to remove it with "zigbee2mqtt/bridge/config/remove"  (or force remove) as described in https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html and then re-pair it again?

XanderTenBoden, 2020-04-04:
That worked, thanks! \o/

Garfonso, 2020-04-04:
It is important to have it successfully paired and configured. If it seems to be paired but does not work, try to send the configure command right after you reset the remote (ie keep all 4 buttons pressed until the led turns green, then release).

XanderTenBoden, 2020-04-05:
> It is important to have it successfully paired and configured. If it seems to be paired but does not work, try to send the configure command right after you reset the remote (ie keep all 4 buttons pressed until the led turns green, then release).

Got it, thanks for the additional information :-)

closed 2020-06-12 by stale[bot]