Koenkk/zigbee2mqtt #2439: hue_power_on_behavior not working despite new firmware

pronobis, 2019-11-29:
# Bug Report
## What happened
Sending `mosquitto_pub -t 'zigbee2mqtt/hue_color1/set' -m '{ "hue_power_on_behavior": "off"}'` to a HUE color light bulb does not change the power on state to OFF.

The problem occurs for the following bulbs:
```
{"model":"9290012573A","manufacturerID":4107,"manufacturerName":"Philips","modelID":"LCT007","hardwareVersion":17,"softwareBuildID":"5.127.1.26420","dateCode":"20181121"}
{"model":"9290012573A","manufacturerID":4107,"manufacturerName":"Philips","modelID":"LCT014","hardwareVersion":1,"softwareBuildID": "1.46.13_r26312","dateCode":"20181115"}
```

## Debug Info
zigbee2mqtt version: 1.7.1
CC253X firmware version: `{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20180815}}`

cskollar, 2019-12-18:
It's the same here, I also tried with and without json (......"zigbee2mqtt/hue_color_1/set/hue_power_on_behavior" -m "recover"), but it didn't help.

zigbee2mqtt version: 1.8.0

coordinator: CC2531, zStack12, 20190608

bulb:
model:9290012573A
manufacturerID: 4107
manufacturerName: Philips
modelID: LCT015
hardwareVersion: 2
softwareBuildID: 1.46.13_r26312
dateCode: 20181115

The bulb connected to the coordinator and 2 router (a diy CC2531 and a Xiaomi power plug).

sjorge, 2019-12-19:
Maybe it's a general issue, I had issues setting the osram settings using 1.8.0+dev too.
My 2 hue bulbs are on an old firmware though, so they don't work either but that is expected.

sti0, 2019-12-27:
We've got a short power loss yesterday and I tried to set the `hue_power_on_behavior` afterwards, but it don't work for me as well. Tried to set it to `recover` and to `off`.

I'm using an CC26X2R1 with Coordinator firmware version: `'{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}}'`

Hue bulbs:
`{"ieeeAddr":"0x00178801029a765a","type":"Router","networkAddress":64096,"model":"8718696695203","friendly_name":"lgt_buro_strahler","manufacturerID":4107,"manufacturerName":"Philips","powerSource":"Mains (single phase)","modelID":"LTW012","hardwareVersion":1,"softwareBuildID":"1.46.13_r26312","dateCode":"20181115","lastSeen":1577442206304},`
`{"ieeeAddr":"0x0017880102b95d30","type":"Router","networkAddress":29925,"model":"8718696548738","friendly_name":"lgt_buro_stehlampe","manufacturerID":4107,"manufacturerName":"Philips","powerSource":"Mains (single phase)","modelID":"LTW001","hardwareVersion":20,"softwareBuildID":"5.127.1.26420","dateCode":"20181121","lastSeen":1577442231957}`

Kryzek, 2019-12-27:
I also have 9290012573A with "new" firmware 20181115 / 1.46.13_r26312. When used with Hue bridge recovery restored the state bulb was on before losing power. I cannot get this to work with Z2M. Issuing `"hue_power_on_behavior": "recover"` to topic `zigbee2mqtt/0x001788010278b40a/set` where 0x001788010278b40a is my bulb:
`:{"id":16,"type":"Router","ieeeAddr":"0x001788010278b40a","nwkAddr":54154,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"LCT010","epList":[11,242],"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,4096,768,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LCT010","manufacturerName":"Philips","powerSource":1,"zclVersion":1,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20181115","swBuildId":"1.46.13_r26312"}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20181115","swBuildId":"1.46.13_r26312","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1577466600765}`

Pockrandt, 2019-12-28:
"version":"1.8.0-dev"
"commit":"2451d536f2c9e446d396df55c9493f256ab9352e"

same issue cannot get `hue_power_on_behavior` or `osram_set_transitionworking` working for my newly purchased bulbs. (Germany - Amazon)

osram (white dim only)
```
"manufId":48042,
"manufName":"OSRAM",
"modelId":"Classic A60 W clear - LIGHTIFY",
"zclVersion":1,
"appVersion":0,
"stackVersion":2,
"hwVersion":1,
"dateCode":"20140331CNEF****",
"swBuildId":"V1.04.12"
``` 

hue: 
``` 
"manufId":4107,
"manufName":"Philips",
"modelId":"LCA001",
"zclVersion":2,
"appVersion":2,
"stackVersion":1,
"hwVersion":5,
"dateCode":"20190122",
"swBuildId":"1.53.3_r27175"
``` 
Note:
The hue bulb (Bluetooth version) is is labelled as white and color with  Modelnr 9290022168, which is not in the wiki. 
It works fine but throws a interview_failed for all 3 bulbs even after multiple remove and touchlink resets.

kitopopo, 2020-01-03:
Dear friend,

I have add to my zigbee network a philips hue 929001953101, I have sended mqqt messages to the bulb for select the "hue_power_on_behavior" , any command sended works. I test these commands:

`sudo mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/0x0017880104dc1f87/set -m "{\"hue_power_on_behavior\": \"off\"}"`

` sudo mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/0x0017880104dc1f87/set -m "{"hue_power_on_brightness": "100"}"
`
No command works, I always receive the following message:

`MQTT publish: topic 'zigbee2mqtt/0x0017880104dc1f87', payload '{"state":"OFF","brightness":0,"color":{"x":0.682,"y":0.2943},"color_temp":37}'`

I have the 1.8.0 version and CC2531_DEFAULT_20190608.zip. 

Any solution? thanks in advanced

ppetru, 2020-01-04:
Also not working for this bulb (the only one I tried with so far):

{"id":9,"type":"Router","ieeeAddr":"0x00178801020a6b33","nwkAddr":57350,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","model$d":"LCT007","epList":[11,242],"endpoints":{"11":{"$rofId":49246,"epId":11,"devId":528,"inClusterList"$
[0,3,4,5,6,8,4096,768,64513],"outClusterList":[25]$
"clusters":{"genBasic":{"attributes":{"modelId":"L$
T007","manufacturerName":"Philips","powerSource":1$
"zclVersion":1,"appVersion":2,"stackVersion":1,"hw$
ersion":17,"dateCode":"20181121","swBuildId":"5.12$
.1.26420"}},"genOnOff":{"attributes":{"onOff":1}},$
genLevelCtrl":{"attributes":{"currentLevel":254}},$
lightingColorCtrl":{"attributes":{"currentX":29980$
"currentY":26875,"colorTemperature":366}}},"binds"$
[]},"242":{"profId":41440,"epId":242,"devId":97,"i$
ClusterList":[33],"outClusterList":[33],"clusters"$
{},"binds":[]}},"appVersion":2,"stackVersion":1,"h$
Version":17,"dateCode":"20181121","swBuildId":"5.1$
7.1.26420","zclVersion":1,"interviewCompleted":tru$
,"meta":{}}

Command I used: mosquitto_pub -t zigbee2mqtt/0r_bulb/set -m '{"hue_power_on_behavior": "recover"}'

Z2M version 1.8.0
Coordinator: {'type': 'zStack3x0', 'meta': {'transportrev': 2, 'product': 1, 'majorrel': 2, 'minorrel': 7, 'maintrel': 1, 'revision': 20191106}}

Koenkk, 2020-01-06:
Should be fixed in latest dev branch, please confirm.

brammp, 2020-01-06:
- Z2M v1.8.0-dev, commit:6245c72
- CC2531 with Firmware source_routing_20190619
- Bulb,mode: 9290011370 Firmware: 5.127.1.26420

hue_power_on_behavior works with default,on and off parameters  but the recover parameter seems not to work tho.

The bulb has a working recover functionality on the Hue Brige

cskollar, 2020-01-08:
I've experienced the same: HUE on and off option works fine, recover not (with 1.8.0-dev branch).

Philips Hue White Single bulb B22
Model: 8718696449691
Firmware: 1.46.13_r26312

kitopopo, 2020-01-08:
> Should be fixed in latest dev branch, please confirm.

Hi,

When will you release the new dev branch? In the 1.8.0 version the hue_power_on_behavior doesn't works.

Thanks in advanced.

Regards

sti0, 2020-01-08:
> > Should be fixed in latest dev branch, please confirm.
> 
> Hi,
> 
> When will you release the new dev branch? In the 1.8.0 version the hue_power_on_behavior doesn't works.
> 
> Thanks in advanced.
> 
> Regards

You can see the latest dev branch @docker hub (https://hub.docker.com/r/koenkk/zigbee2mqtt/tags). If you use the hassio edge addon it has a delay  of some hours if it is available here (2-3h). You need to uninstall and re-install the addon. Please notice that the dev branches have no release no.

Koenkk, 2020-01-11:
Should be fixed in the latest dev branch, after updating to this you need to set the `hue_power_on_behavior` again.

I noticed in order for the recover to work, there needs to be quite some time between pulling the power and turning it on again (I tested with 15 seconds in between), otherwise it doesn't work (got same behaviour on the Hue bridge).

kitopopo, 2020-01-11:
> Should be fixed in the latest dev branch, after updating to this you need to set the `hue_power_on_behavior` again.
>
> I noticed in order for the recover to work, there needs to be quite some time [...]

Hi,

I wait 15 and 20 seconds but the same problem appears. i send the following message but the hue lamps doen't receive anything, 

`sudo mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/0x0017880104dxxxx/set -m "{\"hue_power_on_behavior\": \"off\"}"`

If i cut the current to the lamps and after i send the message, i receive Mac error acknowledge.

I am looking forward to the new version 1.9.0.

Thanks for your job

Regards

sti0, 2020-01-11:
`recover `didn't work for me either. I waited some minutes after cutting the power. `off `works as expected.

kitopopo, 2020-01-17:
Dear Koenkk,

I test the 1.9.0 version and all the states works now correctly: On, Off and revocer. Nice work.

I have found a difference from the bridge hub of Philips. With the philips bridge after a power cut we have the option of choosing a custom color. 

Using your "recover" option we did not recover the last color of the bulb, but simply that if the bulb was on before the power cut the bulb remains on (but in the default philips color -horrible yellow - ) and if the bulb was off After the power cut, the bulb remains off. 

Is there any way to get a "recover" with some custom color or the last color the bulb was in before the power cut?  I've tried to make a domoticz script using "json" so that the light bulb comes on in my favorite color and it works but unfortunately first the lamp is lit a few milliseconds in the philips default colour and after change to my favorite colour.

Thanks for your work.

Best regards

closed 2020-03-24 by stale[bot]

Chrischi-, 2020-05-20:
Is this still unfixed in latest stable?

Getting a `Error: Status 'UNSUPPORTED_ATTRIBUTE'` with [Hue white ambiance GU10 with Bluetooth](https://www.zigbee2mqtt.io/devices/929001953301.html) lamps.

Firmware: 1.65.11_hB798F2BF (latest FW, mid may, removes coil whine via SW 'magic')

```
Mai 20 15:04:57 z2m npm[30184]: zigbee2mqtt:debug 2020-05-20 15:04:57: Received MQTT message on 'zigbee2mqtt/0x00178801084573fa/set' with data '{"hue_power_on_behavior":"recover","hue_power_on_brightness":255,"hue_power_on_color_temperature":1}'
Mai 20 15:04:57 z2m npm[30184]: zigbee2mqtt:debug 2020-05-20 15:04:57: Publishing 'set' 'hue_power_on_color_temperature' to 'kue_re'
Mai 20 15:04:57 z2m npm[30184]: zigbee2mqtt:debug 2020-05-20 15:04:57: Publishing 'set' 'hue_power_on_behavior' to 'kue_re'
Mai 20 15:04:57 z2m npm[30184]: zigbee2mqtt:error 2020-05-20 15:04:57: Publish 'set' 'hue_power_on_behavior' to 'kue_re' failed: 'Error: Write 0x00178801084573fa/11 lightingColorCtrl({"3":{"value":65535,"type":33}}, {"timeout":10000,"manufacturerCode":4107,"disableDefaultResponse":true}) failed (Error: Status 'UNSUPPORTED_ATTRIBUTE')'
Mai 20 15:04:57 z2m npm[30184]: zigbee2mqtt:debug 2020-05-20 15:04:57: Error: Write 0x00178801084573fa/11 lightingColorCtrl({"3":{"value":65535,"type":33}}, {"timeout":10000,"manufacturerCode":4107,"disableDefaultResponse":true}) failed (Error: Status 'UNSUPPORTED_ATTRIBUTE')
Mai 20 15:04:57 z2m npm[30184]:     at Endpoint.<anonymous> (/home/pi/data/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:158:23)
Mai 20 15:04:57 z2m npm[30184]:     at Generator.next (<anonymous>)
Mai 20 15:04:57 z2m npm[30184]:     at fulfilled (/home/pi/data/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:5:58)
Mai 20 15:04:57 z2m npm[30184]: zigbee2mqtt:info  2020-05-20 15:04:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'hue_power_on_behavior' to 'kue_re' failed: 'Error: Write 0x00178801084573fa/11 lightingColorCtrl({\"3\":{\"value\":65535,\"type\":33}}, {\"timeout\":10000,\"manufacturerCode\":4107,\"disableDefaultResponse\":true}) failed (Error: Status 'UNSUPPORTED_ATTRIBUTE')'","meta":{"friendly_name":"kue_re"}}'
```

Koenkk, 2020-05-21:
@Chrischi- can you check if this has been fixed in the latest dev branch?

Chrischi-, 2020-05-21:
@Koenkk 
The problem is fixed and everything works.
Thank you for the quick response.