Koenkk/zigbee2mqtt #2440: Home Assistant discovery info not updated upon rename

kiall, 2019-11-30:
# Bug Report
## What happened

With HA discovery enabled, renaming a device does not update HA discovery info. A restart of Zigbee2MQTT is reguired to update the discovery info.

## What did you expect to happen

HA discover info should have been updated.

## How to reproduce it (minimal and precise)

* Rename a device via `zigbee2mqtt/bridge/config/rename`
* Observe HA discovery info in MQTT is now incorrect

## Debug Info

zigbee2mqtt version: 1.7.1
CC253X firmware version: CC26X2R1_20191106

mmaster23, 2019-12-08:
I'm seeing the same thing. Had to use the zigbee addresses as names or else the HA integration would break. This bug report showed me the workaround, rebooting zigbee2mqtt service fixes the translation table between HA and Z2M. 

```
Version of Zigbee2MqttAssistant: 0.3.110 (0.3.110+Branch.master.Sha.73b8e3cd600f434f60a45e3b575ff901b4b4e334) Release
Version of Zigbee2Mqtt: 1.7.1+dev
Coordinator version: 20190608
```

domas902, 2019-12-24:
Takže první restart ?

piit79, 2020-02-27:
This was a bit inconvenient when I was migrating my Zigbee network from Zigbee 1.2 on CC2531 to Zigbee 3 on CC26X2R1. I had two z2m instances running during the transition. The sensors would continue working with HA seamlessly when leaving the old network and joining the new network - until they were renamed.

I think it would be great if z2m could announce the device to HA again when renamed. I'm not great at JS/nodejs, but I'll see if I can get a PR started later.

kiall, 2020-03-01:
> I think it would be great if z2m could announce the device to HA again when renamed. I'm not great at JS/nodejs, but I'll see if I can get a PR started later.

I pushed a PR :)

https://github.com/Koenkk/zigbee2mqtt/pull/3040

piit79, 2020-03-01:
> I pushed a PR :)
> 
> #3040

Great job, thanks!

PR #3040 (2020-03-04): Update HA Discovery Info on Rename
Update HA discovery info when a device is renamed. Adds a new deviceRenamed
event to the event bus, allowing the HA extension to react to the rename

Fixes #2440
commit 67b4bcf8 (2020-03-04): Update HA Discovery Info on Rename (#3040)

* Update HA Discovery Info on Rename

Update HA discovery info when a device is renamed. Adds a new deviceRenamed
event to the event bus, allowing the HA extension to react to the rename

Fixes #2440

* Handle Groups correctly wrt HA discovery data refresh

* Cleanup group handling

* Cleanup group handling
closed 2020-03-18 by Koenkk