Koenkk/zigbee2mqtt #2442: Communication between 2 Zigbee2MQTT instances - "cannot handle mqtt message"

IWAsGIT, 2019-11-30:
Hello,
after many unsuccessful tries I have to ask if someone can help me. I use a good working Zigbee2MQTT instance on my RPi3 (zigbee2mqtt 1.6.0). Because there are difficulties to reach every device I want to use a RPi Zero (zigbee2mqtt 1.5.1 because I can't get going version 1.6.0).
The RPi3 is the MQTT server. In the configuration.yaml on the RPi Zero I added the server 'mqtt://IP_RPi3'. Everything connects well, but the RPi Zero seems not to re-send the messages. I see the commands from the RPi3 on the RPiZero but the RPiZero says:

"Cannot handle MQTT message on 'zigbee2mqtt/FRIENDLYNAME_OF_THE_DEVICE/set' with data '{"state":"ON"}'"

They have the same configuration.yaml, I copied it from RPi3 to the RPi Zero and only changed the mqtt-server. So all devices and friendlynames are equal.

Can someone please help me with my problems? I really don't know what's to do:-(

pepp86, 2019-12-01:
It's not working how you tried it. 

You need is a repeater. A zigbee network has only one coordinator. So you can use the same configuration on two devices.

IWAsGIT, 2019-12-01:
Thank you for your response! But isn't the repeater working only via Zigbee and not via MQTT? The RPiZero is too far away for repeating the Zigbee signals so I thought I can use MQTT messages instead? Isn't it possible on this way?

closed 2020-02-06 by stale[bot]