Koenkk/zigbee2mqtt #2444: CC2530+CC2591 with USB-to-TTL adapter not working

hansblafoo, 2019-11-30:
# Bug Report
I want to use my new CC2530+CC2591 as a coordinator for zigbee2mqtt. I started with a fresh install of raspbian-lite buster on a Raspberry Pi 3B with packages updated to the most recent versions. I flashed the firmware accordingly and if I wire the CC2530+CC2591 directly to the GPIO Pins on the Raspberry Pi 3b, it runs flawlessly. Since the GPIO port will be occupied by a different module, I need to use a USB-to-serial adapter for connecting the coordinator to the pi. Although I and configured it as decribed here https://github.com/Koenkk/zigbee2mqtt/issues/231#issuecomment-483063954 but zigbee2mqtt refuses to start and quits with an error message. Thus, there seems to be something wrong with the USB-to-serial adapter but the Pi seems to recognize it properly.

Do you have any idea what the problem is? Did I forget to configure or install something?

## What happened
zigbee2mqtt starting log: https://pastebin.com/uTiphHAC
configuration.yaml: https://pastebin.com/Jy2YMauZ
boot-configuration: https://pastebin.com/gsJTTi2u
usb detection of pi: https://pastebin.com/QJmV7WUb

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.7.1
CC2530 firmware version:  CC2530_CC2591_DEFAULT_20190608 from https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/bin/default/CC2530_CC2591_DEFAULT_20190608.zip

Thanks!

hughb8on, 2019-12-04:
I appear to have the same problem, timeout at waitress.js:44:24. 
I am running Z2M version 1.7.1+dev (commit #2fcddb5).
I am using co-ordinator firmware CC2530ZNP-Prod.bin from CC2530_CC2591_DEFAULT_20190608.zip
I have tried several different USB-TTL converters.
My CC2530_CC2591 is the webee version; the same board works fine as a router.
A CC2531 co-ordinator works fine in the same setup.

hansblafoo, 2019-12-04:
I'm also using the same firmware. Which chipsets are used on your USB-TTL converters? I just ordered a CP2102-based adapter which will arrive some time in January. From the linked issue above, I assumed that the CP2102 is supposed to work.

I'm also using the webee version and a CC2531 runs also fine in my setup. ;-) Did you tried to connect the CC2530/CC2591 via GPIO? Does this work?

hughb8on, 2019-12-04:
I've tried a FT232RL (a genuine one!) and a PL2303. 
Like you, I've ordered a CP2102; mine should be here by the weekend.
At the moment I'm testing on an Intel NUC. The coordinator will end up attached to a Pi, but the RXD pin on that is already in use, so I need to get USB working.
Given that, and the fact you've already proved a GPIO connection works, it hardly seems worth me setting up another Pi to test it.
I'll report back when I've tested a CP2102.

hansblafoo, 2019-12-04:
In the first run, my adapter wasn't flashed properly so it didn't work. That's why I've tested the direct GPIO connection to erase the USB adapter as potential root cause. The question is whether your adapters are flashed properly or whether it is really a USB issue.

hughb8on, 2019-12-06:
I tried with a CP2102 today. Same problem.
At some point I may try again to work out what the problem is, but I have other things I want to do at  the moment.

hansblafoo, 2019-12-06:
Did you already check whether it works via GPIO? As mentioned above, my first flashing was not successful so that even after erasing the USB connection as a point of failure, it didn't work.

hughb8on, 2019-12-07:
I have now fixed my problem, but I no longer think it is the same as yours.
Rather than use the bin file in the zip referenced above, I instead used objcopy to create my own from the hex version.
I don't know why, but it is clear the bin file in the zip is not suitable for flashing using the "alternative" method I use, i.e. CCLoader_x86_64.exe and an arduino.
My CC2530+CC2591 now works, and it doesn't matter which chipset serial adaptor I use.
Your problem is clearly different because your board works when directly connected to an RPI.

hansblafoo, 2019-12-07:
So you also suffered from a wrong flash run like I did in the first place. I'm curious to see whether the CP2102 will fix my problem when it has arrived.

mr2c12, 2020-01-05:
Probably related, I have been suffering from the very same problemi for the past three days.
In my case, a usb to ttl based on cp2102 was the solution. The other usb to ttl I was using couldn't z2m connect to my cc2530+cc2591.

hansblafoo, 2020-01-05:
You're absolutely right. I can confirm it. I just switched to a CP2102 based USB Adapter and it works like a charm.

I close this issue here as it is a non-z2m-related issue.

closed 2020-01-05 by hansblafoo

mr2c12, 2020-01-05:
I would suggest to add this piece of information to the wiki page related to connect a CC2530(+CC2591).
Should we create a new issue for this?

epicurean70, 2020-04-29:
I am having these errors for a CC2530+CC2591 connected to a CP2102

,zigbee2mqtt:info  2020-04-29 12:51:18: Logging to console and directory: '/app/data/log/2020-04-29.12-51-18' filename: log.txt
,zigbee2mqtt:info  2020-04-29 12:51:18: Starting zigbee2mqtt version 1.12.2-dev (commit #ce62f43)
,zigbee2mqtt:info  2020-04-29 12:51:18: Starting zigbee-herdsman...
,zigbee2mqtt:error 2020-04-29 12:51:25: Error while starting zigbee-herdsman
,zigbee2mqtt:error 2020-04-29 12:51:25: Failed to start zigbee
,zigbee2mqtt:error 2020-04-29 12:51:25: Exiting...
,zigbee2mqtt:error 2020-04-29 12:51:25: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
,    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:71:23)
,    at Generator.throw (<anonymous>)
,    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
,npm ERR! code ELIFECYCLE
,npm ERR! errno 1
,npm ERR! zigbee2mqtt@1.12.2-dev start: `node index.js`
,npm ERR! Exit status 1
,npm ERR! 
,npm ERR! Failed at the zigbee2mqtt@1.12.2-dev start script.
,npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
,
,npm ERR! A complete log of this run can be found in:
,npm ERR!     /root/.npm/_logs/2020-04-29T04_51_26_915Z-debug.log
,Using '/app/data' as data directory
,
,> zigbee2mqtt@1.12.2-dev start /app
,> node index.js
,
,zigbee2mqtt:info  2020-04-29 12:52:27: Logging to console and directory: '/app/data/log/2020-04-29.12-52-27' filename: log.txt
,zigbee2mqtt:info  2020-04-29 12:52:27: Starting zigbee2mqtt version 1.12.2-dev (commit #ce62f43)
,zigbee2mqtt:info  2020-04-29 12:52:27: Starting zigbee-herdsman...
,zigbee2mqtt:error 2020-04-29 12:52:35: Error while starting zigbee-herdsman
,zigbee2mqtt:error 2020-04-29 12:52:35: Failed to start zigbee
,zigbee2mqtt:error 2020-04-29 12:52:35: Exiting...
,zigbee2mqtt:error 2020-04-29 12:52:35: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
,    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:71:23)
,    at Generator.throw (<anonymous>)
,    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
,npm ERR! code ELIFECYCLE
,npm ERR! errno 1
,npm ERR! zigbee2mqtt@1.12.2-dev start: `node index.js`
,npm ERR! Exit status 1
,npm ERR! 
,npm ERR! Failed at the zigbee2mqtt@1.12.2-dev start script.
,npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
,
,npm ERR! A complete log of this run can be found in:
,npm ERR!     /root/.npm/_logs/2020-04-29T04_52_36_735Z-debug.log
,Using '/app/data' as data directory
,
,> zigbee2mqtt@1.12.2-dev start /app
,> node index.js
,
,zigbee2mqtt:info  2020-04-29 12:53:37: Logging to console and directory: '/app/data/log/2020-04-29.12-53-37' filename: log.txt
,zigbee2mqtt:info  2020-04-29 12:53:37: Starting zigbee2mqtt version 1.12.2-dev (commit #ce62f43)
,zigbee2mqtt:info  2020-04-29 12:53:37: Starting zigbee-herdsman...
,zigbee2mqtt:error 2020-04-29 12:53:45: Error while starting zigbee-herdsman
,zigbee2mqtt:error 2020-04-29 12:53:45: Failed to start zigbee
,zigbee2mqtt:error 2020-04-29 12:53:45: Exiting...
,zigbee2mqtt:error 2020-04-29 12:53:45: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
,    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:71:23)
,    at Generator.throw (<anonymous>)
,    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
,npm ERR! code ELIFECYCLE
,npm ERR! errno 1
,npm ERR! zigbee2mqtt@1.12.2-dev start: `node index.js`
,npm ERR! Exit status 1
,npm ERR! 
,npm ERR! Failed at the zigbee2mqtt@1.12.2-dev start script.
,npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
,
,npm ERR! A complete log of this run can be found in:
,npm ERR!     /root/.npm/_logs/2020-04-29T04_53_46_064Z-debug.log
,Using '/app/data' as data directory
,
,> zigbee2mqtt@1.12.2-dev start /app
,> node index.js
,
,zigbee2mqtt:info  2020-04-29 12:54:46: Logging to console and directory: '/app/data/log/2020-04-29.12-54-46' filename: log.txt
,zigbee2mqtt:info  2020-04-29 12:54:47: Starting zigbee2mqtt version 1.12.2-dev (commit #ce62f43)
,zigbee2mqtt:info  2020-04-29 12:54:47: Starting zigbee-herdsman...
,zigbee2mqtt:error 2020-04-29 12:54:54: Error while starting zigbee-herdsman
,zigbee2mqtt:error 2020-04-29 12:54:54: Failed to start zigbee
,zigbee2mqtt:error 2020-04-29 12:54:54: Exiting...
,zigbee2mqtt:error 2020-04-29 12:54:54: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
,    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:71:23)
,    at Generator.throw (<anonymous>)
,    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
,npm ERR! code ELIFECYCLE
,npm ERR! errno 1
,npm ERR! zigbee2mqtt@1.12.2-dev start: `node index.js`
,npm ERR! Exit status 1
,npm ERR! 
,npm ERR! Failed at the zigbee2mqtt@1.12.2-dev start script.
,npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

Its a zigbee2mqtt docker installation . What did I do wrong?

daborbor, 2020-10-31:
I've been struggling for days trying to use the CC2530+CC2591 as coordinator, only to find out that the 'disable_led: true' makes Zigbee2Mqtt crash. Now it works. 

Just posting this here for future me. (Or anyone else who's struggling with this issue)