Koenkk/zigbee2mqtt #2445: Trouble binding or grouping IKEA Bulb and E1524 Remote

GregSutcliffe, 2019-11-30:
Hi, firstly great work, loving it :)

I have an IKEA LED1623G12 bulb and a E1524 remote. Neither binding nor grouping them appears to be working.

When I try to bind via friendly ID (eg `mosquitto_pub -h 127.0.0.1 -t 'zigbee2mqtt/bridge/bind/lounge_remote' -m 'lounge_bulb_1' `) then I see logs like:

```
debug 2019-11-30T23:48:04: binding cluster 'genOnOff' from 'lounge_remote' to 'lounge_bulb_1'
info  2019-11-30T23:48:10: Successfully bound cluster 'genOnOff' from 'lounge_remote' to 'lounge_bulb_1'
debug 2019-11-30T23:48:10: binding cluster 'genLevelCtrl' from 'lounge_remote' to 'lounge_bulb_1'
info  2019-11-30T23:48:13: Successfully bound cluster 'genLevelCtrl' from 'lounge_remote' to 'lounge_bulb_1'
```

However, clicking the toggle does not toggle the bulb.

I'm unsure how to check the firmware version on the E1524, so I decided to try the grouping method as per the [docs](https://www.zigbee2mqtt.io/devices/E1524_E1810.html). I see logs like:

```
debug 2019-11-30T23:48:41: Received Zigbee message from 'lounge_remote', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
```

So I added a block to my config:

```
groups:
  '0':
    friendly_name: group_lounge
    retain: false
    transition: 2
    optimistic: true
    devices:
      - '0x90fd9ffffe5a7d4b'
      - '0x000d6ffffeaf73e9'
```

However, now when I restart zigbee2mqtt, I see:

```
error 2019-11-30T23:33:40: Failed to call 'Groups' 'onZigbeeStarted' (Error: Timeout - 14060 - 1 - 2 - 4 - 0 after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (timers.js:327:15)
    at processTimers (timers.js:271:5))
```

And again, the toggle does not affect the bulb. What am I doing wrong? :)

# Reproduce
Try binding the devices and pressing the button, or try grouping with above config and then restart zigbee2mqtt.

## Debug Info
zigbee2mqtt version: zigbee2mqtt version 1.7.1 (commit #b459c35)
CC253X firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'

didiht, 2019-12-01:
It seems your remote has the current firmware which sends with groupId 0 after pairing.
In this case you should define a group with a groupId > 0, and then bind the remote to the new group.
Add the device to the new group and hopefully ...

GregSutcliffe, 2019-12-01:
> It seems your remote has the current firmware which sends with groupId 0 after pairing.
> In this case you should define a new group with a groupId > 0, and then bind the remote to the new group.

Thanks! So I changed my group ID to 100, i.e.
```
groups:
  '100':
    friendly_name: group_lounge
...
```

No other changes. I still see the same timeout on startup and the group still doesn't work.

If this is the latest firmware, shouldn't the binding be working? I'd prefer that anyway, as it's direct.

didiht, 2019-12-01:
Do you see in the log file that the remote sends now with groupId 100?

GregSutcliffe, 2019-12-01:
Hmm, no, it still has group ID 0. Does it need to be re-paired?

GregSutcliffe, 2019-12-01:
More interesting. If I publish to 'zigbee2mqtt/bridge/config/groups' I correctly get:

```
zigbee2mqtt/bridge/log {"type":"groups","message":[{"optimistic":true,"devices":["0x90fd9ffffe5a7d4b","0x000d6ffffeaf73e9"],"friendly_name":"group_lounge","retain":false,"transition":2,"ID":100}]}
```

But if I ask the remote for it's group membership via 'zigbee2mqtt/bridge/device/lounge_remote/get_group_membership', it's different:

```
zigbee2mqtt/bridge/device/lounge_remote/get_group_membership (null)
```

So yeah, I guess I'm still holding it wrong :)

didiht, 2019-12-01:
I don't think so. Note that EndDevices like the remote are in standby and don't receive any messages in this state.
Try to send the bind command and immediately press a button on the remote to wake it up.

GregSutcliffe, 2019-12-01:
Binding hasn't changed from my first post (I was pressing buttons then, but I forgot to say so, sorry). It claims successful binding, but nothing actually happens when I click the toggle. 

```
$ mosquitto_pub -t 'zigbee2mqtt/bridge/bind/lounge_remote' -m 'lounge_bulb_1'
debug 2019-12-01T08:11:09: binding cluster 'genOnOff' from 'lounge_remote' to 'lounge_bulb_1'
info  2019-12-01T08:11:10: Successfully bound cluster 'genOnOff' from 'lounge_remote' to 'lounge_bulb_1'
debug 2019-12-01T08:11:10: binding cluster 'genLevelCtrl' from 'lounge_remote' to 'lounge_bulb_1'
info  2019-12-01T08:11:11: Successfully bound cluster 'genLevelCtrl' from 'lounge_remote' to 'lounge_bulb_1'
```

So it seems like zigbee2mqtt thinks we're good - but the bulb doesn't actually change state when I click the remote button (although MQTT logs the click).

(BTW if having both groups and binding in one issue is a bad idea, I'm happy to split it...)

didiht, 2019-12-01:
Hm, what is the firmware version of the remote?
You can see this in file .../data/database.de, something with swBuildId

GregSutcliffe, 2019-12-01:
Looks like "swBuildId":"2.3.014" so I guess binding should work? How can I debug that further?

GregSutcliffe, 2019-12-01:
For the groups, it seems like the issue is definitely the remote - if I send '{"state": "ON"} to MQTT the bulb comes on. I note that even just trying to add the remote fails with a timeout (even while pressing buttons on it):

```
$ mosquitto_pub -t 'zigbee2mqtt/bridge/group/group_lounge/add' -m 'lounge_remote'
<click some buttons repeatedly>
debug 2019-12-01T10:57:25: Received MQTT message on 'zigbee2mqtt/bridge/group/group_lounge/add' with data 'lounge_remote'
<10s timeout>
error 2019-12-01T10:57:35: Failed to call 'Groups' 'onMQTTMessage' (Error: Timeout - 14060 - 1 - 10 - 4 - 0 after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (timers.js:327:15)
    at processTimers (timers.js:271:5))
```

Looks like it *really* doesn't want to update it's group membership ;)

didiht, 2019-12-01:
OK, it seems there is some misunderstanding.

Straight forward:

- re-pair the remote
- add a new group with id > 0
- bind the remote to the new group `zigbee2mqtt/bridge/bind/[remote] -m [new group]`
- check that the remote sends messages with new groupId in logfile
- add bulb to the new group `zigbee2mqtt/bridge/group/[GROUP_FRIENDLY_NAME]/add -m [bulb]`

GregSutcliffe, 2019-12-01:
a-HA, that's got it, thanks! GroupID is right, and it now controls the bulb even when zigbee2mqtt is down.

My confusion seems to have come from the [Binding docs](https://www.zigbee2mqtt.io/information/binding.html) which uses an example of binding a dimmer directly to a bulb. Binding it to a group seems to work just fine. I'll play with it some more, but this has me sorted for the short term. Thanks again!

closed 2019-12-01 by GregSutcliffe

dmak, 2023-04-30:
I am not sure how the following step is supposed to look like in log:
> check that the remote sends messages with new groupId in logfile

I my case I try to bind `remote_control` to group `65289` like this:
```
mosquitto_pub -t zigbee2mqtt/bridge/request/device/bind -m '{"from": "remote_control", "to": "65289"}
```
and it reports that `"status":"ok"`, but `remote_control` still continues to send the command to group `901`:
```
debug 2023-04-30 23:46:05: Received MQTT message on 'zigbee2mqtt/bridge/request/device/bind' with data '{"from": "remote_control", "to": "65289"}'
debug 2023-04-30 23:46:05: binding cluster 'genOnOff' from 'remote_control' to 'Lamp group'
debug 2023-04-30 23:46:12: Received Zigbee message from 'remote_control', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 901
info  2023-04-30 23:46:12: MQTT publish: topic 'zigbee2mqtt/remote_control', payload '{"action":"toggle","battery":60,"linkquality":102,"update":{"installed_version":587286065,"latest_version":587286065,"state":"idle"},"update_available":false}'
info  2023-04-30 23:46:12: MQTT publish: topic 'zigbee2mqtt/remote_control/action', payload 'toggle'
info  2023-04-30 23:46:13: Successfully bound cluster 'genOnOff' from 'remote_control' to 'Lamp group'
info  2023-04-30 23:46:13: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"cluster":"genOnOff","from":"remote_control","to":"Lamp group"},"type":"device_bind"}'
debug 2023-04-30 23:46:13: binding cluster 'genLevelCtrl' from 'remote_control' to 'Lamp group'
info  2023-04-30 23:46:16: Successfully bound cluster 'genLevelCtrl' from 'remote_control' to 'Lamp group'
info  2023-04-30 23:46:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"cluster":"genLevelCtrl","from":"remote_control","to":"Lamp group"},"type":"device_bind"}'
info  2023-04-30 23:46:16: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/bind', payload '{"data":{"clusters":["genOnOff","genLevelCtrl"],"failed":[],"from":"remote_control","to":"65289"},"status":"ok"}'
debug 2023-04-30 23:46:48: Received Zigbee message from 'remote_control', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 901
info  2023-04-30 23:46:48: MQTT publish: topic 'zigbee2mqtt/remote_control', payload '{"action":"toggle","battery":60,"linkquality":105,"update":{"installed_version":587286065,"latest_version":587286065,"state":"idle"},"update_available":false}'
info  2023-04-30 23:46:48: MQTT publish: topic 'zigbee2mqtt/remote_control/action', payload 'toggle'
```
Any idea what went wrong and how to workaround that?

Same issue with IKEA motion sensor (E1525): after successful binding it to the group `65289`, it sends events to wrong group `7090`:
```
debug 2023-04-30 23:29:40: Received Zigbee message from 'motion_sensor', type 'commandOnWithTimedOff', cluster 'genOnOff', data '{"ctrlbits":0,"offwaittime":0,"ontime":600}' from endpoint 1 with groupID 7090
info  2023-04-30 23:29:40: MQTT publish: topic 'zigbee2mqtt/motion_sensor', payload '{"battery":47,"illuminance_above_threshold":false,"linkquality":105,"occupancy":true,"requested_brightness_level":null,"requested_brightness_percent":null,"update":{"installed_version":304170354,"latest_version":304170354,"state":"idle"},"update_available":false}'
```

Z2M v1.29.2.