Koenkk/zigbee2mqtt #2446: Eurotronic Spirit SPZB0001 pairing problem

mrrioes, 2019-12-01:
Hi,

I´ve already read this issue #2292 but it didn´t helped me at all.
I already have paired 1 thermostat which works flawless. I now ordered 2 new ones and wanted to pair them but without any luck.
I´m on Version 1.7.1.

sjorge, 2019-12-01:
I got one yesterday from amazon that I could not get to pair unless I executed a reset. It had a new firmware form october 2019.

I updated the docs so it should show up once the site gets refresh, but meanwhile you can find the steps here: https://github.com/Koenkk/zigbee2mqtt.io/commit/a865ca184a141cde7969444748334e2c33d8935c

mrrioes, 2019-12-01:
I already tried that.
I managed to get one paired by restarting everything a bazillion times but could not reproduce it for the second one....

halfdom, 2019-12-01:
Thanks sjorge - work for me

mrrioes, 2019-12-11:
i´ve ordered a new one. that one works.

closed 2019-12-11 by mrrioes