Koenkk/zigbee2mqtt #2447: Issue with GLEDOPTO GL-S-007Z

grostim, 2019-12-01:
# Bug Report
## What happened

When trying to changewhite leve l, nothing happen.
on/off, brightness and colour adjustments are working well though.

Here is what i can find in the logs:

> zigbee2mqtt:error 2019-12-01 13:50:11: Publish 'set' 'white_value' to '0x00124b001ef9e738' failed: 'TypeError: Cannot read property 'state' of undefined'

I checked the fine print on the bulbs and although it is named GL-S-007Z, it is althought written that the whit temperature is adjustable.
I suspect the manufacturer has updated (upgraded) the specs without changing the modelID .... 👎 
Indeed, i have made a quick and dirty modification of devices.js in order to identify my GL-S-007 as  GL-S-007Z**S** (upgraded version with white temperature control + additional features not zigbee related).
And the bulbs are now working correctly and i can control the color temp .

I am not sure what would be the "clean" way to handle different equipments declaring themselves with the same modelID ?

## Debug Info
zigbee2mqtt version:

> zigbee2mqtt:info  2019-12-01 13:54:16: Starting zigbee2mqtt version 1.7.1+dev (commit #f767b41955335d9ed56f3f0310059b25f7d80b26)

CC253X firmware version: 

> zigbee2mqtt:info  2019-12-01 13:54:18: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'

sjorge, 2019-12-01:
I think the if block here needs updates to the condition to add the new model.

https://github.com/Koenkk/zigbee-herdsman-converters/blob/51a14aaa270df6b9ea7a6f2689675aabb024d9d7/converters/toZigbee.js#L912

milobt, 2019-12-01:
I have the Gledopto GL-S-007ZS bulbs and I am struggling to get them added to Hassio.
After creating a local devices.js file with the addition of a new device for GL-S-007ZS:

```
{
        zigbeeModel: ['GL-S-007ZS'],
        model: 'GL-S-007ZS',
        vendor: 'Gledopto',
        description: 'Smart RGB + CCT GU10',
        extend: gledopto.light,
        //meta: {options: {disableDefaultResponse: true}},
        supports: 'on/off, brightness, color, color temperature',
    },
```

They interview and pair successfully, but will not integrate into Hassio. I recently added another Gleopto LED controller (GL-C-007), which paired successfully and was added to MQTT integrations in Hassio, so I am puzzled as to why the GL-S-007ZS won’t integrate.

My Z2M configuration.yaml contain: "homeassistant": true

My Hassio MQTT configuration:

```
mqtt:
  broker: !secret device_ip
  discovery: true
  birth_message:
    topic: 'hass/status'
    payload: 'online'
  will_message:
    topic: 'hass/status'
    payload: 'offline'
```

I can turn the bulb on/off through MQTT set, so it is connected to the network.

Z2M version: 1.7.1

Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190223}}'

I have 16 bulbs to get connected and am stumped right now as how to proceed, so any help is greatly appreciated.

sjorge, 2019-12-01:
I think for hassio it also needs to be added here: https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/homeassistant.js#L433

milobt, 2019-12-01:
Is that a local file in Hassio dirs or on the coordinator?

sjorge, 2019-12-01:
It where zigbee2mqtt is running, I don't use hassio so I'm not sure how it works when using it as a plugin

PR #2616 (2019-12-27): Use correct Home Assistant configuration for GL-S-007Z
This PR changes the Gledopto `GL-S-007Z` to use color temperature instead, as this bulb supports adjusting color temperature and not just brightness.

I think this fixes #2447 (I don't think we shouldn't be using `white_value`, we should instead be using `color_temp` to control the white LEDs).
closed 2020-02-06 by stale[bot]