Koenkk/zigbee2mqtt #2448: Unable to pair tradfri bulbs with source routing firmware

sjorge, 2019-12-01:
# Bug Report
## What happened
I had some reach ability issues with my new Eurotronic SPZB0001, and occasionally with of my bulbs a bit further away from the coordinator. I had planned on switching to the source route firmware a while ago but was too lazy.

1. stop zigbee2mqtt
2. unplug the stick
3. update the firmware on the stick
4. plug in the stick
5. start zigbee2mqtt

At this stage the network was mostly unreachable... as the coordinator wanted to talk to my 3 bulbs + repeater directly (which is annoying as the bulbs are flacky, hence the reapter is between the coordinator and the bulbs)

6. removed the repeater from the network (left fine)(
7. pair the repeater again (paired fine)

Network was now more stable, I could reach 2 of my bulbs and all sensors were either talking to the bulbs or the repeater! Except one ikea tradfri RGB bulb would not send attReport or be reachable at all. The remote was able to turn it on and off just fine though, but the coordinator only saw the remote and not the bulb.

8. remove bulb (timeout)
9. move bulb next to coordinator
10. remove bulb (timeout)
11. enable join
12. reset bulb (6x on/off, bulb blinks)
13. bulb does not rejoin

I tried this about 10 times next to the coordinator and 3 times next to the repeater.
I then grabbed one of my spare tradfri 1000LM warm white bulbs... could not get it to join.
I tried another tradfri rgb bulb with older firmware... could not get it to join.

I also gave touchlink reset a go... but no effect. It seems very unlikely that all my tradfri bulbs went bad after switching to the source routing firmware.

## What did you expect to happen

The bulb to be able to join.

## How to reproduce it (minimal and precise)
Not sure ?

## Debug Info
zigbee2mqtt version: `1.7.1+dev (commit #694c45f)`
CC253X firmware version: `{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}}`

sjorge, 2019-12-01:
```
digraph G {
node[shape=record];
  "0x00124b001938a7e5" [style="bold, filled", fillcolor="#e04e5d", fontcolor="#ffffff", label="{Coordinator|0x00124b001938a7e5 (0)|2019-12-01T14:13:47+00:00}"];
  "0x00124b001938a7e5" -> "0x000d6ffffe8e8d4f" [penwidth=0.5, weight=0, color="#994444", label="88"]
  "0x00124b001938a7e5" -> "0x0017880104259333" [penwidth=0.5, weight=0, color="#994444", label="2"]
  "0x00158d0003f115c5" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{serverroom/sensor|0x00158d0003f115c5 (61107)|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|2019-12-01T14:08:25+00:00}"];
  "0x000d6ffffe8e8d4f" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{masterbedroom/repeater|0x000d6ffffe8e8d4f (17425)|IKEA TRADFRI signal repeater (E1746)|2019-12-01T14:13:08+00:00}"];
  "0x000d6ffffe8e8d4f" -> "0x00124b001938a7e5" [penwidth=0.5, weight=0, color="#994444", label="42"]
  "0x000d6ffffe8e8d4f" -> "0x0017880104259333" [penwidth=0.5, weight=0, color="#994444", label="155"]
  "0x00158d00033ddfaa" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/sensor|0x00158d00033ddfaa (46734)|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|unknown}"];
  "0x00158d00033ddfaa" -> "0x0017880104259333" [penwidth=1, weight=0, color="#994444", label="252"]
  "0x0017880104259333" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{bedroom/desk_lamp/bulb_1|0x0017880104259333 (7034)|Philips Hue white and color ambiance E26/E27/E14 (9290012573A)|2019-12-01T14:11:51+00:00}"];
  "0x0017880104259333" -> "0x00124b001938a7e5" [penwidth=0.5, weight=0, color="#994444", label="0"]
  "0x0017880104259333" -> "0x000d6ffffe8e8d4f" [penwidth=0.5, weight=0, color="#994444", label="111"]
  "0x00158d0004148f89" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/motion|0x00158d0004148f89 (53667)|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|2019-12-01T14:11:14+00:00}"];
  "0x00158d0004148f89" -> "0x0017880104259333" [penwidth=1, weight=0, color="#994444", label="227"]
  "0x14b457fffecc1315" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/desk_lamp/remote|0x14b457fffecc1315 (45663)|IKEA TRADFRI ON/OFF switch (E1743)|unknown}"];
  "0x14b457fffecc1315" -> "0x0017880104259333" [penwidth=1, weight=0, color="#994444", label="220"]
  "0x14b457fffeca351b" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/night_light/remote|0x14b457fffeca351b (10575)|IKEA TRADFRI ON/OFF switch (E1743)|unknown}"];
  "0x14b457fffeca351b" -> "0x000d6ffffe8e8d4f" [penwidth=1, weight=0, color="#994444", label="0"]
  "0x00158d0001ffaffc" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/radiator|0x00158d0001ffaffc (38868)|Eurotronic Spirit Zigbee wireless heater thermostat (SPZB0001)|2019-12-01T14:10:20+00:00}"];
  "0x00158d0001ffaffc" -> "0x000d6ffffe8e8d4f" [penwidth=1, weight=0, color="#994444", label="232"]
}
```
Note: my other bulbs are missing too, only my hue one still remains as I tried to see if I could remove and rejoin them also. They do not :(

sjorge, 2019-12-01:
I'm considering switching back to the default firmware, but I am not sure that will solve the problem... also my sensors seem more stable now on the source route firmware.

sjorge, 2019-12-01:
I just tried my innr bulb next the coordinator, no effect although the bulb itself seems to think it joined.

sjorge, 2019-12-01:
I will create a new network and rejoin everything... or at least try.

sjorge, 2019-12-01:
Interesting... I started rebuilding the network with a new panid and network key... eveyrthing was going well... until I tried to pair a 6th device! Now it doesn't show up at all.

So I'm guessing there is some issue here as the coordinator can only have 5 directly connected devices! 

```
digraph G {
node[shape=record];
  "0x00124b001938a7e5" [style="bold, filled", fillcolor="#e04e5d", fontcolor="#ffffff", label="{Coordinator|0x00124b001938a7e5 (0)|2019-12-01T15:44:53+00:00}"];
  "0x00124b001938a7e5" -> "0x000d6ffffe8e8d4f" [penwidth=0.5, weight=0, color="#994444", label="81"]
  "0x00124b001938a7e5" -> "0x0017880104259333" [penwidth=0.5, weight=0, color="#994444", label="7"]
  "0x000d6ffffe8e8d4f" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{masterbedroom/repeater|0x000d6ffffe8e8d4f (16396)|IKEA TRADFRI signal repeater (E1746)|2019-12-01T15:43:59+00:00}"];
  "0x000d6ffffe8e8d4f" -> "0x00124b001938a7e5" [penwidth=0.5, weight=0, color="#994444", label="38"]
  "0x000d6ffffe8e8d4f" -> "0x90fd9ffffee77fcf" [penwidth=0.5, weight=0, color="#994444", label="99"]
  "0x000d6ffffe8e8d4f" -> "0x0017880104259333" [penwidth=0.5, weight=0, color="#994444", label="54"]
  "0x90fd9ffffee77fcf" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{bedroom/night_light/bulb|0x90fd9ffffee77fcf (36035)|IKEA TRADFRI LED bulb E14/E26/E27 600 lumen, dimmable, color, opal white (LED1624G9)|2019-12-01T15:44:39+00:00}"];
  "0x90fd9ffffee77fcf" -> "0x00124b001938a7e5" [penwidth=0.5, weight=0, color="#994444", label="1"]
  "0x90fd9ffffee77fcf" -> "0x000d6ffffe8e8d4f" [penwidth=0.5, weight=0, color="#994444", label="107"]
  "0x90fd9ffffee77fcf" -> "0x0017880104259333" [penwidth=0.5, weight=0, color="#994444", label="108"]
  "0x0017880104259333" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{bedroom/desk_lamp/bulb|0x0017880104259333 (21488)|Philips Hue white and color ambiance E26/E27/E14 (9290012573A)|2019-12-01T15:43:24+00:00}"];
  "0x0017880104259333" -> "0x00124b001938a7e5" [penwidth=0.5, weight=0, color="#994444", label="1"]
  "0x0017880104259333" -> "0x000d6ffffe8e8d4f" [penwidth=0.5, weight=0, color="#994444", label="71"]
  "0x0017880104259333" -> "0x90fd9ffffee77fcf" [penwidth=0.5, weight=0, color="#994444", label="80"]
  "0x14b457fffecc1315" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/night_light/remote|0x14b457fffecc1315 (48398)|IKEA TRADFRI ON/OFF switch (E1743)|2019-12-01T15:29:07+00:00}"];
  "0x14b457fffecc1315" -> "0x90fd9ffffee77fcf" [penwidth=1, weight=0, color="#994444", label="255"]
  "0x14b457fffeca351b" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/desk_lamp/remote|0x14b457fffeca351b (497)|IKEA TRADFRI ON/OFF switch (E1743)|2019-12-01T15:31:24+00:00}"];
  "0x14b457fffeca351b" -> "0x0017880104259333" [penwidth=1, weight=0, color="#994444", label="228"]
}
```

Here is another network map... I tried both adding another bulb, or some aqara end devices... nothing works at this point.

sjorge, 2019-12-01:
I'm giving up at this point, nothing works... I will reflash the default firmware and repair everything... hopefully that will work.

I'll love out the Eurotronic TRV as that was causing so many timeouts and lockups.

sjorge, 2019-12-01:
So to summerize:
- no issues on DEFAULT
- using SOURCE ROUTE, cannot add more than 5 devices... even if the first 5 are all routers.

iFloris, 2019-12-23:
@sjorge I ran into the same problem, was entirely confused why bulbs that I wanted to re-pair were not showing up at all. I saw your post here a few days ago, re-flashed to the default firmware and used the new touchlink feature to repair the bulbs. Using the default firmware, this worked as expected. I am having some issues with the network now using default. I think this is because I have 51 devices right now. But I am glad that the bulbs are paring again, so thanks for pointing this out. @Koenkk would you happen to know what might be the cause that source routing firmware stops seeing bulbs like this?

sjorge, 2019-12-23:
I think I lost thje original network with the default -> source firmware because I checked the box to clear NVRAM during flashing.

That still does not explain why the source route one stops accepting devices after 4.

iFloris, 2019-12-23:
@sjorge I have switched between firmware versions quite often, from default to max_stability and a few months ago to source_routing. I always kept the network config so that I could avoid repairing all devices. The problem being unable to add new devices using source_routing only appeared recently, which does not mean that this was not present before as I tend to add devices in batches.

iFloris, 2020-01-03:
Reporting back, after a few days I needed to return to the source routing firmware. Many devices were beginning to either disappear entirely (not responding) or started to respond very slowly (such as lamp turning on/off 10x after a delay of several seconds). As the WAF was nearing freezing temperatures I needed to return the Zigbee2mqtt network to a working state. Since returning to source routing, things have mostly become stable again. However, a lamp just dropped from the zigbee2mqtt network so I fear I will have to re-pair that. But in order to do that, I first have to flash back to the regular firmware, then re-pair, and then flash back to source-routing. Not a very exciting prospect..

closed 2020-03-11 by stale[bot]

caffeineflo, 2020-06-02:
I have the very same issue with the source routing firmware, I can't add another device to the network. Luckily I have two different CC2531, one with the default firmware and one with the source routing firmware, so in this "unfortunate" situation I simply have to switch the USB sticks

ChipWolf, 2020-11-20:
I have this exact issue with source routing @Koenkk

ChipWolf, 2020-11-20:
If anyone stumbles across this ticket as I did, the answer to why this happens is here: https://github.com/Koenkk/zigbee2mqtt/issues/5015#issuecomment-731237434