Koenkk/zigbee2mqtt #2450: Osram RGB Bulbs Error - "Cannot read property 'color' of null'" when setting color via hue and saturation

odx, 2019-12-01:
# Bug Report
## What happened

I just installed a few Osram bulbs in my garden. I attempt to use hue and saturation to set the color (since this is what HomeKit requires, als all conversions I implemented to far are never 100% reversible).

When is send data to the bulbs (They identify as CLA60 RGBW Z3, PAR 16 50 RGBW - LIGHTIFY or PAR16 RGBW Z3) I don't get proper status updates.

Neither the color values, nor the color_mode property seem to update in the MQTT messages correctly although the bulb's color reflects the given values. Also I noticed that there is an error in the log, although the color of the bulb updates correctly:

```
zigbee2mqtt:error 2019-12-01T20:46:23: Publish 'set' 'color' to 'my_fancy_bulb_name' failed: 'TypeError: Cannot read property 'color' of null'
```

I can set the correct color or color temperature with MQTT messages like:
```
{"color":{"hue":360,"saturation":100}}
{"color":{"hue":180,"saturation":100}}
{"color_temp":370}
```

When the error occurs I don't get a MQTT response message.

## What did you expect to happen

When I use hue and saturation as color parameters, I expect the bulb to report hue and saturation and the color_mode to represent value 0.
When I use color_temp to set the "color", I expect the color_mode to represent value 2 but it does not change.

## How to reproduce it (minimal and precise)
Publish:
```
{"state":"ON"}
```
Response:
```
{"state":"ON","linkquality":47,"last_seen":"2019-12-01T20:36:18.826Z","brightness":127,"color":{"x":0.6223,"y":0.3656,"saturation":100,"hue":357.2},"color_temp":370,"color_mode":1,"elapsed":20808}
```

Publish:
```
{"color":{"hue":360,"saturation":100}}
```
Error in the log:
```
zigbee2mqtt:error 2019-12-01T20:46:23: Publish 'set' 'color' to 'my_fancy_bulb_name' failed: 'TypeError: Cannot read property 'color' of null'
```
No Response received but the bulbs color is red as expected. In the following response messages the hue value is not correct (but saturation is). The color mode does not change to 0.

Publish:
```
{"color_temp":370}
```
Response:
```
{"state":"ON","linkquality":47,"last_seen":"2019-12-01T20:54:05.571Z","brightness":1,"color":{"x":0.6223,"y":0.3656,"saturation":50,"hue":357.2},"color_temp":370,"color_mode":1,"elapsed":63013}
```
And the bulbs color is warm white. But color mode did not change to 2.

## Debug Info
zigbee2mqtt version: 1.7.1
CC2530 firmware version: 20190619 ZStack Home 1.2

odx, 2019-12-02:
I captured a debug log. I am not really sure how to read it though...
```
adable" +30ms
zigbee2mqtt:debug 2019-12-02T20:15:33: Received MQTT message on 'zigbee/my_fancy_friendly_name/set' with data '{"color":{"hue":360,"saturation":100}}'
winston:file written true false
zigbee2mqtt:debug 2019-12-02T20:15:33: Publishing 'set' 'color' to 'my_fancy_friendly_name'
winston:file written true false
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":39095,"destendpoint":1,"srcendpoint":1,"clusterid":768,"transid":36,"options":0,"radius":30,"len":8,"data":{"type":"Buffer","data":[1,14,67,255,255,254,0,0]}} +4s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,18,36,1,183,152,1,1,0,3,36,0,30,8,1,14,67,255,255,254,0,0,155] +4s
  serialport/stream _write 23 bytes of data +4s
  serialport/binding-abstract write 23 bytes +4s
winston:file logged 37618 debug 2019-12-02T20:15:33: Received MQTT message on 'zigbee/my_fancy_friendly_name/set' with data '{"color":{"hue":360,"saturation":100}}'

winston:file logged 37699 debug 2019-12-02T20:15:33: Publishing 'set' 'color' to 'my_fancy_friendly_name'

  serialport/bindings/unixWrite Starting write 23 bytes offset 0 bytesToWrite 23 +4s
  serialport/bindings/unixWrite write returned: wrote 23 bytes +2ms
  serialport/bindings/unixWrite Finished writing 23 bytes +0ms
  serialport/stream binding.write write finished +5ms
  serialport/bindings/poller received "readable" +4s
  serialport/binding-abstract read +20ms
  serialport/bindings/unixRead Starting read +4s
  serialport/bindings/unixRead Finished read { bytesRead: 6,
  buffer:
   <Buffer fe 0e 61 02 02 00 02 06 03 9b 15 34 01 02 00 00 00 00 d1 fe 03 61 08 00 01 55 3e fe 03 61 08 00 01 55 3e fe 06 61 08 00 04 00 08 00 00 63 fe 03 61 08 ... > } bytes +1ms
  serialport/stream binding.read finished { bytesRead: 6 } +18ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100] +4s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [] - 100 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +4s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  serialport/stream _read reading { start: 2405, toRead: 63131 } +5ms
  serialport/binding-abstract read +7ms
  serialport/bindings/unixRead Starting read +7ms
  serialport/bindings/unixRead Finished read { bytesRead: 8,
  buffer:
   <Buffer fe 0e 61 02 02 00 02 06 03 9b 15 34 01 02 00 00 00 00 d1 fe 03 61 08 00 01 55 3e fe 03 61 08 00 01 55 3e fe 06 61 08 00 04 00 08 00 00 63 fe 03 61 08 ... > } bytes +1ms
  serialport/stream binding.read finished { bytesRead: 8 } +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,36,226] +4ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,36,226] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [$] - 226 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":36} +4s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  serialport/stream _read reading { start: 2413, toRead: 63123 } +4ms
  serialport/binding-abstract read +7ms
  serialport/bindings/unixRead Starting read +5ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +1ms
  serialport/bindings/poller Polling for "readable" +17ms
  serialport/bindings/poller received "readable" +18ms
  serialport/binding-abstract read +21ms
  serialport/bindings/unixRead Starting read +20ms
  serialport/bindings/unixRead Finished read { bytesRead: 10,
  buffer:
   <Buffer fe 0e 61 02 02 00 02 06 03 9b 15 34 01 02 00 00 00 00 d1 fe 03 61 08 00 01 55 3e fe 03 61 08 00 01 55 3e fe 06 61 08 00 04 00 08 00 00 63 fe 03 61 08 ... > } bytes +1ms
  serialport/stream binding.read finished { bytesRead: 10 } +23ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,183,152,1,128,141,167] +25ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,183,152,1,128,141,167] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [����] - 167 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":39095,"relaycount":1,"relaylist":[36224]} +27ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  serialport/stream _read reading { start: 2423, toRead: 63113 } +4ms
  serialport/binding-abstract read +6ms
  serialport/bindings/unixRead Starting read +5ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +1ms
  serialport/bindings/poller Polling for "readable" +9ms
  serialport/bindings/poller received "readable" +45ms
  serialport/binding-abstract read +48ms
  serialport/bindings/unixRead Starting read +49ms
  serialport/bindings/unixRead Finished read { bytesRead: 30,
  buffer:
   <Buffer fe 0e 61 02 02 00 02 06 03 9b 15 34 01 02 00 00 00 00 d1 fe 03 61 08 00 01 55 3e fe 03 61 08 00 01 55 3e fe 06 61 08 00 04 00 08 00 00 63 fe 03 61 08 ... > } bytes +1ms
  serialport/stream binding.read finished { bytesRead: 30 } +53ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,25,68,129,0,0,0,3,183,152,1,1,0,52,0,128,110,64,0,0,5,8,14,11,67,0,128,141,28,48] +56ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,25,68,129,0,0,0,3,183,152,1,1,0,52,0,128,110,64,0,0,5,8,14,11,67,0,128,141,28,48] +1ms
[... cut ...]
  serialport/binding-abstract read +5ms
winston:file logged 37835 error 2019-12-02T20:15:33: Publish 'set' 'color' to 'my_fancy_friendly_name' failed: 'TypeError: Cannot read property 'color' of null'

winston:file logged 38117 info  2019-12-02T20:15:33: MQTT publish: topic 'zigbee/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'color' to 'my_fancy_friendly_name' failed: 'TypeError: Cannot read property 'color' of null'","meta":{"friendly_name":"my_fancy_friendly_name"}}'
```

Koenkk, 2019-12-03:
Can you provide your `configuration.yaml`? (without password/usernames and ip addresses).

odx, 2019-12-03:
```
homeassistant: false
permit_join: false
advanced:
  network_key:
    - XXX
  last_seen: ISO_8601
  elapsed: true
  report: true
mqtt:
  base_topic: zigbee
  server: 'mqtt://localhost'
  user: XXX
  password: XXX
serial:
  port: /dev/ttyUSB0
devices:
  '0x00158d000273cfdc':
    qos: 2
    friendly_name: temp2
    retain: true
  '0x00158d000273e222':
    qos: 2
    friendly_name: door1
    retain: true
  '0x00158d0002d481c3':
    qos: 2
    friendly_name: terrassentuer_rechts
    retain: true
  '0x00158d0002d48263':
    qos: 2
    friendly_name: terrassentuer_links
    retain: true
  '0x00158d000273e216':
    qos: 2
    friendly_name: terrassentuer_kueche
    retain: true
  '0x00158d0002d489fb':
    qos: 2
    friendly_name: wohnzimmerfenster_links
    retain: true
  '0x00158d0002e20b98':
    qos: 2
    friendly_name: wohnzimmerfenster_rechts
    retain: true
  '0x00158d0002d489e2':
    qos: 2
    friendly_name: wc-fenster
    retain: true
  '0x00158d0002d48259':
    qos: 2
    friendly_name: fenster_abstellraum
    retain: true
  '0x00158d0002e25588':
    qos: 2
    friendly_name: badfenster
    retain: true
  '0x00158d0002ca2d76':
    qos: 2
    friendly_name: dachfenster_1
    retain: true
  '0x00158d0002d48174':
    qos: 2
    friendly_name: garagentor
    retain: true
  '0x00158d0002d481ed':
    qos: 2
    friendly_name: fenster_2_links
    retain: true
  '0x00158d0002d48177':
    qos: 2
    friendly_name: fenster_2_rechts
    retain: true
  '0x00158d0002e20d06':
    qos: 2
    friendly_name: dachfenster_galerie
    retain: true
  '0x00158d0002d481e9':
    qos: 2
    friendly_name: giebelfenster_links
    retain: true
  '0x00158d0002d48919':
    qos: 2
    friendly_name: giebelfenster_rechts
    retain: true
  '0x00158d000273e1a3':
    qos: 2
    friendly_name: balkontuer
    retain: true
  '0x00158d0002d481b3':
    qos: 2
    friendly_name: garage_gartentuer
    retain: true
  '0x00124b001bb8e9e9':
    qos: 2
    friendly_name: router3
    retain: true
  '0x000b57fffe900c86':
    qos: 2
    friendly_name: nachttisch_janina
    retain: true
  '0x000b57fffe976159':
    qos: 2
    friendly_name: nachttisch_ronja
    retain: true
  '0x000d6ffffe24cb50':
    qos: 2
    friendly_name: sofalampe
    retain: true
  '0xd0cf5efffef808ed':
    qos: 2
    friendly_name: RemoteSZ
    retain: true
  '0x00158d0003042554':
    qos: 2
    friendly_name: pir_wz
    retain: true
    occupancy_timeout: 300
  '0x00158d000305875c':
    qos: 2
    friendly_name: pir_bad
    retain: true
    occupancy_timeout: 300
  '0x00158d00031b315d':
    qos: 2
    friendly_name: spare_broken
    retain: true
  '0x000d6ffffe7aca4b':
    qos: 2
    friendly_name: tradfri_repeater
    retain: true
  '0x00158d0002758987':
    qos: 2
    friendly_name: smoke_broken
    retain: true
  '0x00158d0003021a63':
    qos: 2
    friendly_name: pir_abstell
    retain: true
  '0x7cb03eaa0a068a51':
    qos: 2
    friendly_name: plug1
    retain: true
  '0x00158d0002758682':
    qos: 2
    friendly_name: smoke_wz
    retain: true
  '0x00158d0002759edc':
    qos: 2
    friendly_name: smoke_flur_eg
    retain: true
  '0x00158d000275878d':
    qos: 2
    friendly_name: smoke_flur_og
    retain: true
  '0x00158d0002754836':
    qos: 2
    friendly_name: smoke_sz
    retain: true
  '0x00158d000275472a':
    qos: 2
    friendly_name: smoke_1
    retain: true
  '0x00158d0002758628':
    qos: 2
    friendly_name: smoke_galerie
    retain: true
  '0x00158d0002755923':
    qos: 2
    friendly_name: smoke_2
    retain: true
  '0x7cb03eaa0a06a960':
    qos: 2
    friendly_name: cd_player_2
    retain: true
  '0x00158d000349cb43':
    qos: 2
    friendly_name: temp_sz
    retain: true
  '0x7cb03eaa0a071b63':
    qos: 2
    friendly_name: plug2
    retain: true
  '0x00158d00031b28b5':
    qos: 2
    friendly_name: pir_sz
    retain: true
  '0x00158d00027315e8':
    qos: 2
    friendly_name: temp1
    retain: true
  '0x00158d0002a367df':
    friendly_name: lumicube2
    retain: false
  '0x00158d0002c6371f':
    friendly_name: pir_kueche
    retain: false
  '0x00158d0002b6ee13':
    friendly_name: pir_flur_og
    retain: false
  '0xf0d1b80000129fd1':
    friendly_name: garten_spot_1
    retain: true
  '0x00158d00027305c3':
    friendly_name: pir_flur_eg
    retain: false
  '0x70b3d5de0000140c':
    friendly_name: ecozy1
    retain: false
  '0xf0d1b8000010d9a1':
    friendly_name: garten_kugel_gross
    retain: true
  '0xf0d1b8000010b717':
    friendly_name: garten_kugel_klein_sandkasten
    retain: true
  '0x7cb03eaa00aed8fa':
    friendly_name: garten_spot_2
    retain: true
  '0x7cb03eaa00ab61bc':
    friendly_name: garten_spot_3
    retain: true
  '0x7cb03eaa00aee03d':
    friendly_name: garten_spot_4
    retain: true
  '0x7cb03eaa00aee1ca':
    friendly_name: garten_spot_5
    retain: true
  '0xf0d1b8000010af88':
    friendly_name: garten_kugel_klein_baum
    retain: true
```

Koenkk, 2019-12-04:
I don't know where the error comes from yet. Therefore I've added extra logging in the latest dev branch. Please:

- Update to the latest dev branch
- Enable debug logging
- Provide the log again

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

sjorge, 2019-12-04:
@Koenkk I see the hue value is coming form the 'currentHue' attribute, I fixed those for the Innr bulbs because they were not being read a few weeks ago.

I however did not do the toZigbee for it because I never figured out how to decide between enhancedMoveToHueAndSaturation, enhancedMoveToHue, moveToHueAndSaturation, or moveToHue... I know the Innr bulb does not respond with a enhancedMoveToHueAndSaturation or enhancedMoveToHue, so it might be the same for Osram?

odx, 2019-12-04:
Here is the requested log. I am not sure if the first timeout is related to this at all or this is some other issue with my network. But I checked this with two different Osram Models with both the same "Cannot read property" error...

```
Dec 04 21:07:00 raspberrypi npm[462]: zigbee2mqtt:debug 2019-12-04 21:07:00: Received MQTT message on 'zigbee/garten_spot_5/set' with data '{"color":{"hue":250, "saturation":50}}'
Dec 04 21:07:00 raspberrypi npm[462]: zigbee2mqtt:debug 2019-12-04 21:07:00: Publishing 'set' 'color' to 'garten_spot_5'
Dec 04 21:07:15 raspberrypi npm[462]: zigbee2mqtt:error 2019-12-04 21:07:15: Publish 'set' 'color' to 'garten_spot_5' failed: 'Error: Timeout - 24803 - 3 - 5 - 768 - 11 after 15000ms'
Dec 04 21:07:15 raspberrypi npm[462]: zigbee2mqtt:debug 2019-12-04 21:07:15: Error: Timeout - 24803 - 3 - 5 - 768 - 11 after 15000ms
Dec 04 21:07:15 raspberrypi npm[462]:     at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
Dec 04 21:07:15 raspberrypi npm[462]:     at ontimeout (timers.js:436:11)
Dec 04 21:07:15 raspberrypi npm[462]:     at tryOnTimeout (timers.js:300:5)
Dec 04 21:07:15 raspberrypi npm[462]:     at listOnTimeout (timers.js:263:5)
Dec 04 21:07:15 raspberrypi npm[462]:     at Timer.processTimers (timers.js:223:10)
Dec 04 21:07:15 raspberrypi npm[462]: zigbee2mqtt:info  2019-12-04 21:07:15: MQTT publish: topic 'zigbee/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'color' to 'garten_spot_5' failed: 'Error: Timeout - 24803 - 3 - 5 - 768 - 11 after 15000ms'","meta":{"friendly_name":"garten_spot_5"}}'

...

Dec 04 21:07:29 raspberrypi npm[462]: zigbee2mqtt:debug 2019-12-04 21:07:29: Received MQTT message on 'zigbee/garten_spot_5/set' with data '{"color":{"hue":350, "saturation":50}}'
Dec 04 21:07:29 raspberrypi npm[462]: zigbee2mqtt:debug 2019-12-04 21:07:29: Publishing 'set' 'color' to 'garten_spot_5'
Dec 04 21:07:30 raspberrypi npm[462]: zigbee2mqtt:error 2019-12-04 21:07:30: Publish 'set' 'color' to 'garten_spot_5' failed: 'TypeError: Cannot read property 'color' of null'
Dec 04 21:07:30 raspberrypi npm[462]: zigbee2mqtt:debug 2019-12-04 21:07:30: TypeError: Cannot read property 'color' of null
Dec 04 21:07:30 raspberrypi npm[462]:     at Object.convertSet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:429:73)
Dec 04 21:07:30 raspberrypi npm[462]: zigbee2mqtt:info  2019-12-04 21:07:30: MQTT publish: topic 'zigbee/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'color' to 'garten_spot_5' failed: 'TypeError: Cannot read property 'color' of null'","meta":{"friendly_name":"garten_spot_5"}}'
```

```
Dec 04 21:09:14 raspberrypi npm[462]: zigbee2mqtt:debug 2019-12-04 21:09:14: Received MQTT message on 'zigbee/garten_kugel_klein_baum/set' with data '{"color":{"hue":250, "saturation":50}}'
Dec 04 21:09:14 raspberrypi npm[462]: zigbee2mqtt:debug 2019-12-04 21:09:14: Publishing 'set' 'color' to 'garten_kugel_klein_baum'
Dec 04 21:09:14 raspberrypi npm[462]: zigbee2mqtt:debug 2019-12-04 21:09:14: Received Zigbee message from 'garten_kugel_klein_baum', type 'attributeReport', cluster 'lightingColorCtrl', data '{"currentX":36936,"currentY":25251,"colorTemperature":370}' from endpoint 1 with groupID 0
Dec 04 21:09:14 raspberrypi npm[462]: zigbee2mqtt:info  2019-12-04 21:09:14: MQTT publish: topic 'zigbee/garten_kugel_klein_baum', payload '{"state":"ON","linkquality":47,"last_seen":"2019-12-04T21:09:14.161Z","brightness":50,"color":{"x":0.5636,"y":0.3853,"saturation":50,"hue":351},"color_temp":370,"color_mode":1,"elapsed":15939}'
Dec 04 21:09:14 raspberrypi npm[462]: zigbee2mqtt:error 2019-12-04 21:09:14: Publish 'set' 'color' to 'garten_kugel_klein_baum' failed: 'TypeError: Cannot read property 'color' of null'
Dec 04 21:09:14 raspberrypi npm[462]: zigbee2mqtt:debug 2019-12-04 21:09:14: TypeError: Cannot read property 'color' of null
Dec 04 21:09:14 raspberrypi npm[462]:     at Object.convertSet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:429:73)
Dec 04 21:09:14 raspberrypi npm[462]: zigbee2mqtt:info  2019-12-04 21:09:14: MQTT publish: topic 'zigbee/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'color' to 'garten_kugel_klein_baum' failed: 'TypeError: Cannot read property 'color' of null'","meta":{"friendly_name":"garten_kugel_klein_baum"}}'
```

Koenkk, 2019-12-05:
Should be fixed in latest dev.

odx, 2019-12-05:
Thanks @Koenkk. I have updated the branch. Although the error is gone, the behavior is not really different. Besides the fact that after a while I get the correct hue / saturation values delivered. (May be due to reporting that I enabled recently). Anyways there is still a "UnhandledPromiseRejectionWarning".

Here a log from a request, where the hue / saturation data in response is ok:
```
Dec 05 20:51:11 raspberrypi npm[10027]: zigbee2mqtt:debug 2019-12-05 20:51:11: Received MQTT message on 'zigbee/garten_kugel_klein_baum/set' with data '{"color":{"hue":360,"saturation":100}}'
Dec 05 20:51:11 raspberrypi npm[10027]: zigbee2mqtt:debug 2019-12-05 20:51:11: Publishing 'set' 'color' to 'garten_kugel_klein_baum'
Dec 05 20:51:11 raspberrypi npm[10027]: (node:10042) UnhandledPromiseRejectionWarning: TypeError: Cannot set property 'last_seen' of null
Dec 05 20:51:11 raspberrypi npm[10027]:     at Controller.publishEntityState (/opt/zigbee2mqtt/lib/controller.js:248:31)
Dec 05 20:51:11 raspberrypi npm[10027]:     at EntityPublish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/entityPublish.js:178:30)
Dec 05 20:51:11 raspberrypi npm[10027]: (node:10042) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 8)
Dec 05 20:51:11 raspberrypi npm[10027]: zigbee2mqtt:debug 2019-12-05 20:51:11: Received Zigbee message from 'garten_kugel_klein_baum', type 'attributeReport', cluster 'lightingColorCtrl', data '{"currentHue":254,"currentSaturation":254,"currentX":36936,"currentY":25251,"colorTemperature":370}' from endpoint 1 with groupID 0
Dec 05 20:51:11 raspberrypi npm[10027]: zigbee2mqtt:info  2019-12-05 20:51:11: MQTT publish: topic 'zigbee/garten_kugel_klein_baum', payload '{"state":"ON","linkquality":47,"last_seen":"2019-12-05T20:51:11.605Z","brightness":50,"color":{"x":0.5636,"y":0.3853,"saturation":100,"hue":360},"color_temp":370,"color_mode":1,"elapsed":77023}'
```

But I still often get wrong data:
```
Not ok

Dec 05 20:53:09 raspberrypi npm[10027]: zigbee2mqtt:debug 2019-12-05 20:53:09: Received MQTT message on 'zigbee/garten_kugel_klein_baum/set' with data '{"color":{"hue":120,"saturation":100}}'
Dec 05 20:53:09 raspberrypi npm[10027]: zigbee2mqtt:debug 2019-12-05 20:53:09: Publishing 'set' 'color' to 'garten_kugel_klein_baum'
Dec 05 20:53:09 raspberrypi npm[10027]: (node:10042) UnhandledPromiseRejectionWarning: TypeError: Cannot set property 'last_seen' of null
Dec 05 20:53:09 raspberrypi npm[10027]:     at Controller.publishEntityState (/opt/zigbee2mqtt/lib/controller.js:248:31)
Dec 05 20:53:09 raspberrypi npm[10027]:     at EntityPublish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/entityPublish.js:178:30)
Dec 05 20:53:09 raspberrypi npm[10027]: (node:10042) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 11)
Dec 05 20:53:09 raspberrypi npm[10027]: zigbee2mqtt:debug 2019-12-05 20:53:09: Received Zigbee message from 'garten_kugel_klein_baum', type 'attributeReport', cluster 'lightingColorCtrl', data '{"currentX":36936,"currentY":25251,"colorTemperature":370}' from endpoint 1 with groupID 0
Dec 05 20:53:09 raspberrypi npm[10027]: zigbee2mqtt:info  2019-12-05 20:53:09: MQTT publish: topic 'zigbee/garten_kugel_klein_baum', payload '{"state":"ON","linkquality":49,"last_seen":"2019-12-05T20:53:09.819Z","brightness":50,"color":{"x":0.5636,"y":0.3853,"saturation":100,"hue":360},"color_temp":370,"color_mode":1,"elapsed":7950}'

...

Dec 05 20:53:35 raspberrypi npm[10027]: zigbee2mqtt:debug 2019-12-05 20:53:35: Received Zigbee message from 'garten_kugel_klein_baum', type 'attributeReport', cluster 'lightingColorCtrl', data '{"currentHue":85,"currentSaturation":254}' from endpoint 1 with groupID 0
Dec 05 20:53:35 raspberrypi npm[10027]: zigbee2mqtt:info  2019-12-05 20:53:35: MQTT publish: topic 'zigbee/garten_kugel_klein_baum', payload '{"state":"ON","linkquality":47,"last_seen":"2019-12-05T20:53:35.615Z","brightness":50,"color":{"x":0.5636,"y":0.3853,"saturation":100,"hue":120},"color_temp":370,"color_mode":1,"elapsed":25798}'
```

Besides the fact that the responses still don't reliably reflect the hue / saturation values. What I also need is the correct color_mode. Since all values are transmitted through the same MQTT topic, I need to be aware about which of the values are currently really up to date. My pattern would be:

color_mode 0: process color -> hue / saturation
color_mode 2: process color_temp
any color mode: process brightness

Any I idea why the color mode never changes? Some of my bulbs are stuck in color_mode 2 some in color_mode 1...

Koenkk, 2019-12-07:
The exception should also be fixed.

The other stuff is harder to solve. In order to immediately have the new x/y reported after a color change via hue/saturation we need to find an algorithm which can do this. After that we have to set the `newState` here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/toZigbee.js#L386

sjorge, 2019-12-07:
I have an HS(V) to RGB and RGB -> XY converters in node-red that seems to work...

```javascript
function hsv2rgb (h, s, v = 100) {
    var RGB_MAX = 255;
    var HUE_MAX = 360;
    var SV_MAX = 100;
    
    if (typeof h === 'object') {
        const args = h
        h = args.h; s = args.s; v = args.v;
    }

    h = (h % 360 + 360) % 360;
    h = (h === HUE_MAX) ? 1 : (h % HUE_MAX / parseFloat(HUE_MAX) * 6)
    s = (s === SV_MAX) ? 1 : (s % SV_MAX / parseFloat(SV_MAX))
    v = (v === SV_MAX) ? 1 : (v % SV_MAX / parseFloat(SV_MAX))

    var i = Math.floor(h)
    var f = h - i
    var p = v * (1 - s)
    var q = v * (1 - f * s)
    var t = v * (1 - (1 - f) * s)
    var mod = i % 6
    var r = [v, q, p, p, t, v][mod]
    var g = [t, v, v, q, p, p][mod]
    var b = [p, p, t, v, v, q][mod]

    return {
        r: Math.floor(r * RGB_MAX),
        g: Math.floor(g * RGB_MAX),
        b: Math.floor(b * RGB_MAX),
    }
}
function rgb2hsv (r, g, b) {
    let rabs, gabs, babs, rr, gg, bb, h, s, v, diff, diffc, percentRoundFn;
    rabs = r / 255;
    gabs = g / 255;
    babs = b / 255;
    v = Math.max(rabs, gabs, babs),
    diff = v - Math.min(rabs, gabs, babs);
    diffc = c => (v - c) / 6 / diff + 1 / 2;
    percentRoundFn = num => Math.round(num * 100) / 100;
    if (diff === 0) {
        h = s = 0;
    } else {
        s = diff / v;
        rr = diffc(rabs);
        gg = diffc(gabs);
        bb = diffc(babs);

        if (rabs === v) {
            h = bb - gg;
        } else if (gabs === v) {
            h = (1 / 3) + rr - bb;
        } else if (babs === v) {
            h = (2 / 3) + gg - rr;
        }
        if (h < 0) {
            h += 1;
        }else if (h > 1) {
            h -= 1;
        }
    }
    return {
        h: Math.round(h * 360),
        s: percentRoundFn(s * 100),
        v: percentRoundFn(v * 100)
    };
}
function xy2rgb(x, y, brightness){
	//Set to maximum brightness if no custom value was given (Not the slick ECMAScript 6 way for compatibility reasons)
	if (brightness === undefined) {
		brightness = 254;
	}

	var z = 1.0 - x - y;
	var Y = (brightness / 254).toFixed(2);
	var X = (Y / y) * x;
	var Z = (Y / y) * z;

	//Convert to RGB using Wide RGB D65 conversion
	var red 	=  X * 1.656492 - Y * 0.354851 - Z * 0.255038;
	var green 	= -X * 0.707196 + Y * 1.655397 + Z * 0.036152;
	var blue 	=  X * 0.051713 - Y * 0.121364 + Z * 1.011530;

	//If red, green or blue is larger than 1.0 set it back to the maximum of 1.0
	if (red > blue && red > green && red > 1.0) {

		green = green / red;
		blue = blue / red;
		red = 1.0;
	}
	else if (green > blue && green > red && green > 1.0) {

		red = red / green;
		blue = blue / green;
		green = 1.0;
	}
	else if (blue > red && blue > green && blue > 1.0) {

		red = red / blue;
		green = green / blue;
		blue = 1.0;
	}

	//Reverse gamma correction
	red 	= red <= 0.0031308 ? 12.92 * red : (1.0 + 0.055) * Math.pow(red, (1.0 / 2.4)) - 0.055;
	green 	= green <= 0.0031308 ? 12.92 * green : (1.0 + 0.055) * Math.pow(green, (1.0 / 2.4)) - 0.055;
	blue 	= blue <= 0.0031308 ? 12.92 * blue : (1.0 + 0.055) * Math.pow(blue, (1.0 / 2.4)) - 0.055;

	//Convert normalized decimal to decimal
	red 	= Math.round(red * 255);
	green 	= Math.round(green * 255);
	blue 	= Math.round(blue * 255);

	if (isNaN(red))
		red = 0;

	if (isNaN(green))
		green = 0;

	if (isNaN(blue))
		blue = 0;

	return {
        r: red,
        g: green,
        b: blue,
    }
}

rgb = hsv2rgb(hue, sat, 100);
```

I think there is already a rgb -> xy converter in zigbee2mqtt so it should not be super hard.

odx, 2019-12-07:
OK. Now there are no more exceptions in the log. I don't think the approach with the "manual" xy -> hsv conversion will work. The problem I see is that the bulbs also don't update the xy values when the color is set through hsv.

See this log, the x/y values never change (also in the attributeReports):
```
Dec 07 20:04:54 raspberrypi npm[27420]: zigbee2mqtt:debug 2019-12-07 20:04:54: Received MQTT message on 'zigbee/garten_kugel_klein_baum/set' with data '{"color":{"hue":360, "saturation":100}, "brightness":100 }'
Dec 07 20:04:54 raspberrypi npm[27420]: zigbee2mqtt:debug 2019-12-07 20:04:54: Publishing 'set' 'brightness' to 'garten_kugel_klein_baum'
Dec 07 20:04:55 raspberrypi npm[27420]: zigbee2mqtt:info  2019-12-07 20:04:55: MQTT publish: topic 'zigbee/garten_kugel_klein_baum', payload '{"state":"ON","linkquality":47,"last_seen":"2019-12-07T20:04:55.115Z","brightness":100,"color":{"x":0.5636,"y":0.3853,"saturation":100,"hue":360},"color_temp":370,"color_mode":1}'
Dec 07 20:04:55 raspberrypi npm[27420]: zigbee2mqtt:debug 2019-12-07 20:04:55: Publishing 'set' 'color' to 'garten_kugel_klein_baum'
Dec 07 20:04:55 raspberrypi npm[27420]: zigbee2mqtt:debug 2019-12-07 20:04:55: Received Zigbee message from 'garten_kugel_klein_baum', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":100}' from endpoint 1 with groupID 0
Dec 07 20:04:55 raspberrypi npm[27420]: zigbee2mqtt:info  2019-12-07 20:04:55: MQTT publish: topic 'zigbee/garten_kugel_klein_baum', payload '{"state":"ON","linkquality":47,"last_seen":"2019-12-07T20:04:55.141Z","brightness":100,"color":{"x":0.5636,"y":0.3853,"saturation":100,"hue":360},"color_temp":370,"color_mode":1,"elapsed":22375}'
Dec 07 20:05:10 raspberrypi npm[27420]: zigbee2mqtt:debug 2019-12-07 20:05:10: Received MQTT message on 'zigbee/garten_kugel_klein_baum/set' with data '{"color":{"hue":120, "saturation":100}, "brightness":100 }'
Dec 07 20:05:10 raspberrypi npm[27420]: zigbee2mqtt:debug 2019-12-07 20:05:10: Publishing 'set' 'brightness' to 'garten_kugel_klein_baum'
Dec 07 20:05:10 raspberrypi npm[27420]: zigbee2mqtt:info  2019-12-07 20:05:10: MQTT publish: topic 'zigbee/garten_kugel_klein_baum', payload '{"state":"ON","linkquality":47,"last_seen":"2019-12-07T20:05:10.962Z","brightness":100,"color":{"x":0.5636,"y":0.3853,"saturation":100,"hue":360},"color_temp":370,"color_mode":1}'
Dec 07 20:05:10 raspberrypi npm[27420]: zigbee2mqtt:debug 2019-12-07 20:05:10: Publishing 'set' 'color' to 'garten_kugel_klein_baum'
Dec 07 20:05:10 raspberrypi npm[27420]: zigbee2mqtt:debug 2019-12-07 20:05:10: Received Zigbee message from 'garten_kugel_klein_baum', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":100}' from endpoint 1 with groupID 0
Dec 07 20:05:10 raspberrypi npm[27420]: zigbee2mqtt:info  2019-12-07 20:05:10: MQTT publish: topic 'zigbee/garten_kugel_klein_baum', payload '{"state":"ON","linkquality":47,"last_seen":"2019-12-07T20:05:10.987Z","brightness":100,"color":{"x":0.5636,"y":0.3853,"saturation":100,"hue":360},"color_temp":370,"color_mode":1,"elapsed":15846}'
Dec 07 20:05:11 raspberrypi npm[27420]: zigbee2mqtt:debug 2019-12-07 20:05:11: Received Zigbee message from 'garten_kugel_klein_baum', type 'attributeReport', cluster 'lightingColorCtrl', data '{"currentHue":85,"currentSaturation":254,"currentX":36936,"currentY":25251,"colorTemperature":370}' from endpoint 1 with groupID 0
Dec 07 20:05:11 raspberrypi npm[27420]: zigbee2mqtt:info  2019-12-07 20:05:11: MQTT publish: topic 'zigbee/garten_kugel_klein_baum', payload '{"state":"ON","linkquality":47,"last_seen":"2019-12-07T20:05:11.173Z","brightness":100,"color":{"x":0.5636,"y":0.3853,"saturation":100,"hue":120},"color_temp":370,"color_mode":1,"elapsed":189}'
Dec 07 20:05:15 raspberrypi npm[27420]: zigbee2mqtt:debug 2019-12-07 20:05:15: Received MQTT message on 'zigbee/garten_kugel_klein_baum/set' with data '{"color":{"hue":240, "saturation":100}, "brightness":100 }'
Dec 07 20:05:15 raspberrypi npm[27420]: zigbee2mqtt:debug 2019-12-07 20:05:15: Publishing 'set' 'brightness' to 'garten_kugel_klein_baum'
Dec 07 20:05:15 raspberrypi npm[27420]: zigbee2mqtt:info  2019-12-07 20:05:15: MQTT publish: topic 'zigbee/garten_kugel_klein_baum', payload '{"state":"ON","linkquality":47,"last_seen":"2019-12-07T20:05:15.981Z","brightness":100,"color":{"x":0.5636,"y":0.3853,"saturation":100,"hue":120},"color_temp":370,"color_mode":1}'
Dec 07 20:05:15 raspberrypi npm[27420]: zigbee2mqtt:debug 2019-12-07 20:05:15: Publishing 'set' 'color' to 'garten_kugel_klein_baum'
Dec 07 20:05:16 raspberrypi npm[27420]: zigbee2mqtt:debug 2019-12-07 20:05:16: Received Zigbee message from 'garten_kugel_klein_baum', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":100}' from endpoint 1 with groupID 0
Dec 07 20:05:16 raspberrypi npm[27420]: zigbee2mqtt:info  2019-12-07 20:05:16: MQTT publish: topic 'zigbee/garten_kugel_klein_baum', payload '{"state":"ON","linkquality":47,"last_seen":"2019-12-07T20:05:16.185Z","brightness":100,"color":{"x":0.5636,"y":0.3853,"saturation":100,"hue":120},"color_temp":370,"color_mode":1,"elapsed":5009}'
Dec 07 20:05:16 raspberrypi npm[27420]: zigbee2mqtt:debug 2019-12-07 20:05:16: Received Zigbee message from 'garten_kugel_klein_baum', type 'attributeReport', cluster 'lightingColorCtrl', data '{"currentX":36936,"currentY":25251,"colorTemperature":370}' from endpoint 1 with groupID 0
[... cut ...]
- Can we somehow trick the device to feed back the correct values (if possible through hsv) back after change? This looks a bit like an implementation error on the manufacturer's side. So there might not be a real solution?
- Can I turn off the caching of the state of the other values and only send the values transmitted by the Zigbee device in the MQTT message, so I can distinguish what has changed and is up-to-date? (I need this to make homebridge (mqttthing) understand wether the color_temp or the color-hsv values should be processed.)

Koenkk, 2019-12-09:
@sjorge what to fill in for the V value? Always filling in 100% won't work

@odx
- I prefer to keep values in xy only, otherwise we have to work with different color expressions
- Yes, set `cache_state` to `false` https://www.zigbee2mqtt.io/information/configuration.html

sjorge, 2019-12-09:
@Koenkk 0 will always result in black, 50 will result in very dark color tints.

I can do a xy -> hue/sat -> xy with the same result before and after if I use V=100.

odx, 2019-12-09:
Here is what I use too convert colors for my tradfri RGB bulb in addition to what @sjorge posted already (which does not support hue / saturation at all):

```
/* accepts parameters
     * r  Object = {r:x, g:y, b:z}
     * OR
     * r, g, b
     */
function RGBtoHSV(r, g, b) {
    if (arguments.length === 1) {
        g = r.g, b = r.b, r = r.r;
    }
    var max = Math.max(r, g, b), min = Math.min(r, g, b),
        d = max - min,
        h,
        s = (max === 0 ? 0 : d / max),
        v = max / 255;

    switch (max) {
        case min: h = 0; break;
        case r: h = (g - b) + d * (g < b ? 6 : 0); h /= 6 * d; break;
        case g: h = (b - r) + d * 2; h /= 6 * d; break;
        case b: h = (r - g) + d * 4; h /= 6 * d; break;
    }

    return {
        h: h,
        s: s,
        v: v
    };
}
```

and

```
/*
     * HSV to RGB conversion from https://stackoverflow.com/questions/17242144/javascript-convert-hsb-hsv-color-to-rgb-accurately
     * accepts parameters
     * h  Object = {h:x, s:y, v:z}
     * OR
     * h, s, v
     */
function HSVtoRGB(h, s, v) {
    var r, g, b, i, f, p, q, t;
    if (arguments.length === 1) {
        s = h.s, v = h.v, h = h.h;
    }
    i = Math.floor(h * 6);
    f = h * 6 - i;
    p = v * (1 - s);
    q = v * (1 - f * s);
    t = v * (1 - (1 - f) * s);
    switch (i % 6) {
        case 0: r = v, g = t, b = p; break;
        case 1: r = q, g = v, b = p; break;
        case 2: r = p, g = v, b = t; break;
        case 3: r = p, g = q, b = v; break;
        case 4: r = t, g = p, b = v; break;
        case 5: r = v, g = p, b = q; break;
    }
    return {
        r: Math.round(r * 255),
        g: Math.round(g * 255),
        b: Math.round(b * 255)
    };
}

function ScaledHSVtoRGB(h, s, v) {
    return HSVtoRGB(h / 360, s / 100, v / 100);
}
```

For "Value" V in HSV i use the bulbs brightness.

My issue applying these conversions to the xy (which describe by the way die CIE color model) is that not all models cover the same color values so by nature not all conversions are reversible. (Besides the numeric errors and possibly implementation inaccuracies).
For the tradfri remote which only supports a specific set of colors this is not such a big deal for me, but for real RGB bulbs applying multiple conversions after each other will result in a shift in the color when colors are converted back and forth multiple times. This is the reason why I prefer to apply the direct hue / saturation messages because this is what my smart home platform talks (and I have no choice about this).

@Koenkk i will give cache_state = false a try. That should do the trick for me.

closed 2020-02-15 by stale[bot]