Koenkk/zigbee2mqtt #2451: MQTT Broker : Publish States on subscribe

moelski, 2019-12-02:
# Bug Report
## What happened
I use iobroker for my home environment and the iobroker mqtt broker for zigbee2mqtt. There is a (very useful) option in the mqtt broker which is called "Publish States on subscribe". If this option is enabled a client will get the topic data directly once after the subscription. 

If the option is enabled z2m will produce a lot of errors like this: 
```
zigbee2mqtt:error 2019-12-02T06:59:56: Publish 'set' 'state' to 'EZ_RGB_GLEDOPTO_Decke2' failed: 'Error: Data request failed with error: 'MAC no ack' (233)'
zigbee2mqtt:info  2019-12-02T06:59:56: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'EZ_RGB_GLEDOPTO_Decke2' failed: 'Error: Data request failed with error: 'MAC no ack' (233)'","meta":{"friendly_name":"EZ_RGB_GLEDOPTO_Decke2"}}'
zigbee2mqtt:error 2019-12-02T06:59:56: Publish 'set' 'state' to 'WZ_DIMM_Tradfri_Decke1' failed: 'Error: Data request failed with error: 'MAC no ack' (233)'
zigbee2mqtt:info  2019-12-02T06:59:56: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'WZ_DIMM_Tradfri_Decke1' failed: 'Error: Data request failed with error: 'MAC no ack' (233)'","meta":{"friendly_name":"WZ_DIMM_Tradfri_Decke1"}}'
zigbee2mqtt:error 2019-12-02T06:59:56: Publish 'set' 'state' to 'WZ_DIMM_Tradfri_Decke2' failed: 'Error: Data request failed with error: 'MAC no ack' (233)'
zigbee2mqtt:info  2019-12-02T06:59:56: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'WZ_DIMM_Tradfri_Decke2' failed: 'Error: Data request failed with error: 'MAC no ack' (233)'","meta":{"friendly_name":"WZ_DIMM_Tradfri_Decke2"}}'
```
If the option is disabled all works well ... 

## What did you expect to happen
Would be nice if z2m could handle the initial topic receiving from the broker after the subscription. 

## How to reproduce it (minimal and precise)
In iobroker just enable the option "Publish states on subscribe". 
I don know if this is implemented in mosquitto. Couldn't find an option for this. 

## Debug Info
zigbee2mqtt version: 1.7.0
CC253X firmware version: latest

tetzlav, 2019-12-02:
I have had a similar problem with really a lot of publish logs at zigbee2mqtt (re)start. I never found out what the reason of this. Sometimes it takes about 3-5min to startup to calm down this publishing messages throu my 12 devices and get a quiet log. I searched the internet an here in the issue list, but nobody lese seems to have the same problem.

Now - thanks for hint - i disabled "Publish states on subscribe" in iobroker mqtt adapter and *bingo* there is a quiet and "normal" startup of zigbee2mqtt daemon...

There is something like ping-pong publish-subscribe game between zigbee2mqtt and iobroker?

I use:
iobroker 2.2.1 with MQTT Broker Adapter 2.0.4
and zigbee2mqtt 1.7.1 with firmware CC2531_DEFAULT_20190608

sjorge, 2019-12-02:
It would be great with a QoS higher than 0 it would retry the message with a backdown if the network is too busy.

moelski, 2019-12-02:
@sjorge: Well I could say my broker was never too busy. To be honest Z2M is the only tool which uses the broker. And there is only one subscriber (also Z2M).

sjorge, 2019-12-02:
network being the zigbee mesh network and not the mqtt traffic, sorry for the confusing.

moelski, 2019-12-03:
Oh ok, didn't got that. 
Anyway ... There is a problem in Z2M when the broker sends the topic data after the subscription. 
Would be very useful if this could be fixed.

closed 2020-02-08 by stale[bot]