Koenkk/zigbee2mqtt #2453: Imposible to repair my Aqara wired switch without a fresh install of hassio and Z2M

uxdesignerhector, 2019-12-02:
# Bug Report
## What happened
- I am not able to repair my Aqara switch any more without a fresh install of Z2M and hassio. I think it's a software problem.
## What did you expect to happen
- To repair my Aqara wired switch without having to start a new fresh installation again.
## How to reproduce it (minimal and precise)
-  I believe the problem is caused by a malfunctioning in the light installation, sometimes when I plug my lamp on my computer loses the mouse signal for a few seconds. This can be caused by a fan I have in my room, as I have read in some engineering forum the internal electronics can create electromagnetic fields and thanks to physics it ends resetting my switch and freezing my mouse for seconds (notice my switch doesn't turn off when this happens and it's turned on) So I'm not sure if this is the real reason.

- Anyway, the switch disconnects itself from time to time, this is not a problem as I've been able to repair it. But after some "magic" conditions meets, it's impossible to repair it again. This doesn't happens with other switches with the same model that are in other rooms.

## Debug Info
- zigbee2mqtt version: 1.7.1 (hassio addon)
- zigbee2mqtt log: I don't have access to it, but I remember it said something like this: **"Transation Mac expired. Unable to reach ..."**

- CC2531 firmware version: Z-Stack_Home_1.2
- Hassio version: 0.102.3

## What I have tried to solve it and it doesn't work.
- Removing the device from Z2M (zigbee2mqtt/bridge/config/remove). 
- Deleting it manually from the Z2M database.
- Resetting ZNP stick. (zigbee2mqtt/bridge/config/reset)
- Re-flashing the CC2531. (Erase program and verify)
- Switching off the main power from my house in case the switch has some kind of memory.
- Deleting the mosquito Integration in Home Assistant.
- Uninstalling and reinstalling Mosquito.
- Uninstalling and reinstalling Z2M hassio add-on.
- Deleting the device from the core_entity_registry in Home Assistant.

## To take in consideration.
- As soon as I run and a fresh install of hassio and Z2M, everything works again.
- when pairing my switch it turns from red lights to blue, then blue lights get stuck flashing and finally turn off, after that if I turn on again my switch they are still red, so I suppose the pairing process was not successful. 
- Zigbee to mqtt in debug mode doesn't receive any package from my switch in pairing mode. 
- This has happened to me two times.

PaeseSono, 2019-12-03:
It might be a defect switch or the light issue you mention, I suggest you could try moving your switch location to a different light and let the previous 1 disconnect while testing. Do you have other switch to test as well or the other is working well except this?

uxdesignerhector, 2019-12-03:
I don't have another switch with the  same model, the next time if it happens again I will change the location of this switch in the room. Right now I'm using a fresh install again.  The other switches are double switches, this one is single. There are not problems with the double switches which are in other rooms and are from the same manufacturer. I will leave this issue opened just to track everything and find out were is the error taking place.

**Main suspicions right now:**
- The switch changes its emission channel? (Due to Z2M stopped to receive messages from this switch when in pairing mode)
- Memory, like a cached or just a stored database value gets registered in the dark and when pairing with the new values (assuming the switch changes its internal values like a hash or id... something like that) cause an unregistered error.

WouterJN, 2019-12-04:
I'm having the same issue with one of my Aqara switches, it could be broken hardware.

uxdesignerhector, 2019-12-31:
Today the problem said hello again. I've been able to fix it by repairing it at the third time. Today I can confirm is not a malfunctioning at the installation level. I was trying a new automation while it has happened. **It's seems to be a connectivity network  problem**. 

`zigbee2mqtt:error 2019-12-31 17:51:51: Publish 'set' 'state' to '0x00158d000xxxxxxx' failed: 'Error: Data request failed with error: 'MAC transaction expired' `(240)'`

It's funny this is the only switch doing this, others wit low quality range don't loose the signal. As far I have been able to find the **main problem, 'connectivity network'**, I will apply some solutions that are suggested at the docs to improve my network quality.

Closing the issue. Thank you all for helping me.

closed 2019-12-31 by uxdesignerhector