Koenkk/zigbee2mqtt #2454: Add a new device (Halo Smoke/CO Detector)

alex-savin, 2019-12-02:
# Bug Report
Support for a new device (Halo Smoke/CO Detector)
## What happened
Not Supported
## What did you expect to happen
Add a support
## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.7.1

```
zigbee2mqtt:info  2019-11-30T16:19:59: Device '0x000d6f000c0464b1' joined
zigbee2mqtt:info  2019-11-30T16:19:59: MQTT publish: topic 'zigbee2mqtt01/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x000d6f000c0464b1"}}'
zigbee2mqtt:info  2019-11-30T16:19:59: Starting interview of '0x000d6f000c0464b1'
zigbee2mqtt:info  2019-11-30T16:19:59: MQTT publish: topic 'zigbee2mqtt01/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x000d6f000c0464b1"}}'
zigbee2mqtt:warn  2019-11-30T16:19:59: Received message from unsupported device with Zigbee model 'undefined'
zigbee2mqtt:warn  2019-11-30T16:19:59: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:warn  2019-11-30T16:19:59: Received message from unsupported device with Zigbee model 'haloWX'
```

```
zigbee2mqtt:debug 2019-11-30T19:18:39: Device '0x000d6f000c0464b1' announced itself
zigbee2mqtt:debug 2019-11-30T19:18:39: Received Zigbee message from '0x000d6f000c0464b1', type 'raw', cluster '64768', data '{"type":"Buffer","data":[13,1,18,0,0,0]}' from endpoint 4 with groupID 0
```

After I was trying to add it myself, I got more at Z2M log
```
zigbee2mqtt:debug 2019-11-30T19:18:50: Received Zigbee message from '0x000d6f000c0464b1', type 'readResponse', cluster 'genBasic', data '{"modelId":"haloWX","manufacturerName":"HaloSmartLabs","powerSource":3}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2019-11-30T19:18:50: No converter available for 'SABDA1' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"haloWX","manufacturerName":"HaloSmartLabs","powerSource":3}'
zigbee2mqtt:debug 2019-11-30T19:18:50: Received Zigbee message from '0x000d6f000c0464b1', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1,"appVersion":1,"stackVersion":0}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2019-11-30T19:18:50: No converter available for 'SABDA1' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":1,"appVersion":1,"stackVersion":0}'
zigbee2mqtt:debug 2019-11-30T19:18:50: Received Zigbee message from '0x000d6f000c0464b1', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2019-11-30T19:18:50: No converter available for 'SABDA1' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1}'
```

I found a couple files which could be used as a source od information for this device

Device handler for SmartThings Hub
https://github.com/jsconstantelos/SmartThings/blob/40a50aabcf667a04538dc42de9e08ad6588672a5/devicetypes/jsconstantelos/my-halo-smoke-alarm.src/my-halo-smoke-alarm.groovy

Device handler for Habitat Hub
https://github.com/hubitat/HubitatPublic/blob/8b87f7c6e7bd19a904d9c19888feedbe3b551060/examples/drivers/haloSmokeCoDetector.groovy

and Arcus Platform
https://github.com/t1minator/arcusplatform/blob/ed2a70ed9f88f46bb64a1e7d263141c48f26e530/platform/arcus-containers/driver-services/src/main/resources/ZB_Halo_HaloPlus_2_3.driver

My changes to file `devices.js`
```
    {
        zigbeeModel: ['haloWX', 'halo+', 'SABDA1'],
        model: 'SABDA1',
        vendor: 'HaloSmartLabs',
        description: 'Halo Smoke/CO Alarm',
        supports: 'on/off, brightness, color xy, smoke, temperature and humidity',
        fromZigbee: [
            fz.color_colortemp,
            fz.on_off,
            fz.brightness,
            fz.humidity,
            fz.generic_pressure,
            fz.temperature
        ],
        toZigbee: [
            tz.light_onoff_brightness,
            tz.light_color,
            tz.ignore_transition,
            tz.light_alert
        ],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint1 = device.getEndpoint(1);
            const endpoint2 = device.getEndpoint(2);
            const endpoint3 = device.getEndpoint(3);
            const endpoint4 = device.getEndpoint(4);
            const endpoint5 = device.getEndpoint(5);
            await bind(endpoint1, coordinatorEndpoint, ['genBasic', 'genPowerCfg']);
            await bind(endpoint2, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl', 'lightingColorCtrl']);
            await bind(endpoint3, coordinatorEndpoint, ['msTemperatureMeasurement', 'msPressureMeasurement']);
            await configureReporting.batteryVoltage(endpoint1);
        },
```

file `homeassistant.js`
```
    'SABDA1': [cfg.light_brightness_colortemp_colorxy, cfg.sensor_temperature, cfg.sensor_humidity, cfg.sensor_pressure],
```
But without success
Could you please point me out have I can add it on my own, or give any help to make it possible to be supported

Thank you in advance,
Alex

closed 2020-02-07 by stale[bot]

vesomo, 2020-04-01:
I am interested in zigbee2mqtt supporting Halo detectors as well. Please help to figure it out. I use some at my house and happy to test

genelevi, 2022-07-09:
Yes please  :-)