Koenkk/zigbee2mqtt #2455: Add E12 to existing trådfri device LED1624G9

thefarelkid, 2019-12-02:
The LED1624G9 bulb comes in a few sizes.  I have a newer E12 model.  I've updated it to the devices.js file and it works great for me.  Adding it by default though would be great!

        zigbeeModel: [
            'TRADFRI bulb E27 CWS opal 600lm',
            'TRADFRI bulb E26 CWS opal 600lm',
            'TRADFRI bulb E14 CWS opal 600lm',
            'TRADFRI bulb E12 CWS opal 600lm'],
        model: 'LED1624G9',
        vendor: 'IKEA',
        description: 'TRADFRI LED bulb E14/E26/E27/E12 600 lumen, dimmable, color, opal white',
        extend: generic.light_onoff_brightness_colorxy,

sjorge, 2019-12-03:
I've made a PR for this against zigbee-herdsman-convertors, once @Koenkk merged this it should be available in future versions.

Koenkk, 2019-12-03:
Merged the pr from @sjorge thanks again! Will be available in the dev version in a few hours.

closed 2019-12-03 by Koenkk