Koenkk/zigbee2mqtt #2456: Yale YMF40 lock "Timeout .... after 15000ms"

BrunoZumba, 2019-12-03:
# Bug Report
Can't get any data FROM YMF40 lock TO Zigbee.
## What happened
A few weeks ago I was able to connect my YMF40 lock with Zigbee2MQTT. Everything was working ok. But after rebooting my raspberry pi, I couldn't get any data FROM the LOCK to the zigbee stick.

I can send Zigbee messages from MQTT to the Lock. I send the message to open and it unlocks... everything seems ok. But no data gets transmitted from the lock to the stick.

After I send the unlock command, for example, it unlocks but it doesn't send the message confirming that it is unlocket... I don't get the confirmation on MQTT, as I used to. Similarly, when I lock/unlock manually, I don't get the MQTT message.

One thing that is weird is that after I publish the unlock message, I get this:

```
Dec 02 22:42:07 raspberrypi npm[20192]: zigbee2mqtt:debug 2019-12-02 23:42:07: Received MQTT message on 'zigbee2mqtt/fechadura_yale/set' with data 'UNLOCK'
Dec 02 22:42:07 raspberrypi npm[20192]: zigbee2mqtt:debug 2019-12-02 23:42:07: Publishing 'set' 'state' to 'fechadura_yale'
Dec 02 22:42:22 raspberrypi npm[20192]: zigbee2mqtt:error 2019-12-02 23:42:22: Publish 'set' 'state' to 'fechadura_yale' failed: 'Error: Timeout - 54749 - 1 - 3 - 257 - 11 after 15000ms'
Dec 02 22:42:22 raspberrypi npm[20192]: zigbee2mqtt:info  2019-12-02 23:42:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'fechadura_yale' failed: 'Error: Timeout - 54749 - 1 - 3 - 257 - 11 after 15000ms'","meta":{"friendly_name":"fechadura_yale"}}'
```

I've been struggling with this for a few weeks. I tried everything I could think of:
1. Erased lock entries at configuration.yaml, database.db and pairing again (a lot of times)
2. Moved the raspberry pi to about 5 feets from the lock and repaired item 1.
3. Just now I updated zigbee2mqtt, reflashed the stick, repeated steps 1 and 2... and nothing.

## Debug Info
zigbee2mqtt version: `zigbee2mqtt@1.7.1` (DEV version)
CC253X firmware version: `Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'`

Koenkk, 2019-12-03:
You can probably fix this by changing https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L4534 to `meta: {options: {disableDefaultResponse: true}, configureKey: 1},`. See this on how to get to `devices.js`: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

BrunoZumba, 2019-12-05:
> You can probably fix this by changing https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L4534 to `meta: {options: {disableDefaultResponse: true}, configureKey: 1},`. See this on how to get to `devices.js`: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Thanks, Koenkk!!!! That did the trick... or at least a little bit.

Now I don't get the timeout and I can lock/unlock the door... but it's like the "fromZigbee" part is not working.

I get **no** messages from the lock. When I lock/unlock it MANUALLY, I don't get the zigbee messages.

As a test, I ereased all the "converters" I had in the "fromZigbee" part of my devices.js... and everything continues the same (i.e. I can lock/unlock via MQTT, but don't get any messages from the lock).

Any idea?

Devices.js (for some reason, my Zigbee Adapter is 0700000001 instead of iZBModule01)

```
{
//        zigbeeModel: ['iZBModule01'],
        zigbeeModel: ['0700000001'],
        model: 'YMF40',
        vendor: 'Yale',
        description: 'Real living lock',
        supports: 'lock/unlock, battery',
        fromZigbee: [fz.lock_operation_event, fz.battery_200, fz.lock],
        toZigbee: [tz.generic_lock],
//        meta: {configureKey: 1},
        meta: {options: {disableDefaultResponse: true}, configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await configureReporting.lockState(endpoint);
            await configureReporting.batteryPercentageRemaining(endpoint);
        },
    },
```

fromZigbee.js
```
lock_operation_event: {
        cluster: 'closuresDoorLock',
        type: 'commandOperationEventNotification',
        convert: (model, msg, publish, options) => {
            const unlockCodes = [2, 9, 14];
            return {
                state: unlockCodes.includes(msg.data['opereventcode']) ? 'UNLOCK' : 'LOCK',
                user: msg.data['userid'],
                source: msg.data['opereventsrc'],
            };
        },
    },
battery_200: {
        cluster: 'genPowerCfg',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options) => {
            const batt = msg.data.batteryPercentageRemaining;
            const battLow = msg.data.batteryAlarmState;
            const voltage = msg.data.batteryVoltage;
            const results = {};
            if (batt != null) {
                const value = Math.round(batt/200.0*10000)/100; // Out of 200
                results['battery'] = value;
            }
            if (battLow != null) {
                if (battLow) {
                    results['battery_low'] = true;
                } else {
                    results['battery_low'] = false;
                }
            }
            if (voltage != null) {
                results['voltage'] = voltage * 100;
            }
            return results;
        },
    },
lock: {
        cluster: 'closuresDoorLock',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options) => {
            if (msg.data.hasOwnProperty('lockState')) {
                return {state: msg.data.lockState == 2 ? 'UNLOCK' : 'LOCK'};
            }
        },
    },
```

toZigbee.js
```
generic_lock: {
        key: ['state'],
        convertSet: async (entity, key, value, meta) => {
            await entity.command(
                'closuresDoorLock',
                `${value.toLowerCase()}Door`,
                {'pincodevalue': ''},
                getOptions(meta)
            );

            return {readAfterWriteTime: 200, state: {state: value.toUpperCase()}};
        },
        convertGet: async (entity, key, meta) => {
            await entity.read('closuresDoorLock', ['lockState']);
        },
    },
```

configuration.yaml

```
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://***.***.***.***'
  user: *****
  password: ******
serial:
  port: /dev/ttyACM1
  disable_led: true
advanced:
  rtscts: true
  log_level: debug
devices:
  '0x00158d0002edb3a1':
    friendly_name: cortina
    retain: false
  '0x00158d0002b5472b':
    friendly_name: temp_hum_sensor
  '0x000d6f0012bb1db5':
    friendly_name: fechadura_yale
```

Here are a few parts of my log.
When I send the "UNLOCK" message via MQTT (home assistant):
```
Dec 04 21:46:18 raspberrypi npm[18706]: zigbee2mqtt:debug 2019-12-04 22:46:18: Received MQTT message on 'zigbee2mqtt/fechadura_yale/set' with data 'UNLOCK'
Dec 04 21:46:18 raspberrypi npm[18706]: zigbee2mqtt:debug 2019-12-04 22:46:18: Publishing 'set' 'state' to 'fechadura_yale'
Dec 04 21:46:20 raspberrypi npm[18706]: zigbee2mqtt:info  2019-12-04 22:46:20: MQTT publish: topic 'zigbee2mqtt/fechadura_yale', payload '{"state":"UNLOCK","user":0,"source":2,"linkquality":94,"battery":45}'
```

At this point, my home assistant get the confirmation that the lock is unlocked. 
A few seconds later, the lock autolock... and nothing happens in the log.
Just as well, if I lock/unlock the door manually, I get nothing in the logs.

Sorry for the long post and thanks!

BrunoZumba, 2019-12-05:
Ok, I just saw that the thread #2339 mentioned this thread. I went to read it... and just like this thread helped them, their thread helped me lol

As you said there, I changed my devices.js to add the "bind" thing and I started to get some messages from the lock. I'm fiddling with it right now... It seems I am making some advance... I'll report back in a while.

Here's my "new" devices.js with the bind thing
```
//        zigbeeModel: ['iZBModule01'],
        zigbeeModel: ['0700000001'],
        model: 'YMF40',
        vendor: 'Yale',
        description: 'Real living lock',
        supports: 'lock/unlock, battery',
        fromZigbee: [fz.lock_operation_event, fz.battery_200, fz.lock],
        toZigbee: [tz.generic_lock],
//        meta: {configureKey: 1},
        meta: {options: {disableDefaultResponse: true}, configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
            await configureReporting.lockState(endpoint);
            await configureReporting.batteryPercentageRemaining(endpoint);
        },
    },
```

BrunoZumba, 2019-12-05:
Yep, that was it... after adding the line below everything seems to be working correctly now.
Thanks

`await bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);`

closed 2019-12-05 by BrunoZumba

Koenkk, 2019-12-05:
Thanks, will be updated in next zigbee2mqtt release.

moistgun, 2019-12-27:
Only very slightly related note, I am using the YRD226HA2619 lock, and I am able to control and get information from the lock, but everytime I initiate a lock or unlock command, the log has an entry as seen below.

`No converter available for 'YRD226HA2619' with cluster 'closuresDoorLock' and type 'commandOperationEventNotification' `

(I created the initial PR for this lock and at the time didn't really know what I was doing)

arduino43, 2019-12-27:
> Only very slightly related note, I am using the YRD226HA2619 lock, and I am able to control and get information from the lock, but everytime I initiate a lock or unlock command, the log has an entry as seen below.
> 
> `No converter available for 'YRD226HA2619' with cluster 'closuresDoorLock' and type 'commandOperationEventNotification' `
> 
> (I created the initial PR for this lock and at the time didn't really know what I was doing)
> 
> Edit:
> 
> Took a later look at the logs and randomly got a similar timeout error as OP
> `zigbee2mqtt:error 2019-12-26 20:24:55: Publish 'set' 'state' to '0x000d6f000fedxxxx' failed: 'Error: Timeout - 39666 - 1 - 11 - 257 - 11 after 15000ms'`
> 
> If at all possible, it would be great if YRD226HA2619 could get the same treatment as the YMF40 I would be a happy camper

I have the exact same problem with my YRD226HA2619

BrunoZumba, 2019-12-27:
> Only very slightly related note, I am using the YRD226HA2619 lock, and I am able to control and get information from the lock, but everytime I initiate a lock or unlock command, the log has an entry [...]

About the "no converter available..." message: did you check the page on supporting new devices? ( https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html )
Sections 2 and 3 on that page explain a little bit on how to add converters.

moistgun, 2019-12-28:
> About the "no converter available..." message: did you check the page on supporting new devices? ( https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html )
> Sections 2 and 3 on that page explain a little bit on how to add converters.

At the time, sure, but I mentioned I didn't really know what I was doing, otherwise I would have made it work without the errors. 

Basically once it started showing the states and responding to commands, I stopped and didn't even know it was logging out the timeouts until after a while it was misbehaving.

Now I don't have a setup to test with, and am just pulling the hassio addon

Koenkk, 2020-01-03:
@moistgun @BrunoZumba these issues are fixed now, will be available in the dev branch in a few hours.