Koenkk/zigbee2mqtt #2457: Issue with Yale YDM3168

msadeghz, 2019-12-03:
Hi, 
I'm trying to join YDM3168 Yale door lock to CC2531 based coordinator.
the door lock sends Beacon request and the coordinator sends the beacon. but the door lock doesn't continue join procedure. this door lock model is compatible with Samsung SmartThings hub and I think it can be compatible with zigbee2mqtt.

msadeghz, 2019-12-03:
I tried with another zigbee module and the issue is resolved. actually, the zigbee module was damaged.

closed 2019-12-03 by msadeghz

gagandeep90, 2020-05-28:
> I tried with another zigbee module and the issue is resolved. actually, the zigbee module was damaged.

Hi, could you please send a link or a photo to the zigbee module you are using in the YDM3168 as i am planning to buy it. And are all of the functions and states of the lock working correctly in Zigbee2Mqtt?