Koenkk/zigbee2mqtt #2458: Bitron Video Relay Wall Thermostat Home Assistent Intergration

RienkDen, 2019-12-03:
I am working on intergrating the Bitron Video Relay Wall Thermostat to Home Assistant.

**Model Nummer:** AV2010/32
**Zigbee2Mqtt Version:** 1.7.1

Currently the home intergration is the following file _lib/extension/homeassistant.js_ is: 
`'AV2010/32': [],`

I changed this to `'AV2010/32': [cfg.thermostat()],` now it shows up in home assistant but it does not report back its state (heat/off/auto).

When slightly changing the cfg.thermostat() from :
`mode_state_template: '{{ value_json.system_mode }}',` **to:** `mode_state_template: '{{ value_json.running_state }}'` it reports back the right state (heat/off/auto) mode back to home assistant.

But when clicking on the different states (heat/off/auto) in home assistant it gives the following error in zigbee2mqtt:

```
Dec 03 09:26:11 raspberrypi npm[1324]: zigbee2mqtt:error 2019-12-03T09:26:11: No converter available for 'system_mode' (auto)
Dec 03 09:26:19 raspberrypi npm[1324]: zigbee2mqtt:error 2019-12-03T09:26:19: No converter available for 'system_mode' (off)
Dec 03 09:26:26 raspberrypi npm[1324]: zigbee2mqtt:error 2019-12-03T09:26:19: No converter available for 'system_mode' (heat)
```

When looking at the package which is being send from the thermostat it indeed seems that the thermostat does not have a **system_mode** variable but it does have a **running_state** variable.

`{"local_temperature":21.5,"linkquality":68,"occupied_heating_setpoint":21.5,"local_temperature_calibration":0,"running_state":"heat","battery":"71.43","voltage":3000,"battery_alarm_state":0}`

My assumption would be that we need to send a running_state back instead of system_mode to correctly send back the state it is running in from Home Assistant. How would I be able to do this?

SBTephan, 2019-12-04:
what widget are you using in home assistant to change the temp's etc. ?

Koenkk, 2019-12-05:
You should first check if this thermostat supports the system mode, that can be done by adding to https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L3643 `tz.thermostat_system_mode`

closed 2020-02-10 by stale[bot]