Koenkk/zigbee2mqtt #2459: IKEA E1525 - IKEA TRADFRI motion sensor without converter in DEFAULT branch

uxdesignerhector, 2019-12-03:
# Bug Report
## What happened
- IKEA TRADFRI motion sensor without converter in **default branch**, like in #1322 
## What did you expect to happen
- To get the status of IKEA TRADFRI motion sensor using MQTT
## How to reproduce it (minimal and precise)
`zigbee2mqtt/[friendly_name_motion_sensor_ikea]/get`
```
{
  "state": ""
}
```
## Debug Info
```
zigbee2mqtt:error 2019-12-03T19:26:26: No converter available for 'state' (occupancy)
zigbee2mqtt:error 2019-12-03T19:26:57: No converter available for 'state' (linkquality)
zigbee2mqtt:error 2019-12-03T19:38:54: No converter available for 'state' () "<-just in case I'm typing it wrong"
```
- zigbee2mqtt  version: *1.7.1 default*
- CC2531 firmware version: Z-Stack_Home_1.2

zhapwey, 2019-12-04:
hello:

the end devices do not support recovering the status, they have to send this status when it changes. The final devices are asleep most of the time and cannot return their status at any time.
This can be done in rooters who, when connected to constant current, never sleep.

uxdesignerhector, 2019-12-04:
Thank you for the info. Closing it.

closed 2019-12-04 by uxdesignerhector