Koenkk/zigbee2mqtt #2461: Device from the past appears in new installation

HaflingerWallach, 2019-12-04:
Hello!

After moving to a new home I made a new Raspberry Pi SD-card and installed it from scratch. Obviously I installed zigbee2mqtt also.

After starting the new installation an error appears that the system can't find an TRADFRI Entity which was installed on the old card. Eventually I can't start it as a service, it exists immediatly.

The "joke": The SD-card is new, database.db is empty and I even flashed a new CC2531-stick with the latest firmware - it still remembers the old device in the fresh installation.

After several trys (deleting /opt/zigbee2mqtt and clone from github) I kindly ask for your advise!

TIA

uxdesignerhector, 2019-12-04:
Some tips that can help you:
- When flashing the SD card, make sure you slow format it previously I was having similar problems a year ago with Hassio.
- I believe, but I'm not sure at all, that the stick have some kind of memory where it stores the Zigbee IEEE address, and some data. Have you select the  the option in Actions, "Erase, program and verify", in Texas Instruments SmartRF Flash Programmer when flahsing?

sjorge, 2019-12-04:
I noticed if you have joining enable and use the same panId/network key about 2/3 of my devices automatically reapear... did you keep the same panId/netkey?

closed 2020-02-09 by stale[bot]