Koenkk/zigbee2mqtt #2463: My Gledopto devices do not contribute to the mesh

grostim, 2019-12-04:
# Bug Report
## What happened

I am not sure if this issue is related to Zigbee2mqtt or to the gledopto firmware.
I have been facing a lot of communication issues.
After further analysis, i realize that none of my (many) gledopto device seems to act as a router:
See network map:
https://svgshare.com/i/GX4.svg
The other devices (innr, philips, osram) are working as expected.
Is this a known issue with the Gledopto devices ?
Am i the only one here to face this issue ?

## What did you expect to happen
Proper meshing of the gledopto devices.

## Debug Info
zigbee2mqtt version: 1.7.1-dev
CC253X firmware version: CC2531_DEFAULT_20190608.zip

Koenkk, 2019-12-04:
I think they already provided you the answer here: https://www.reddit.com/r/Gledopto/comments/e4n8d0/my_gledopto_zll_lights_seems_to_have_a_poor_range/ From the zigbee2mqtt side we cannot solve this.

grostim, 2019-12-04:
As i said, i am not sure if the issue is related to my setup, the gledopto firmware or zigbee2mqtt. 
I am trying to gather additional testimony in order to identify the root cause.

At this stage the manufacturer answer is not very helpful.
1) They tend to blame the distance. This can not be the cause. Osram & philips device communicate 20meters appart with 2 walls in between while gledopto are not able to connect to each other even at 1 meter distance.
I will contact their customer service as per their suggestion.
2) They say "Some customers told us they can not "see" the mesh with dongle.". I am not sure if this is a way to confirm that they are aware that this is an issue with their firmware of if they blame the dongle .
3) Auto-transmitting and Auto-sync function: This is completely unrelated. This is a feature of their "Plus" products which use both zigbee and RF communication. These features is related to the RF remote only, not zigbee.

closed 2020-02-09 by stale[bot]