Koenkk/zigbee2mqtt #2465: OSRAM Flex Transition

sti0, 2019-12-04:
#Bug Report
## What happened
If you turn a Osram Flex light strip on and set the transition to 0.4 it does not respect that transition time.

## What did you expect to happen
If I set a transition time it should be fade in/out on turning it on or off. Like it works for HUE light strips. In my tests the transition time only works by color changing.

## How to reproduce it (minimal and precise)
Set transition time in devices.yaml

```
'0x7cb0334a00ae938f':
  friendly_name: some_light_strip
  transition: 0.4
```

## Debug Info
zigbee2mqtt version: 1.7.1 dev
CC26X2R1 firmware version: 20191106

Koenkk, 2019-12-05:
Zigbee does not support on/off transitions, for Hue bulbs also don't, but they just always go off with a transition (without specifying it).

sti0, 2019-12-06:
Ok. Thanks for the information.

closed 2019-12-06 by sti0