Koenkk/zigbee2mqtt #2466: Taking control over a paired device

d4s1r, 2019-12-04:
# Bug Report
## What happened
I am able to take control of my paired OSRAM AB3257001NJ (Smart+ Plug) with my IKEA E1810 (TRADFRI remote control)
## What did you expect to happen
I expected my device to stick to the broker instead of pairing without pressing buttons
## How to reproduce it (minimal and precise)
Press the button next to the battery (inside the tray) of the Tradfri remote for 10 seconds close to the Smart+ Plug
## Debug Info
zigbee2mqtt version: 1.6.0
CC253X firmware version: 20190608

piit79, 2019-12-07:
Let me see if I understand correctly. You are saying that a device paired with zigbee2mqtt on its own? Or what exactly are you saying? The report is not very clear.

d4s1r, 2019-12-07:
The Osram-Device was paired to zigbee2mqtt, which runs on HassOS, however, I was able to depair the Osram-Device (Hass/zigbee2mqtt lost the route (because the device refused the connection I assume). Does that clarify?

tbowmo, 2019-12-07:
I do not se this as a bug in zigbee2mqtt, that a random device disconnects and then pairs to another controller.

Koenkk, 2019-12-07:
This is expected, the remote uses touchlink to pair the device to itself. This cannot be avoided as this is just how touchlink works. Read more about it e.g. here: https://redsilico.com/blog/ikea-tradri-remote-with-hue-hub

closed 2019-12-07 by Koenkk