Koenkk/zigbee2mqtt #2467: Oblo living smart plug - model "undefined"

andrej-peterka, 2019-12-05:
# Bug Report
## What happened

I'm trying to get an unsupported zigbee smart plug smart plug working with zigbee2mqtt, but I'm hitting a problem which seems to be kinda known, but not quite fixed yet.

When trying to pair the plug, I'm getting an error:

```
Device '0x00124b0014cd8374' with Zigbee model 'undefined' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
```

Here's the full debug log:
https://pastebin.com/n2eZcWNy

The plug is this one:
https://www.obloliving.com/oblo-smart-plug/

Here's the manual:
https://www.obloliving.com/wp-content/uploads/2017/05/OBLO_Living-Smart_Plug_SPW35ZD0-User_Manual-20170525.pdf

I'm trying to pair it in "Join the gateway’s network" mode (1x short press in SETUP mode). But I'm getting the same issues when trying other modes, though I'm honestly not sure what they mean. Any insight into this would be greatly appreciated.

The plug seems to be a rebranded Schwaiger ZHS-15 (model SPW35Z-D0). I see something was being worked on here: https://github.com/Koenkk/zigbee2mqtt/issues/393

## What did you expect to happen
I expected the device to pair so I can try adding support for it.

## How to reproduce it (minimal and precise)
Try pairing the device.

## Debug Info
zigbee2mqtt version: latest (git)
CC253X firmware version: latest

Koenkk, 2019-12-05:
The problem is that device does not provide a modelID, zigbee2mqtt tries to read it but the device indicates it doesn't have it. As a workaround you can set the modelID manually in `database.db` for this device. Set it for example to `OBLO_PLUG`, after that you can continue as normal.

andrej-peterka, 2019-12-06:
Thanks for the quick and helpful reply!

I got the device working up to the point I can turn the power on and off via mqtt publish. This is the initial implementation:

```javascript
{
    zigbeeModel: ['OBLO_PLUG'],
    model: 'SPW35Z-D0',
    vendor: 'Oblo',
    description: 'Power socket on/off with power consumption monitoring',
    supports: 'on/off, power measurement',
    fromZigbee: [],
    toZigbee: [tz.on_off],
},
```

As a basis, I used the code @tb-killa posted in the issue #393 and noticed it doesn't work anymore. I assume this is due to an API change.

I would love to update below code to the new API and test, but need some guidance.

```javascript
{
    zigbeeModel: ['SPW35Z-D0'],
    model: 'ZHS-15',
    vendor: 'Schwaiger',
    description: 'Power socket on/off with power consumption monitoring',
    supports: 'on/off, power measurement',
    fromZigbee: [fz.generic_state, fz.ignore_onoff_change, fz.ignore_electrical_change, fz.Z809A_power],
    toZigbee: [tz.onoff],
    configure: (ieeeAddr, shepherd, coordinator, callback) => {
        const device = shepherd.find(ieeeAddr, 1);
        const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 1000, repChange: 0};
        const actions = [
	(cb) => device.bind('genOnOff', coordinator, cb),
            (cb) => device.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
            (cb) => device.report('haElectricalMeasurement', 'rmsVoltage', 10, 1000, 1, cb),
            (cb) => device.report('haElectricalMeasurement', 'rmsCurrent', 10, 1000, 1, cb),
            (cb) => device.report('haElectricalMeasurement', 'activePower', 10, 1000, 1, cb),
            (cb) => device.report('haElectricalMeasurement', 'powerFactor', 10, 1000, 1, cb),
        ];

        execute(device, actions, callback);
    },
},
```

Koenkk, 2019-12-07:
The api has been changed indeed, good example: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L3695

closed 2020-02-12 by stale[bot]