Koenkk/zigbee2mqtt #2468: zigbee2mqtt cc2531 communication lost to devices after restart domoticz

ariekanarie2005, 2019-12-05:
Bug Report
## What happened
After restart domoticz all paired devices loose connection.

##What did you expect to happen
after restart domoticz all devices manageable

## How to reproduce it (minimal and precise)
editing ip adress from 127.0.0.1 to ip adress domoticz under hardware domoticz zigbee2mqtt usb device causes advertising all paired devices from zigbee2mqtt bridge.
After this everyting seems te be normal.
When i reset domoticz again i have to edit under hardware zigbee2mqtt ip adress again from domoticz adress to localhost adres before all paired devices are published from zigbee2mqtt bridge in domoticz.

## Debug Info
zigbee2mqtt version: 1.71
CC253X firmware version:  CC2531_SOURCE_ROUTING_20190619.zip

ariekanarie2005, 2019-12-05:
log:
after restart domoticz (raspberry pi3)
 2019-12-05 17:06:04.777 Status: Domoticz V4.10717 (c)2012-2019 GizMoCuz
2019-12-05 17:06:04.798 Status: Build Hash: b38b49e5, Date: 2019-05-09 13:04:08
2019-12-05 17:06:04.799 Status: Startup Path: /home/pi/domoticz/
2019-12-05 17:06:05.329 Sunrise: 08:33:00 SunSet: 16:29:00
2019-12-05 17:06:05.329 Day length: 07:56:00 Sun at south: 12:31:00
2019-12-05 17:06:05.329 Civil twilight start: 07:52:00 Civil twilight end: 17:09:00
2019-12-05 17:06:05.329 Nautical twilight start: 07:09:00 Nautical twilight end: 17:53:00
2019-12-05 17:06:05.329 Astronomical twilight start: 06:28:00 Astronomical twilight end: 18:34:00
2019-12-05 17:06:05.910 Active notification Subsystems: telegram (1/13)
2019-12-05 17:06:05.899 Status: PluginSystem: Started, Python version '3.5.3'.
2019-12-05 17:06:05.957 Status: WebServer(HTTP) started on address: :: with port 8080
2019-12-05 17:06:05.976 Status: WebServer(SSL) started on address: :: with port 443
2019-12-05 17:06:05.980 Status: Camera: settings (re)loaded
2019-12-05 17:06:05.998 Status: RxQueue: queue worker started...
2019-12-05 17:06:05.997 Error: Exception starting shared server: bind: Address already in use
2019-12-05 17:06:07.999 Status: (Zigbee2mqtt) Started.
2019-12-05 17:06:07.999 Status: (yamaha receiver woonkamer) Started.
2019-12-05 17:06:07.999 Status: NefitEasy: Worker started...
2019-12-05 17:06:08.000 Status: (Eq3 max draadloze thermostaat) Started.
2019-12-05 17:06:08.000 Status: EventSystem: reset all events...
2019-12-05 17:06:08.049 Status: PluginSystem: Entering work loop.
2019-12-05 17:06:08.055 Status: EventSystem: reset all device statuses...
2019-12-05 17:06:08.949 Status: (Zigbee2mqtt) Entering work loop.
2019-12-05 17:06:08.950 Status: (Zigbee2mqtt) Initialized version 0.0.19
2019-12-05 17:06:09.121 Status: (yamaha receiver woonkamer) Entering work loop.
2019-12-05 17:06:09.121 Status: (yamaha receiver woonkamer) Initialized version 2.0.2, author 'thomasvillagers'
2019-12-05 17:06:09.541 Status: (Eq3 max draadloze thermostaat) Initialized version 0.6.6, author 'mvzut'
2019-12-05 17:06:09.564 Status: (Eq3 max draadloze thermostaat) Entering work loop.
2019-12-05 17:06:09.643 Status: Python EventSystem: Initalizing event module.
2019-12-05 17:06:09.643 Status: EventSystem: Started
2019-12-05 17:06:09.644 Status: EventSystem: Queue thread started...
2019-12-05 17:06:09.014 Error: NefitEasy: Error getting http data!
2019-12-05 17:06:09.016 Error: NefitEasy: Error getting http data!
2019-12-05 17:06:12.400 (Zigbee2mqtt) Connected to MQTT Server: 127.0.0.1:1883
2019-12-05 17:06:15.860 (Zigbee2mqtt) Zigbee2mqtt bridge is online
2019-12-05 17:06:16.011 (Zigbee2mqtt) Joining new devices is enabled on the zigbee bridge
2019-12-05 17:07:00.275 EventSystem: Event triggered: licht keuken spot 1_1
2019-12-05 17:07:00.276 EventSystem: Event triggered: licht keuken spot 2_1
2019-12-05 17:07:00.329 (Zigbee2mqtt) Device spot 1 keuken does not have registered zigbee2mqtt device
2019-12-05 17:07:00.329 (Zigbee2mqtt) Device spot 2 keuken does not have registered zigbee2mqtt device 
--------------------------------------------------------------------------------------------------------
after changing ip adress in hardware domoticz from localhost to ip adress domoticz:
 2019-12-05 17:07:44.791 Status: (Zigbee2mqtt) Stop directive received.
2019-12-05 17:07:44.791 Status: (Zigbee2mqtt) Exiting work loop.
2019-12-05 17:07:44.991 Status: (Zigbee2mqtt) Stopping threads.
2019-12-05 17:07:44.991 Status: (Zigbee2mqtt) Stopped.
2019-12-05 17:07:44.991 Status: (Zigbee2mqtt) Started.
2019-12-05 17:07:44.866 Error: (Zigbee2mqtt) Disconnected from MQTT Server: 127.0.0.1:1883
2019-12-05 17:07:45.410 (Zigbee2mqtt) Connected to MQTT Server: 192.168.2.78:1883
2019-12-05 17:07:45.511 (Zigbee2mqtt) Zigbee2mqtt bridge is online
2019-12-05 17:07:45.511 (Zigbee2mqtt) Joining new devices is enabled on the zigbee bridge
2019-12-05 17:07:45.563 (Zigbee2mqtt) Received available devices list from bridge
2019-12-05 17:07:45.563 (Zigbee2mqtt) Device RTCGQ11LM 0x00158d0002e34b3c (pir gang)
2019-12-05 17:07:45.563 (Zigbee2mqtt) Device MFKZQ01LM 0x00158d00027d9112 (cube)
2019-12-05 17:07:45.564 (Zigbee2mqtt) Device LED1623G12 0x000d6ffffeada2bb (lamp gang)
2019-12-05 17:07:45.564 (Zigbee2mqtt) Device AB3257001NJ 0x7cb03eaa0a012f7e (voortuin lamp)
2019-12-05 17:07:45.564 (Zigbee2mqtt) Device AB3257001NJ 0x7cb03eaa0a012f5b (garage lamp)
2019-12-05 17:07:45.564 (Zigbee2mqtt) Device AB3257001NJ 0x7cb03eaa0a014ad5 (wifi schakelen)
2019-12-05 17:07:45.564 (Zigbee2mqtt) Device RTCGQ11LM 0x00158d0002ebff8c (pir keuken)
2019-12-05 17:07:45.564 (Zigbee2mqtt) Device LED1650R5 0x90fd9ffffe1356fe (spot 1 keuken)
2019-12-05 17:07:45.564 (Zigbee2mqtt) Device LED1650R5 0x000d6ffffe270a86 (spot 2 keuken)
2019-12-05 17:07:45.615 (Zigbee2mqtt) Joining new devices is enabled on the zigbee bridge
2019-12-05 17:07:45.350 Status: (Zigbee2mqtt) Entering work loop.
2019-12-05 17:07:45.350 Status: (Zigbee2mqtt) Initialized version 0.0.19
2019-12-05 17:07:48.014 (nefit easy) Thermostat (ingestelde temp easy)
2019-12-05 17:07:48.018 (nefit easy) Temp (temp easy woonkamer)
2019-12-05 17:07:48.022 (nefit easy) General/Text (Boiler Status)
2019-12-05 17:07:48.257 (nefit easy) Temp (Flow Temperature nefit easy)
2019-12-05 17:07:48.461 (nefit easy) General/Text (Display Code)
2019-12-05 17:08:00.280 EventSystem: Event triggered: licht keuken spot 1_1
2019-12-05 17:08:00.280 EventSystem: Event triggered: licht keuken spot 2_1

Koenkk, 2019-12-07:
Can you still control the device by publishing to the MQTT topic: `zigbee2mqtt/0x000d6ffffeada2bb/set` payload: `TOGGLE`?

closed 2020-02-12 by stale[bot]