Koenkk/zigbee2mqtt #2470: Swann Motion detector 'SWO-MOS1PA'. Please, help me to add to Support

loglux, 2019-12-05:
I checked databasse.db and added this record to devices.js

```
{
    zigbeeModel: ['SWO-MOS1PA'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'SWO-MOS1PA', // Vendor model number, look on the device for a model number
    vendor: 'SwannONe', // Vendor of the device (only used for documentation and startup logging)
    description: 'Swann Motion and Temperature Sensor ', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    supports: 'motion and temperature', // Actions this device supports (only used for documentation)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
},
```
But now I stuck how to modify fromZigbee.js correctly.
Yeah, I have a message in the console:
```
Dec 05 13:39:59 littlehelp npm[10992]: zigbee2mqtt:debug 2019-12-05T13:39:59: Received Zigbee message from '0x00124b00063b5aa2', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":49,"extendedstatus":0}' from endpoint 1 with groupID 0
Dec 05 13:39:59 littlehelp npm[10992]: zigbee2mqtt:warn  2019-12-05T13:39:59: No converter available for 'SWO-MOS1PA' with cluster 'ssIasZone' and type 'commandStatusChangeNotification' and data '{"zonestatus":49,"extendedstatus":0}'
```

#https://github.com/Koenkk/zigbee2mqtt/issues/212

loglux, 2019-12-06:
A record from database.db`

{"id":6,"type":"EndDevice","ieeeAddr":"0x00124b00063b5aa2","nwkAddr":57159,"manufId":0,"manufName":"SwannONe","powerSource":"Battery","modelId":"SWO-MOS1PA","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,2,1,1280,64512],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"modelId":"SWO-MOS1PA","manufacturerName":"SwannONe","powerSource":3,"zclVersion":1,"hwVersion":1,"dateCode":"20160606        "}}},"binds":[]}},"hwVersion":1,"dateCode":"20160606        ","zclVersion":1,"interviewCompleted":true,"meta":{}}`

loglux, 2019-12-06:
I've changed in devices.js to 
```
{
    zigbeeModel: ['SWO-MOS1PA'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'SWO-MOS1PA', // Vendor model number, look on the device for a model number
    vendor: 'Swann', // Vendor of the device (only used for documentation and startup logging)
    description: 'Motion and temperature sensor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)

        supports: 'occupancy and temperature',
        fromZigbee: [
            fz.temperature, fz.iaszone_occupancy_2,
            fz.iaszone_occupancy_1, fz.battery_3V,
        ],
        toZigbee: [],
        meta: {configureKey: 1},
},
```
And got a result in my log:
Dec 06 20:10:10 littlehelp npm[27030]: zigbee2mqtt:info  2019-12-06T20:10:10: MQTT publish: topic 'zigbee2mqtt/SwannONe Motion Sensor', payload '{"occupancy":true,"tamper":false,"battery_low":false,"linkquality":63}'
Dec 06 20:11:25 littlehelp npm[27030]: zigbee2mqtt:info  2019-12-06T20:11:25: MQTT publish: topic 'zigbee2mqtt/SwannONe Motion Sensor', payload '{"occupancy":false,"tamper":false,"battery_low":false,"linkquality":65}'

loglux, 2019-12-06:
The log shows occupancy true and false. It's a kind of progress, isn't it?
Please, guys, help me.

Koenkk, 2019-12-07:
Looks good already, questions:
- Does the temperature actually work?
- You should use both `fz.iaszone_occupancy_2, fz.iaszone_occupancy_1`, try only using one of them and check with correctly published occpuancy true/false.

loglux, 2019-12-07:
>     * Does the temperature actually work?

No, it doesn't work with this configuration.

>     * You should use both `fz.iaszone_occupancy_2, fz.iaszone_occupancy_1`, try only using one of them and check with correctly published occupancy true/false.

When the only **fz.iaszone_occupancy_2** is used the sensor sends '**occupancy: false**' signal every time when it catches the motion. That's wrong,

```
ec 07 22:54:22 orangepiprime npm[601]: zigbee2mqtt:info  2019-12-07T22:54:22: MQTT publish: topic 'zigbee2mqtt/SwannONe Motion Sensor', payload '{"occupancy":false,"tamper":false,"battery_low":false,"linkquality":105}'
Dec 07 22:55:30 orangepiprime npm[601]: zigbee2mqtt:info  2019-12-07T22:55:30: MQTT publish: topic 'zigbee2mqtt/SwannONe Motion Sensor', payload '{"occupancy":false,"tamper":false,"battery_low":false,"linkquality":102}'
Dec 07 22:56:40 orangepiprime npm[601]: zigbee2mqtt:info  2019-12-07T22:56:40: MQTT publish: topic 'zigbee2mqtt/SwannONe Motion Sensor', payload '{"occupancy":false,"tamper":false,"battery_low":false,"linkquality":105}
```'
I was waiitng about 5 minutes and no other signal was sent (like occupancy: true etc.)

2. When both fz.iaszone_occupancy_1 and fz.iaszone_occupancy_2 are used, the sensors send occupancy: true when it see motion and occupancy: false after some time of inactivity (I mean no motion). 

3. When the only fz.iaszone_occupancy_1 parameter is active. It definitely sends a correct 'occupation: true' when it sees motion and 'occupation: false' after inactivity:

```
Dec 07 23:05:16 orangepiprime npm[5960]: zigbee2mqtt:info  2019-12-07T23:05:16: MQTT publish: topic 'zigbee2mqtt/SwannONe Motion Sensor', payload '{"occupancy":true,"tamper":false,"battery_low":false,"linkquality":102}'
Dec 07 23:06:26 orangepiprime npm[5960]: zigbee2mqtt:info  2019-12-07T23:06:26: MQTT publish: topic 'zigbee2mqtt/SwannONe Motion Sensor', payload '{"occupancy":false,"tamper":false,"battery_low":false,"linkquality":105}'

```

loglux, 2019-12-07:
So, it seems that the  fz.iaszone_occupancy_1 is enough. But I don't understand how it actually works.

Koenkk, 2019-12-08:
Ok, that is good, only thing is temperature, please try with:

```js
{
    zigbeeModel: ['SWO-MOS1PA'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'SWO-MOS1PA', // Vendor model number, look on the device for a model number
    vendor: 'Swann', // Vendor of the device (only used for documentation and startup logging)
    description: 'Motion and temperature sensor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    supports: 'occupancy and temperature',
    fromZigbee: [
        fz.temperature, fz.iaszone_occupancy_1, fz.battery_3V,
    ],
    toZigbee: [],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['msTemperatureMeasurement']);
        await configureReporting.temperature(endpoint);
    },
},
```

You might need to re-pair the device.

loglux, 2019-12-08:
Now I have after restart:

```
Dec 08 13:13:02 orangepiprime npm[10293]: zigbee2mqtt:error 2019-12-08T13:13:02: Failed to configure 'SwannONe Motion Sensor', attempt 1 (Error: AREQ - ZDO - bindRsp after 10000ms
Dec 08 13:13:02 orangepiprime npm[10293]:     at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
Dec 08 13:13:02 orangepiprime npm[10293]:     at ontimeout (timers.js:436:11)
Dec 08 13:13:02 orangepiprime npm[10293]:     at tryOnTimeout (timers.js:300:5)
Dec 08 13:13:02 orangepiprime npm[10293]:     at listOnTimeout (timers.js:263:5)
Dec 08 13:13:02 orangepiprime npm[10293]:     at Timer.processTimers (timers.js:223:10))
```
Then I'll try to re pair it

loglux, 2019-12-08:
> You might need to re-pair the device.

How can I do it properly?
Just reset the device or I must delete a record from database.db?

Koenkk, 2019-12-08:
The configure fails because the device is sleeping, try waking it up right after starting zigbee2mqtt by pressing a button on it.

loglux, 2019-12-08:
Something happened, waiting for results now

```
Dec 08 13:25:14 orangepiprime npm[10293]: zigbee2mqtt:info  2019-12-08T13:25:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"SwannONe Motion Sensor"}}'
Dec 08 13:25:14 orangepiprime npm[10293]: zigbee2mqtt:info  2019-12-08T13:25:14: Starting interview of 'SwannONe Motion Sensor'
Dec 08 13:25:14 orangepiprime npm[10293]: zigbee2mqtt:info  2019-12-08T13:25:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"SwannONe Motion Sensor"}}'
Dec 08 13:25:19 orangepiprime npm[10293]: zigbee2mqtt:info  2019-12-08T13:25:19: MQTT publish: topic 'zigbee2mqtt/SwannONe Motion Sensor', payload '{"occupancy":true,"tamper":false,"battery_low":false,"linkquality":89}'
Dec 08 13:25:19 orangepiprime npm[10293]: zigbee2mqtt:info  2019-12-08T13:25:19: Successfully interviewed 'SwannONe Motion Sensor', device has successfully been paired
Dec 08 13:25:19 orangepiprime npm[10293]: zigbee2mqtt:info  2019-12-08T13:25:19: Device 'SwannONe Motion Sensor' is supported, identified as: Swann Motion and temperature sensor (SWO-MOS1PA)
Dec 08 13:25:19 orangepiprime npm[10293]: zigbee2mqtt:info  2019-12-08T13:25:19: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"SwannONe Motion Sensor","model":"SWO-MOS1PA","vendor":"Swann","description":"Motion and temperature sensor","supported":true}}'
```

loglux, 2019-12-08:
There are three deferent blinking red signals on the sensor itself. 
All red LEDs at the top
One led blink at the top
One led blink at the middle of the sensor.

loglux, 2019-12-08:
Nothing had happened so I tried to reset again.
It sent me occupancy: false signal being in my hands

Dec 08 13:36:35 orangepiprime npm[10293]: zigbee2mqtt:info  2019-12-08T13:36:35: Device 'SwannONe Motion Sensor' is supported, identified as: Swann Motion and temperature sensor (SWO-MOS1PA)
Dec 08 13:36:35 orangepiprime npm[10293]: zigbee2mqtt:info  2019-12-08T13:36:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"SwannONe Motion Sensor","model":"SWO-MOS1PA","vendor":"Swann","description":"Motion and temperature sensor","supported":true}}'
Dec 08 13:36:35 orangepiprime npm[10293]: zigbee2mqtt:info  2019-12-08T13:36:35: MQTT publish: topic 'zigbee2mqtt/SwannONe Motion Sensor', payload '{"occupancy":false,"tamper":false,"battery_low":false,"linkquality":84}'

loglux, 2019-12-08:
I'll leave it alone for a while, then check again after 20-30 minutes

loglux, 2019-12-08:
I can see full LEDs blinks but there is nothing in the log.

loglux, 2019-12-08:
So the sensor is working but zigbee2mqtt can't properly interpret the signals. I don't see any trace of activity in the log.

loglux, 2019-12-08:
Restarted the service and got the output:

```
roduct":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":true}'
Dec 08 13:48:50 orangepiprime npm[12830]: zigbee2mqtt:error 2019-12-08T13:48:50: Failed to configure 'SwannONe Motion Sensor', attempt 1 (Error: AREQ - ZDO - bindRsp after 10000ms
Dec 08 13:48:50 orangepiprime npm[12830]:     at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
Dec 08 13:48:50 orangepiprime npm[12830]:     at ontimeout (timers.js:436:11)
Dec 08 13:48:50 orangepiprime npm[12830]:     at tryOnTimeout (timers.js:300:5)
Dec 08 13:48:50 orangepiprime npm[12830]:     at listOnTimeout (timers.js:263:5)
Dec 08 13:48:50 orangepiprime npm[12830]:     at Timer.processTimers (timers.js:223:10))
```

Koenkk, 2019-12-08:
Are you sure that it does support temperature reporting?

loglux, 2019-12-08:
Yeah, It does. 
https://www.swann.com/uk/swo-mos1pa

loglux, 2019-12-08:
P.S.
Not sure is it useful info but:
Protocol Zigbee HA 1.2

loglux, 2019-12-09:
But what this device doesn't have, it doesn't have any tamper.
So, the report is not 100% correct.
`Dec 09 00:20:32 orangepiprime npm[15287]: zigbee2mqtt:info  2019-12-09T00:20:32: MQTT publish: topic 'zigbee2mqtt/SwannONe Motion Sensor', payload '{"occupancy":true,"tamper":false,"battery_low":false,"linkquality":94}'
`

loglux, 2019-12-10:
We could at least reach a compromise adding the devices as partly-supported, continuing the research for fully-functional support. In this case, I could ask the Domoticz plugin developer to add the support to and use these motion sensors. 

This sensor shows good activity and in contrast with Xiaomi and IKEA sensors it seems does not have 'blind time' after sending an 'occupancy: false' signal.

Koenkk, 2019-12-11:
Let's give the temperature one more try, please try with the config below and re-pair the device. If you get an error while configuring make sure you wake up the device (by pressing some buttons on it) when start zigbee2mqtt.

```js
{
    zigbeeModel: ['SWO-MOS1PA'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'SWO-MOS1PA', // Vendor model number, look on the device for a model number
    vendor: 'Swann', // Vendor of the device (only used for documentation and startup logging)
    description: 'Motion and temperature sensor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    supports: 'occupancy and temperature',
    fromZigbee: [
        fz.temperature, fz.iaszone_occupancy_1, fz.battery_3V,
    ],
    toZigbee: [],
    meta: {configureKey: 2},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['msTemperatureMeasurement']);
        const payload = [{
            attribute: 'measuredValue',
            minimumReportInterval: 0,
            maximumReportInterval: 3,
            reportableChange: 0,
        }];
        await endpoint.configureReporting('msTemperatureMeasurement', payload);
    },
},
```

Ryanmt-Deep3, 2020-01-03:
Hi @Koenkk 

I have the same device too so I have tested the file above.

I reset the device a few times and tried triggering it etc to get to configure but I get

```
zigbee2mqtt:error 2020-01-03 16:06:40: Failed to configure '0x00124b0006286315', attempt 3 (Error: Timeout - 48151 - 1 - 3 - 1026 - 7 after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/zigbee2mqtt-1.8.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10))
zigbee2mqtt:info  2020-01-03 16:07:23: MQTT publish: topic 'zigbee2mqtt/0x00124b0006286315', payload '{"occupancy":false,"tamper":false,"battery_low":false,"linkquality":89}'
zigbee2mqtt:info  2020-01-03 16:07:37: MQTT publish: topic 'zigbee2mqtt/0x00124b0006286315', payload '{"occupancy":true,"tamper":false,"battery_low":false,"linkquality":115}'
zigbee2mqtt:info  2020-01-03 16:07:44: MQTT publish: topic 'zigbee2mqtt/0x00124b0006286315', payload '{"occupancy":false,"tamper":false,"battery_low":false,"linkquality":113}'
```

Ryanmt-Deep3, 2020-01-03:
Also worth noting that the SWO-WDS1PA which is currently in the supported device list is also supposed to provide temperature according to Swann. I have 2 of each so am happy to aid debugging as much as I can.

loglux, 2020-01-29:
I apologise for long delay. 

I have tested the configuration. It doesn't work. And motion detection itself is working with some delay I suppose. 

So we can easy to get motion signals only for now but the temperature is still a miracle.

loglux, 2020-01-29:
I'd like to speed up the process of support because I have a bunch of these devices and it's a good sensor, very quick.

loglux, 2020-01-29:
My proposal is still the same. We could spit the process of Support into two stages. 
1. Implement motion on/off, that is very easy to do
2. Continue the research to disclose all other signals.  I could set up a separate stick for that on the separate system.

Koenkk, 2020-01-30:
So with https://github.com/Koenkk/zigbee2mqtt/issues/2470#issuecomment-564696952 occupancy is working?

loglux, 2020-02-18:
> So with [#2470 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/2470#issuecomment-564696952) occupancy is working?

It's working but sending 1 signal from 2. 
Actually, after we put additional parameters bellow 
```
    toZigbee: [],
    meta: {configureKey: 1},
```
The motion sensor is not always sending a signal or maybe we have a delay. I'm not 100% sure what is wrong.

Ryanmt-Deep3, 2020-04-14:
What do we need to get a PR together for this @Koenkk ? 

Do you want me to verify the issue on my side @loglux ? Can you share your config?

Koenkk, 2020-04-14:
@Ryanmt-Deep3 verify that the code from https://github.com/Koenkk/zigbee2mqtt/issues/2470#issuecomment-564696952 is working

loglux, 2020-04-14:
> @Ryanmt-Deep3 verify that the code from #2470 (comment) is working
I would say it's working but not as was expected and delay or miss alerts.

I use a docker now for zigbee2mqtt so it's not possible to make all these tests now.
I need time to set up another server with another stick and another network just for this sensor.
I'm still busy at work despite all this virus mess. But I'll do it as soon as I can.

Ryanmt-Deep3, 2020-04-15:
I just edit the docker image for short term testing like this. 

I will do it tonight and report back with some logs.

MattEng0, 2020-06-05:
Apologies for the block of code. Got this same sensor today and after getting to the same point with regards to the motion detecting capabilities of the device I'm now stuck with getting the temp details from it. This is the errors I'm getting from the log:
```
zigbee2mqtt:error 2020-06-05 15:53:21: Failed to configure 'Swann Sensor', attempt 1 (Error: Bind 0x00124b00063c5146/1 msTemperatureMeasurement from '0x00124b00014d96ff/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.13.1/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:244:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.13.1/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
zigbee2mqtt:debug 2020-06-05 15:58:08: Saving state to file /share/zigbee2mqtt/state.json
zigbee2mqtt:debug 2020-06-05 15:59:38: Received Zigbee message from '0x00158d00047e68d9', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-06-05 15:59:38: MQTT publish: topic 'zigbee2mqtt/0x00158d00047e68d9', payload '{"contact":false,"linkquality":55,"battery":100,"voltage":3025}'
zigbee2mqtt:debug 2020-06-05 16:00:03: Received Zigbee message from 'Swann Sensor', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":49,"extendedstatus":0}' from endpoint 1 with groupID 0
zigbee2mqtt:error 2020-06-05 16:00:03: Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
    at /zigbee2mqtt-1.13.1/lib/extension/receive.js:105:42
    at Array.filter (<anonymous>)
    at Receive.onZigbeeEvent (/zigbee2mqtt-1.13.1/lib/extension/receive.js:104:65)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.13.1/lib/controller.js:350:44)
    at Controller.onZigbeeEvent (/zigbee2mqtt-1.13.1/lib/controller.js:236:14)
    at Zigbee.emit (events.js:223:5)
    at Controller.<anonymous> (/zigbee2mqtt-1.13.1/lib/zigbee.js:67:52)
    at Controller.emit (events.js:223:5)
    at Controller.<anonymous> (/zigbee2mqtt-1.13.1/node_modules/zigbee-herdsman/dist/controller/controller.js:496:22)
    at Generator.next (<anonymous>))
zigbee2mqtt:info  2020-06-05 16:00:03: Configuring 'Swann Sensor'
zigbee2mqtt:error 2020-06-05 16:01:08: Failed to configure 'Swann Sensor', attempt 2 (Error: ConfigureReporting 0x00124b00063c5146/1 msTemperatureMeasurement([{"attribute":"measuredValue","minimumReportInterval":0,"maximumReportInterval":3,"reportableChange":0}], {"timeout":10000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Data request failed with error: 'MAC transaction expired' (240))
    at Endpoint.<anonymous> (/zigbee2mqtt-1.13.1/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:326:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.13.1/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
zigbee2mqtt:debug 2020-06-05 16:01:13: Received Zigbee message from 'Swann Sensor', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":48,"extendedstatus":0}' from endpoint 1 with groupID 0
zigbee2mqtt:error 2020-06-05 16:01:13: Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
    at /zigbee2mqtt-1.13.1/lib/extension/receive.js:105:42
    at Array.filter (<anonymous>)
    at Receive.onZigbeeEvent (/zigbee2mqtt-1.13.1/lib/extension/receive.js:104:65)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.13.1/lib/controller.js:350:44)
    at Controller.onZigbeeEvent (/zigbee2mqtt-1.13.1/lib/controller.js:236:14)
    at Zigbee.emit (events.js:223:5)
    at Controller.<anonymous> (/zigbee2mqtt-1.13.1/lib/zigbee.js:67:52)
    at Controller.emit (events.js:223:5)
    at Controller.<anonymous> (/zigbee2mqtt-1.13.1/node_modules/zigbee-herdsman/dist/controller/controller.js:496:22)
    at Generator.next (<anonymous>))
zigbee2mqtt:info  2020-06-05 16:01:13: Configuring 'Swann Sensor'
zigbee2mqtt:debug 2020-06-05 16:02:16: Received Zigbee message from 'Swann Sensor', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":49,"extendedstatus":0}' from endpoint 1 with groupID 0
zigbee2mqtt:error 2020-06-05 16:02:16: Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
    at /zigbee2mqtt-1.13.1/lib/extension/receive.js:105:42
    at Array.filter (<anonymous>)
    at Receive.onZigbeeEvent (/zigbee2mqtt-1.13.1/lib/extension/receive.js:104:65)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.13.1/lib/controller.js:350:44)
    at Controller.onZigbeeEvent (/zigbee2mqtt-1.13.1/lib/controller.js:236:14)
    at Zigbee.emit (events.js:223:5)
    at Controller.<anonymous> (/zigbee2mqtt-1.13.1/lib/zigbee.js:67:52)
    at Controller.emit (events.js:223:5)
    at Controller.<anonymous> (/zigbee2mqtt-1.13.1/node_modules/zigbee-herdsman/dist/controller/controller.js:496:22)
    at Generator.next (<anonymous>))
zigbee2mqtt:error 2020-06-05 16:02:26: Failed to configure 'Swann Sensor', attempt 3 (Error: ConfigureReporting 0x00124b00063c5146/1 msTemperatureMeasurement([{"attribute":"measuredValue","minimumReportInterval":0,"maximumReportInterval":3,"reportableChange":0}], {"timeout":10000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 9306 - 1 - 6 - 1026 - 7 after 10000ms)
[... cut ...]
    at Controller.callExtensionMethod (/zigbee2mqtt-1.13.1/lib/controller.js:350:44)
    at Controller.onZigbeeEvent (/zigbee2mqtt-1.13.1/lib/controller.js:236:14)
    at Zigbee.emit (events.js:223:5)
    at Controller.<anonymous> (/zigbee2mqtt-1.13.1/lib/zigbee.js:67:52)
    at Controller.emit (events.js:223:5)
    at Controller.<anonymous> (/zigbee2mqtt-1.13.1/node_modules/zigbee-herdsman/dist/controller/controller.js:496:22)
    at Generator.next (<anonymous>))
zigbee2mqtt:debug 2020-06-05 16:08:08: Saving state to file /share/zigbee2mqtt/state.json
```

loglux, 2020-07-19:
Please, reactivate the subject.
I have a bunch of these devices. They are good doing their primary function. Let them work.
As I offered before, we could implement basic motion detection functions and then could continue to deep in temperature and etc, if it possible (maybe not).

MattEng0, 2020-07-19:
> 
> 
> Please, reactivate the subject.
> I have a bunch of these devices. They are good doing their primary function. Let them work.
> As I offered before, we could implement basic motion detection functions and then could continue to deep in temperature and etc, if it possible (maybe not).

I have add the basic feature of this device and its been pushed to the main distrubution. As you have said it only works for motion sensing at the moment, I just haven't gotten around to properly sniffing the traffic to see if I can get the temperature detection to work. 

It's on my list of things to do but being properly back to work has meant I have had to put it on the back burner. 

Do you have the SWANN smart home hub?

#3678 - for reference to the pushed features.

loglux, 2020-07-19:
> Do you have the SWANN smart home hub?

No I haven't. I just bought a Swann smart home set with their rubbish gateway in Scan for a very cheap price on last Christmas.

loglux, 2020-08-19:
I spotted one thing. The sensor reacts in time but the signal is sent with a delay. I can go in and out of the kitchen and the lights are on after I actually left.

loglux, 2020-08-19:
> Do you have the SWANN smart home hub?

I have but don't use it. SWANN is spying on you through the hub.

closed 2020-08-27 by github-actions[bot]