Koenkk/zigbee2mqtt #2471: Z2M Devices Dissapear after HA Restart

carpenike, 2019-12-05:
# Bug Report
## What happened
First, my environment:

1. Separate, highly available MQTT broker located in a K8S cluster. Both HASS and Z2M use this broker:

[image]

2. HASS.IO installed with Z2M on the same box. Z2M configuration here:

```
{
  "data_path": "/share/zigbee2mqtt",
  "devices": "devices.yaml",
  "groups": "groups.yaml",
  "homeassistant": true,
  "permit_join": false,
  "mqtt": {
    "base_topic": "zigbee2mqtt",
    "server": "mqtts://<MQTTServer>:8883",
    "user": "username",
    "password": "password",
    "reject_unauthorized": "false"
  },
  "serial": {
    "port": "/dev/ttyACM0"
  },
  "advanced": {
    "pan_id": 6754,
    "channel": 11,
    "network_key": [
      1,
      3,
      5,
      7,
      9,
      11,
      13,
      15,
      0,
      2,
      4,
      6,
      8,
      10,
      12,
      13
    ],
    "availability_blacklist": [],
    "report": "true"
  },
  "ban": [],
  "whitelist": [],
  "queue": {},
  "socat": {
    "enabled": false,
    "master": "pty,raw,echo=0,link=/dev/ttyZ2M,mode=777",
    "slave": "tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5",
    "restartdelay": 1,
    "initialdelay": 1,
    "options": "-d -d",
    "log": false
  }
}
```
Devices located here:

[image]

When Z2M starts up, it correctly loads the devices into the homeassistant/# topic and devices are created in HA. However, if HA restarts, the devices go into an unavailable state, even though Z2M continues to make updates to the zigbee2mqtt/# topic.
[... 49 lines cut ...]
zigbee2mqtt:info  2019-12-05T14:01:17: MQTT publish: topic 'homeassistant/sensor/0x00178801033d0a48/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/sunroom_ceiling_light_1","json_attributes_topic":"zigbee2mqtt/sunroom_ceiling_light_1","name":"sunroom_ceiling_light_1_linkquality","unique_id":"0x00178801033d0a48_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00178801033d0a48"],"name":"sunroom_ceiling_light_1","sw_version":"Zigbee2mqtt 1.7.1","model":"Hue white A60 bulb E27 (9290011370)","manufacturer":"Philips"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2019-12-05T14:01:17: MQTT publish: topic 'homeassistant/light/0x90fd9ffffe4ba263/light/config', payload '{"brightness":true,"color_temp":true,"schema":"json","command_topic":"zigbee2mqtt/livingroom_couch_light/set","state_topic":"zigbee2mqtt/livingroom_couch_light","name":"livingroom_couch_light_light","unique_id":"0x90fd9ffffe4ba263_light_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x90fd9ffffe4ba263"],"name":"livingroom_couch_light","sw_version":"Zigbee2mqtt 1.7.1","model":"TRADFRI LED bulb E26/E27 950 lumen, dimmable, white spectrum, clear (LED1546G12)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2019-12-05T14:01:17: MQTT publish: topic 'homeassistant/sensor/0x90fd9ffffe4ba263/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/livingroom_couch_light","json_attributes_topic":"zigbee2mqtt/livingroom_couch_light","name":"livingroom_couch_light_linkquality","unique_id":"0x90fd9ffffe4ba263_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x90fd9ffffe4ba263"],"name":"livingroom_couch_light","sw_version":"Zigbee2mqtt 1.7.1","model":"TRADFRI LED bulb E26/E27 950 lumen, dimmable, white spectrum, clear (LED1546G12)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2019-12-05T14:01:17: MQTT publish: topic 'homeassistant/light/0x001788010349e7b3/light/config', payload '{"brightness":true,"schema":"json","command_topic":"zigbee2mqtt/laundryroom_floor_light_1/set","state_topic":"zigbee2mqtt/laundryroom_floor_light_1","name":"laundryroom_floor_light_1_light","unique_id":"0x001788010349e7b3_light_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x001788010349e7b3"],"name":"laundryroom_floor_light_1","sw_version":"Zigbee2mqtt 1.7.1","model":"Hue white A60 bulb E27 (9290011370)","manufacturer":"Philips"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2019-12-05T14:01:17: MQTT publish: topic 'homeassistant/sensor/0x001788010349e7b3/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/laundryroom_floor_light_1","json_attributes_topic":"zigbee2mqtt/laundryroom_floor_light_1","name":"laundryroom_floor_light_1_linkquality","unique_id":"0x001788010349e7b3_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x001788010349e7b3"],"name":"laundryroom_floor_light_1","sw_version":"Zigbee2mqtt 1.7.1","model":"Hue white A60 bulb E27 (9290011370)","manufacturer":"Philips"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2019-12-05T14:01:17: MQTT publish: topic 'homeassistant/light/0x00178801030734d3/light/config', payload '{"brightness":true,"schema":"json","command_topic":"zigbee2mqtt/laundryroom_floor_light_2/set","state_topic":"zigbee2mqtt/laundryroom_floor_light_2","name":"laundryroom_floor_light_2_light","unique_id":"0x00178801030734d3_light_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00178801030734d3"],"name":"laundryroom_floor_light_2","sw_version":"Zigbee2mqtt 1.7.1","model":"Hue white A60 bulb E27 (9290011370)","manufacturer":"Philips"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2019-12-05T14:01:17: MQTT publish: topic 'homeassistant/sensor/0x00178801030734d3/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/laundryroom_floor_light_2","json_attributes_topic":"zigbee2mqtt/laundryroom_floor_light_2","name":"laundryroom_floor_light_2_linkquality","unique_id":"0x00178801030734d3_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00178801030734d3"],"name":"laundryroom_floor_light_2","sw_version":"Zigbee2mqtt 1.7.1","model":"Hue white A60 bulb E27 (9290011370)","manufacturer":"Philips"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2019-12-05T14:01:17: MQTT publish: topic 'homeassistant/binary_sensor/0x00158d0003f2705c/contact/config', payload '{"payload_on":false,"payload_off":true,"value_template":"{{ value_json.contact }}","device_class":"door","state_topic":"zigbee2mqtt/sunroom_door","json_attributes_topic":"zigbee2mqtt/sunroom_door","name":"sunroom_door_contact","unique_id":"0x00158d0003f2705c_contact_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d0003f2705c"],"name":"sunroom_door","sw_version":"Zigbee2mqtt 1.7.1","model":"Aqara door & window contact sensor (MCCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2019-12-05T14:01:17: MQTT publish: topic 'homeassistant/sensor/0x00158d0003f2705c/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/sunroom_door","json_attributes_topic":"zigbee2mqtt/sunroom_door","name":"sunroom_door_battery","unique_id":"0x00158d0003f2705c_battery_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d0003f2705c"],"name":"sunroom_door","sw_version":"Zigbee2mqtt 1.7.1","model":"Aqara door & window contact sensor (MCCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2019-12-05T14:01:17: MQTT publish: topic 'homeassistant/sensor/0x00158d0003f2705c/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/sunroom_door","json_attributes_topic":"zigbee2mqtt/sunroom_door","name":"sunroom_door_linkquality","unique_id":"0x00158d0003f2705c_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d0003f2705c"],"name":"sunroom_door","sw_version":"Zigbee2mqtt 1.7.1","model":"Aqara door & window contact sensor (MCCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2019-12-05T14:01:17: MQTT publish: topic 'homeassistant/binary_sensor/0x00158d0003f1bdfb/contact/config', payload '{"payload_on":false,"payload_off":true,"value_template":"{{ value_json.contact }}","device_class":"door","state_topic":"zigbee2mqtt/front_door","json_attributes_topic":"zigbee2mqtt/front_door","name":"front_door_contact","unique_id":"0x00158d0003f1bdfb_contact_zigbee2mqtt","dev [... cut ...] "device":{"identifiers":["zigbee2mqtt_0x00158d0003f1bdfb"],"name":"front_door","sw_version":"Zigbee2mqtt 1.7.1","model":"Aqara door & window contact sensor (MCCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
```

## What did you expect to happen
HA/Z2M should re-populate the devices after HA restarts

## How to reproduce it (minimal and precise)
1. Enable HA autodiscovery
2. Restart HA

## Debug Info
zigbee2mqtt version: 1.7.1
CC253X firmware version: 20190608

Koenkk, 2019-12-07:
Do you use the configuration under 'MQTT discovery'? https://www.zigbee2mqtt.io/integration/home_assistant.html

carpenike, 2019-12-07:
> Do you use the configuration under 'MQTT discovery'? https://www.zigbee2mqtt.io/integration/home_assistant.html

Yes sir. Discovery does work as expected initially, but after a restart of home assistant it’s like it stops looking for the devices in zigbee2mqtt/#

Koenkk, 2019-12-08:
I still don't understand what the issue is. What do you mean with *looking for the devices in zigbee2mqtt/#*? If the devices completely disappear from Home Assistant this is not a zigbee2mqtt bug?

carpenike, 2019-12-08:
You may very well be correct. My understanding of the flow is the below, can you correct me if I'm wrong?

1. Z2M publishes a message to homeassistant/# matching the required autodiscovery format that Home Assistant expects. Message looks like the below:

```
zigbee2mqtt:info  2019-12-05T14:01:17: MQTT publish: topic 'homeassistant/sensor/0x00158d0003f2705c/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/sunroom_door","json_attributes_topic":"zigbee2mqtt/sunroom_door","name":"sunroom_door_linkquality","unique_id":"0x00158d0003f2705c_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d0003f2705c"],"name":"sunroom_door","sw_version":"Zigbee2mqtt 1.7.1","model":"Aqara door & window contact sensor (MCCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
```

2. Within the above message is the state topics that HomeAssistant should be looking for updates on the device. In the case above, `state_topic":"zigbee2mqtt/sunroom_door`, and similar.

3. When HomeAssistant restarts, it sends an 'offline' Will message to the hass/status topic at stop, and an 'online' Birth message when it starts back up again. This causes Z2M to re-publish the last state for each device on the state topic located within zigbee2mqtt/#. However, no autodiscovery messages are re-published to the homeassistant/# topic.

4. In my case, HomeAssistant now sees the devices as Unavailable, even though Z2M continues publishing messages to their state topics within zigbee2mqtt/#.

Is this expected behavior and a problem on the HASS side?

Koenkk, 2019-12-09:
> However, no autodiscovery messages are re-published to the homeassistant/# topic.

This shouldn't be needed on every home assistant restart, home assistant stores the devices in the `.storage` folder.

> In my case, HomeAssistant now sees the devices as Unavailable, even though Z2M continues publishing messages to their state topics within zigbee2mqtt/#.

With unavailable you mean greyed out? Do you use the availability feature? https://www.zigbee2mqtt.io/information/availability.html, If so please disable it.

carpenike, 2019-12-10:
Is availability enabled by default? The default configuration doesn't have it, unless `report: true` also turns it on.

  "advanced": {
    "pan_id": 6754,
    "channel": 11,
    "network_key": [
      1,
      3,
      5,
      7,
      9,
      11,
      13,
      15,
      0,
      2,
      4,
      6,
      8,
      10,
      12,
      13
    ],
    "availability_blacklist": [],
    "report": "true"
  },

carpenike, 2019-12-10:
OK, modified the configuration to look like the below, forcing availability to be disabled via `"availability_timeout": 0`:

```
{
  "data_path": "/share/zigbee2mqtt",
  "devices": "devices.yaml",
  "groups": "groups.yaml",
  "homeassistant": true,
  "permit_join": false,
  "mqtt": {
    "base_topic": "zigbee2mqtt",
    "server": "mqtts://mqtt.<address>.net:8883",
    "user": "username",
    "password": "password",
    "reject_unauthorized": "false"
  },
  "serial": {
    "port": "/dev/ttyACM0"
  },
  "advanced": {
    "pan_id": 6754,
    "channel": 11,
    "network_key": [
      1,
      3,
      5,
      7,
      9,
      11,
      13,
      15,
      0,
      2,
      4,
      6,
      8,
      10,
      12,
      13
    ],
    "availability_blacklist": [],
    "availability_timeout": 0
  },
  "ban": [],
  "whitelist": [],
  "queue": {},
  "socat": {
    "enabled": false,
    "master": "pty,raw,echo=0,link=/dev/ttyZ2M,mode=777",
    "slave": "tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5",
    "restartdelay": 1,
    "initialdelay": 1,
    "options": "-d -d",
    "log": false
  }
}
```

Logs of a successful startup here:

```
ryan@hass:/usr/share/hassio/share/zigbee2mqtt$ docker logs -f addon_7ad98f9c_zigbee2mqtt
./run.sh: line 16: [Info] Configuration backup found in /share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No such file or directory
[Info] Socat is DISABLED and not started
2019-12-10T14:07:50: PM2 log: Launching in no daemon mode
2019-12-10T14:07:50: PM2 log: App [npm:0] starting in -fork mode-
2019-12-10T14:07:50: PM2 log: App [npm:0] online
> zigbee2mqtt@1.7.1 start /zigbee2mqtt-1.7.1
> node index.js
zigbee2mqtt:info  2019-12-10T19:07:56: Logging to directory: '/share/zigbee2mqtt/log/2019-12-10.14-07-55'
[... 44 lines cut ...]
zigbee2mqtt:info  2019-12-10T19:08:03: MQTT publish: topic 'zigbee2mqtt/bridge/config/devices', payload '[{"ieeeAddr":"0x00124b0014d9e1be","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack12","dateCode":"20190608","lastSeen":1576004883538},{"ieeeAddr":"0x000b57fffe902c0b","type":"Router","networkAddress":46586,"model":"LED1649C5","friendly_name":"master_bathroom_sink_light","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelID":"TRADFRI bulb E12 W op/ch 400lm","hardwareVersion":1,"softwareBuildID":"1.2.214","dateCode":"20170302","lastSeen":null},{"ieeeAddr":"0x00178801033d3c67","type":"Router","networkAddress":49583,"model":"9290011370","friendly_name":"sunroom_ceiling_light_2","manufacturerID":4107,"manufacturerName":"Philips","powerSource":"Mains (single phase)","modelID":"LWB014","hardwareVersion":1,"softwareBuildID":"1.23.0_r20156","dateCode":"20170311","lastSeen":null},{"ieeeAddr":"0x00178801033d0a48","type":"Router","networkAddress":663,"model":"9290011370","friendly_name":"sunroom_ceiling_light_1","manufacturerID":4107,"manufacturerName":"Philips","powerSource":"Mains (single phase)","modelID":"LWB014","hardwareVersion":1,"softwareBuildID":"1.23.0_r20156","dateCode":"20170311","lastSeen":null},{"ieeeAddr":"0x90fd9ffffe4ba263","type":"Router","networkAddress":59180,"model":"LED1546G12","friendly_name":"livingroom_couch_light","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelID":"TRADFRI bulb E26 WS clear 950lm","hardwareVersion":1,"softwareBuildID":"1.2.217","dateCode":"20170331","lastSeen":null},{"ieeeAddr":"0x001788010349e7b3","type":"Router","networkAddress":59051,"model":"9290011370","friendly_name":"laundryroom_floor_light_1","manufacturerID":4107,"manufacturerName":"Philips","powerSource":"Mains (single phase)","modelID":"LWB014","hardwareVersion":1,"softwareBuildID":"1.23.0_r20156","dateCode":"20170311","lastSeen":null},{"ieeeAddr":"0x00178801030734d3","type":"Router","networkAddress":19198,"model":"9290011370","friendly_name":"laundryroom_floor_light_2","manufacturerID":4107,"manufacturerName":"Philips","powerSource":"Mains (single phase)","modelID":"LWB014","hardwareVersion":1,"softwareBuildID":"1.23.0_r20156","dateCode":"20170311","lastSeen":null},{"ieeeAddr":"0x00158d0003f2705c","type":"EndDevice","networkAddress":9947,"model":"MCCGQ11LM","friendly_name":"sunroom_door","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_magnet.aq2","hardwareVersion":2,"softwareBuildID":"3000-0001","dateCode":"20161128","lastSeen":null},{"ieeeAddr":"0x00158d0003f1bdfb","type":"EndDevice","networkAddress":14408,"model":"MCCGQ11LM","friendly_name":"front_door","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_magnet.aq2","lastSeen":null}]'
zigbee2mqtt:info  2019-12-10T19:08:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[{"ieeeAddr":"0x00124b0014d9e1be","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack12","dateCode":"20190608","lastSeen":1576004883556},{"ieeeAddr":"0x000b57fffe902c0b","type":"Router","networkAddress":46586,"model":"LED1649C5","friendly_name":"master_bathroom_sink_light","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelID":"TRADFRI bulb E12 W op/ch 400lm","hardwareVersion":1,"softwareBuildID":"1.2.214","dateCode":"20170302","lastSeen":null},{"ieeeAddr":"0x00178801033d3c67","type":"Router","networkAddress":49583,"model":"9290011370"," [... cut ...]
zigbee2mqtt:info  2019-12-10T19:14:57: MQTT publish: topic 'zigbee2mqtt/laundryroom_floor_light_2', payload '{"state":"OFF","linkquality":86,"brightness":254}'
zigbee2mqtt:info  2019-12-10T19:14:57: MQTT publish: topic 'zigbee2mqtt/sunroom_door', payload '{"contact":true,"linkquality":28,"battery":86,"voltage":2975}'
zigbee2mqtt:info  2019-12-10T19:14:57: MQTT publish: topic 'zigbee2mqtt/front_door', payload '{"battery":91,"voltage":2985,"contact":true,"linkquality":0}'
```

However, device in HomeAssistant is now unavailable:

[image]

I can force the device to re-appear in HomeAssistant by restarting Z2M.

peterforeman, 2019-12-11:
Same here - I first tried the mqtt discovery feature with HASS and this problem occurs. Exactly the same as mentioned above.

It seems zigbee2mqtt does not respond to the birth message from HASS. I guess HASS needs this to re-enable the devices after restart? (using 1.7.1+dev)

peterforeman, 2019-12-11:
Well, I took a look at the source and can't really find anything wrong with that. The theory is that when HASS restarts it will get the discovery data via retained mqtt topics/messages. In my case it seems like the retained messages are not being retained. Hence, when restarting HASS, it will not receive any data about the devices.

After z2m detected the HASS birth message via mqtt it will start a 20-second timer whereafter it will send the current state of all devices to mqtt. This seems to work fine.

@carpenike: which MQTT broker are you using? I'm using RabbitMQ with mqtt plugin. RabbitMQ in high-availability setup does not support propagating the retained flag throughout the cluster. (in my case I'm not running a cluster so retained messages should work)

peterforeman, 2019-12-11:
And I replaced RabbitMQ with mosquitto and the retained messages stuff is working find -and- above mentioned problem disappeared.

So I guess you should take a look at your MQTT broker configuration.

carpenike, 2019-12-11:
Ahh, very interesting. I'm using RabbitMQ and it's configured in a HA fashion.

Configuration here: https://github.com/billimek/k8s-gitops/blob/master/default/rabbitmq/rabbitmq.yaml

So the thought is that if I move to one of the Hass.IO MQTT brokers the issue will go away as the MQTT messages can be retained?

Have you configured your devices with the retain flag?

peterforeman, 2019-12-12:
> Ahh, very interesting. I'm using RabbitMQ and it's configured in a HA fashion.

Then this is the problem. In the docs (https://www.rabbitmq.com/mqtt.html) you can read the following:

> Both implementations have limitations and trade-offs. With the first one, maximum number of messages that can be retained is limited by RAM. With the second one, there is a limit of 2 GB per vhost. Both are node-local (messages retained on one broker node are not replicated to other nodes in the cluster).

> So the thought is that if I move to one of the Hass.IO MQTT brokers the issue will go away as the MQTT messages can be retained?

Yes.

> Have you configured your devices with the retain flag?

No. This is not needed as z2m will send the discovery-data as retained by default. The 'retain' flag of the  devices is meant to retain the values.

My problem is solved by switching to mosquitto. It seems that for one reason or another my RabbitMQ does not want to retain MQTT messages at all. I'm not going to investigate further as mosquitto is fine for me too.

carpenike, 2020-01-07:
@peterforeman thank you very much for the help! I moved MQTT out of the cluster and onto hass directly and Z2M works properly.

closed 2020-01-07 by carpenike