Koenkk/zigbee2mqtt #2472: ZNCZ02LM not working as router

sander816, 2019-12-05:
# Bug Report
## What happened
ZNCZ02LM is not recognized/working as router
Its not a battery powered device, so it should work as router

## Debug Info
Version of Zigbee2Mqtt: 1.7.1+dev
Coordinator version: 20181119

This device seems to work correctly. Last seen 7 minutes, 55 seconds ago.
Zigbee Id is 0x00158d000268d9e6.
Address on Zigbee Network is 0xEF64.
Link quality is 112/255.
Model is lumi.plug by LUMI.
Hardware version is unknown.
This device type is EndDevice

Clement87, 2020-01-04:
Hi, same "issue" on my side.
I have two ZNCZ02LM, one that act as a router and the second act as an end device.
(the ZNLDP12LM is acting as an end device two...)

zigbee2mqtt version:1.8.0 (commit da4d26a)
coordinator type: zStack12, revision: 20190608

[image]

Clement87, 2020-01-05:
I repaired the devices near the coordinator and now they are seen as routers. For me the issue is solved.

closed 2020-03-12 by stale[bot]

Alxspb, 2022-02-10:
For me it's still appearing as `battery` & `end-device`
Strange, because it has no battery and directly connected to 220V
Also it has `router` capabilities in description from official vendor