Koenkk/zigbee2mqtt #2475: Issue Starting zigbee2mqtt with CC2531

Matze2310, 2019-12-05:
Hello,
i have an issue starting the zigbee2mqtt Server.
i'm using a rasbperry pi with raspbian jessie and an CC2531 Stick.
I tried to install and use it under user pi and root but both did not help.

i also looked at your FAQ's but these did not help.

```
root@raspberrypi:/opt/zigbee2mqtt# ls -l /dev/serial/by-id/
total 0
lrwxrwxrwx 1 root root 13 Dec  5 21:01 usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B000BE89048-if00 -> ../../ttyACM0
```
```
root@raspberrypi:/opt/zigbee2mqtt# cat /root/.npm/_logs/2019-12-05T20_37_47_578Z-debug.log
0 info it worked if it ends with ok
1 verbose cli [ '/root/.nvm/versions/node/v10.17.0/bin/node',
1 verbose cli   '/root/.nvm/versions/node/v10.17.0/bin/npm',
1 verbose cli   'start' ]
2 info using npm@6.11.3
3 info using node@v10.17.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.7.1~prestart: zigbee2mqtt@1.7.1
6 info lifecycle zigbee2mqtt@1.7.1~start: zigbee2mqtt@1.7.1
7 verbose lifecycle zigbee2mqtt@1.7.1~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.7.1~start: PATH: /root/.nvm/versions/node/v10.17.0/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/root/.nvm/versions/node/v10.17.0/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
9 verbose lifecycle zigbee2mqtt@1.7.1~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.7.1~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.7.1~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.7.1~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.7.1 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/root/.nvm/versions/node/v10.17.0/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:198:13)
13 verbose stack     at ChildProcess.<anonymous> (/root/.nvm/versions/node/v10.17.0/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:198:13)
13 verbose stack     at maybeClose (internal/child_process.js:982:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid zigbee2mqtt@1.7.1
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.9.35-v7+
17 verbose argv "/root/.nvm/versions/node/v10.17.0/bin/node" "/root/.nvm/versions/node/v10.17.0/bin/npm" "start"
18 verbose node v10.17.0
19 verbose npm  v6.11.3
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.7.1 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.7.1 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
root@raspberrypi:/opt/zigbee2mqtt# nvm use system
Now using system version of node: v11.12.0 (npm v6.7.0)
root@raspberrypi:/opt/zigbee2mqtt# nvm run js10
Running node v10.17.0 (npm v6.11.3)
>
(To exit, press ^C again or type .exit)
>
[1]+  Stopped                 NODE_VERSION="${VERSION}" "${NVM_DIR}/nvm-exec" "$@"
root@raspberrypi:/opt/zigbee2mqtt# nvm run js10 index.js
Running node v10.17.0 (npm v6.11.3)
zigbee2mqtt:info  2019-12-05T20:39:13: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-12-05.21-39-12'
zigbee2mqtt:info  2019-12-05T20:39:13: Starting zigbee2mqtt version 1.7.1 (commit #b459c35)
zigbee2mqtt:info  2019-12-05T20:39:13: Starting zigbee-herdsman...
zigbee2mqtt:error 2019-12-05T20:39:21: Error while starting zigbee-herdsman
zigbee2mqtt:error 2019-12-05T20:39:21: Failed to start zigbee
zigbee2mqtt:error 2019-12-05T20:39:21: Exiting...
zigbee2mqtt:error 2019-12-05T20:39:21: Error: SRSP - SYS - version after 6000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
root@raspberrypi:/opt/zigbee2mqtt# cat /opt/zigbee2mqtt/data/log/2019-12-05.21-39-12
cat: /opt/zigbee2mqtt/data/log/2019-12-05.21-39-12: Is a directory
root@raspberrypi:/opt/zigbee2mqtt# ls /opt/zigbee2mqtt/data/log/2019-12-05.21-39-12
log.txt
root@raspberrypi:/opt/zigbee2mqtt# cat /opt/zigbee2mqtt/data/log/2019-12-05.21-39-12/log.txt
info  2019-12-05T20:39:13: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-12-05.21-39-12'
info  2019-12-05T20:39:13: Starting zigbee2mqtt version 1.7.1 (commit #b459c35)
info  2019-12-05T20:39:13: Starting zigbee-herdsman...
error 2019-12-05T20:39:21: Error while starting zigbee-herdsman
root@raspberrypi:/opt/zigbee2mqtt# cat /opt/zigbee2mqtt/data/log/2019-12-05.21-39-12/log.txt
info  2019-12-05T20:39:13: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-12-05.21-39-12'
info  2019-12-05T20:39:13: Starting zigbee2mqtt version 1.7.1 (commit #b459c35)
info  2019-12-05T20:39:13: Starting zigbee-herdsman...
error 2019-12-05T20:39:21: Error while starting zigbee-herdsman
root@raspberrypi:/opt/zigbee2mqtt#
```

i hope you can help.
thanks.

[zigbee2mqtt.txt](https://github.com/Koenkk/zigbee2mqtt/files/3929096/zigbee2mqtt.txt)

Koenkk, 2019-12-07:
Are you sure you don't have a CC2540? https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start

Matze2310, 2019-12-08:
i bought it as CC2531, so i think it should be this one.

```
pi@raspberrypi ~ $ test -w /dev/ttyACM0 && echo success || echo failure
success
pi@raspberrypi ~ $ sudo test -w /dev/ttyACM0 && echo success || echo failure
success
pi@raspberrypi ~ $ ls -l /dev/ttyACM0
crw-rw---- 1 root dialout 166, 0 Dec  7 13:49 /dev/ttyACM0
pi@raspberrypi ~ $
```

Matze2310, 2019-12-08:
some aditional logs:
```
pi@raspberrypi /opt/zigbee2mqtt $ sudo su
root@raspberrypi:/opt/zigbee2mqtt# ls
azure-pipelines.yaml  data  docker  images  index.js  lib  LICENSE  node_modules  npm-shrinkwrap.json  package.json  README.md  scripts  test  update.sh
root@raspberrypi:/opt/zigbee2mqtt# nvm use js10
Now using node v10.17.0 (npm v6.11.3)
root@raspberrypi:/opt/zigbee2mqtt# npm start

> zigbee2mqtt@1.7.1 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2019-12-08T15:07:17: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-12-08.16-07-17'
zigbee2mqtt:info  2019-12-08T15:07:18: Starting zigbee2mqtt version 1.7.1 (commit #b459c35)
zigbee2mqtt:info  2019-12-08T15:07:18: Starting zigbee-herdsman...
zigbee2mqtt:error 2019-12-08T15:07:25: Error while starting zigbee-herdsman
zigbee2mqtt:error 2019-12-08T15:07:25: Failed to start zigbee
zigbee2mqtt:error 2019-12-08T15:07:25: Exiting...
zigbee2mqtt:error 2019-12-08T15:07:25: Error: SRSP - SYS - version after 6000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.7.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.7.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-12-08T15_07_25_603Z-debug.log
root@raspberrypi:/opt/zigbee2mqtt# node index.js
zigbee2mqtt:info  2019-12-08T15:07:37: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-12-08.16-07-36'
zigbee2mqtt:info  2019-12-08T15:07:37: Starting zigbee2mqtt version 1.7.1 (commit #b459c35)
zigbee2mqtt:info  2019-12-08T15:07:37: Starting zigbee-herdsman...
zigbee2mqtt:error 2019-12-08T15:07:45: Error while starting zigbee-herdsman
zigbee2mqtt:error 2019-12-08T15:07:45: Failed to start zigbee
zigbee2mqtt:error 2019-12-08T15:07:45: Exiting...
zigbee2mqtt:error 2019-12-08T15:07:45: Error: SRSP - SYS - version after 6000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
```

Matze2310, 2019-12-08:
also tried
```
git checkout npm-shrinkwrap.json
git checkout package.json
rm -rf node_modules && npm install
npm start
```
but did not worked

Koenkk, 2019-12-09:
What firmware did you flash on your cc2531?

Matze2310, 2019-12-12:
i have to ask this. I bought it with flashed Firmware

Snufffel, 2019-12-13:
I am getting the exact same error after (I think) an update.
Did not change anything in the config, did not add anything, just all of the sudden my zigbee2mqtt is not working anymore.

I changed PanID from 6754 to 6755, rebooted the thing a couple of time, now I get this message:

```
at Timeout.object.timer.setTimeout [as _onTimeout] (/zigbee2mqtt-1.7.1/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
(node:57) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 27)
(node:57) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
```

solder-soldier, 2019-12-13:
I had the same "at Timeout.object.timer.setTimeout" error. my set up was:
- z2m 1.7.1
- adapter: cc2531
- firmware 20190608

I tried recloning, reinstalling nodejs, etc - nothing worked. then I verifed the firmeware image and it failed. Flash&Verify to CC2531_SOURCE_ROUTING_20190619 solved it finally.

Matze2310, 2019-12-16:
now i tried your Docker Image but with the same issue.

Matze2310, 2019-12-21:
maybe somebody has an idea why it is not working:
I tried the stick under Windows with iobroker and the Stick is working.
used default zigbee sheperd and also upgraded to Zigbee herdsman, both running.

When using the same stick under raspberry with Linux, Zigbee2mqtt Herdsman does not recognize the Stick.

rnijenhu, 2019-12-21:
run into the same error

Must say it's my first attempt to run zigbee2mqtt. system is ubuntu buster and I just followed the manual on: 
https://www.zigbee2mqtt.io/getting_started/flashing_the_cc2531.html 
https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html

Matze2310, 2019-12-22:
what the hell is this...
yersterday i flashed the Stick and it was working.
now i again tried to run Zigbee2mqtt and again it is not working....
it, seems that i need to flash the stick again to make it running.

is this normal?
Do i have to flash the stick every time i reboot or stop the Zigbee process?

cptCloud, 2019-12-28:
I have the same issue, but I noticed the issue is popping as soon as I stop the container running. Afterwards, I cannot start again zigbee2mqtt and get this error each time.

What's the proper way to stop zigbee2mqtt container?
I tried crtl+c, docker kill. Each of those are generating this issue at reboot...

Matze2310, 2019-12-30:
i also used and tried CC2531_SOURCE_ROUTING_20190619 but with the same issue...
Stick is recognized by OS but not by Zigbee2MQTT...

```
pi@raspberrypi ~/flash_cc2531 $ ./cc_chipid
  ID = b524.
```
Does the Problem has to do someting with these Logs??
```
 raspberrypi kernel: [515774.765631] usb 1-1.5.1: USB disconnect, device number 13
Dec 30 21:32:53 raspberrypi kernel: [515774.765953] cdc_acm 1-1.5.1:1.0: failed to set dtr/rts
Dec 30 21:32:53 raspberrypi kernel: [515775.062601] usb 1-1.5.1: new full-speed USB device number 14 using dwc_otg
Dec 30 21:32:53 raspberrypi kernel: [515775.210406] usb 1-1.5.1: New USB device found, idVendor=0451, idProduct=16a8, bcdDevice= 0.09
Dec 30 21:32:53 raspberrypi kernel: [515775.210421] usb 1-1.5.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Dec 30 21:32:53 raspberrypi kernel: [515775.210431] usb 1-1.5.1: Product: TI CC2531 USB CDC
Dec 30 21:32:53 raspberrypi kernel: [515775.210440] usb 1-1.5.1: Manufacturer: Texas Instruments
Dec 30 21:32:53 raspberrypi kernel: [515775.210450] usb 1-1.5.1: SerialNumber: __0X00124B000BE89048
Dec 30 21:32:53 raspberrypi kernel: [515775.213145] cdc_acm 1-1.5.1:1.0: ttyACM0: USB ACM device
Dec 30 21:32:54 raspberrypi kernel: [515776.301552] usb 1-1.5.1: USB disconnect, device number 14
```

rnijenhu, 2019-12-31:
> 
> 
> run into the same error
> 
> Must say it's my first attempt to run zigbee2mqtt. system is ubuntu buster and I just followed the manual on:
> https://www.zigbee2mqtt.io/getting_started/flashing_the_cc2531.html
> https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html

they send me an cc2540 instead of the cc2531...waiting for a new delivery

Matze2310, 2020-01-01:
> they send me an cc2540 instead of the cc2531...waiting for a new delivery

can you send me the logs from your Stick with idVendor, idProduct 
and maybe your ChipID?

Matze2310, 2020-01-03:
> 
> 
> I have the same issue, but I noticed 2 things:
> 
>     * the issue is popping as soon as I stop the container running. Afterwards, I cannot start again zigbee2mqtt and get this error each time.
> 
>     * if I change the data folder with a brand new location, it runs smoothly
> 
> 
> What's the proper way to stop zigbee2mqtt container?
> I tried crtl+c, docker kill. Each of those are generating this issue at reboot...
> 
> How to restart using the same data folder sourced as container volume?

@cptCloud 
use "docker run" command for the first time and "docker start" after every new start of this container
You can Stop the container with "docker stop".
I hope this helps you.

Coder2012, 2020-01-19:
This has been running for a month now, rebooted my pi without changing anything now I get the following error.

`pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.7.1 start /opt/zigbee2mqtt
> node index.js

		Zigbee2mqtt requires node version 10, you are running v13.3.0!

zigbee2mqtt:info  2020-01-19T19:00:32: Logging to directory: '/opt/zigbee2mqtt/data/log/2020-01-19.19-00-32'
zigbee2mqtt:info  2020-01-19T19:00:32: Starting zigbee2mqtt version 1.7.1 (commit #b459c35)
zigbee2mqtt:info  2020-01-19T19:00:32: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-01-19T19:00:40: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-01-19T19:00:40: Failed to start zigbee
zigbee2mqtt:error 2020-01-19T19:00:40: Exiting...
zigbee2mqtt:error 2020-01-19T19:00:40: Error: SRSP - SYS - version after 6000ms
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:536:17)
    at processTimers (internal/timers.js:480:7)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.7.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.7.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-01-19T19_00_40_044Z-debug.log
pi@raspberrypi:/opt/zigbee2mqtt $ 
`

Koenkk, 2020-01-20:
@Coder2012 probably your cc2531 crashed, try reflashing with the source routing firmware (improves stability) https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

closed 2020-03-20 by Koenkk