Koenkk/zigbee2mqtt #2476: reached 55 devices

yaniv1111, 2019-12-06:
Hi,

I've reached 55 devices ( lights and sensors) .
how is it possible to extend the network cluster? 
Should i build a second ZB network and combine them on the same MQTT?
what about ZB 3.0 with CC26X2R1 \  CC1352P-2?

Thanks 
Yaniv

Koenkk, 2019-12-07:
55 devices is a lot for the CC2531/CC2530, if you are experiencing instability issues consider switching to the CC26X2R1 or CC1352-P2 (and if not, don't). https://www.zigbee2mqtt.io/information/supported_adapters.html

yaniv1111, 2019-12-08:
any other supplier rather than TI?

zhapwey, 2019-12-18:
hello

One question, for the issue of zigbee network stability all devices based on cc2530 and cc2531 have the same capacity?
In other words, to make the network more stable to many devices, there are only two options or cc253X or the new CC26X2R1 or CC1352-P2 equipment?

Do we know if there is any device based on CC26X2R1 or CC1352-P2 that is not a development board in progress or in project?****

Koenkk, 2019-12-19:
CC26X2R1 and CC1352-P2 will greatly improve stability, they are only available as a development board for now and can only be bought at TI

elsbrock, 2019-12-27:
Is the switch from CC253X to CC26X2R1 / CC1352 seamless? Docs are not clear. Can I just exchange it?

Koenkk, 2019-12-27:
You need to repair (https://www.zigbee2mqtt.io/information/FAQ.html#requires-repairing)

pcb1962, 2020-01-01:
"CC26X2R1 and CC1352-P2 will greatly improve stability, they are only available as a development board for now and can only be bought at TI"
I'm in the UK, I have just ordered a [CC26X2R1 from Digi-Key](https://www.digikey.co.uk/products/en?keywords=CC26X2R1), total cost was £52 including shipping and customs charges. Looking forward to the improvement, currently my 51 device network is refusing to add any further devices.

closed 2020-03-08 by stale[bot]