Koenkk/zigbee2mqtt #2477: removing device with error

yaniv1111, 2019-12-06:
Dec 06 09:26:07 pi3b2 npm[29196]: zigbee2mqtt:info  2019-12-06T07:26:07: Removing '0x00124b001936acfe'
Dec 06 09:26:17 pi3b2 npm[29196]: zigbee2mqtt:error 2019-12-06T07:26:17: Failed to remove 0x00124b001936acfe (Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms)

sjorge, 2019-12-06:
Not all devices will cleanly leave the network, you can try with /force_remove instead of /remove.

Koenkk, 2019-12-07:
This is expected, see https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove Added this to the logging.

closed 2019-12-07 by Koenkk

SPEC1AL1ST, 2020-09-25:
Zigbee2MQTT:info  2020-09-25 09:36:10: Removing 'Температура Зал'
Zigbee2MQTT:error 2020-09-25 09:36:20: Failed to remove Температура Зал (Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms)
Zigbee2MQTT:error 2020-09-25 09:36:20: See https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove for more info
Zigbee2MQTT:info  2020-09-25 09:36:20: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Температура Зал","type":"device_removed_failed"}'