Koenkk/zigbee2mqtt #2482: Bitron Smart Plug AV2010/25 not pairable

CodeFinder2, 2019-12-07:
Hi all,

first of all thanks for this great piece of software! I am using zigbee2mqtt on HassIo/Home Assistant and trying to pair a [Bitron Smart Plug (AV2010/25)](http://bvsite.smabit.eu/index.php/produkte/smart-home-produkte/smart-plug-mit-verbrauchsdatenerfassung/) which always fails.

Here's the log: https://pastebin.com/NS55HzHk. ("0x00124b0006227235" is the device of interest.)

Please note that the device is listed as being [supported](https://www.zigbee2mqtt.io/devices/AV2010_25.html). I entered pairing mode by plugging it into a wall socket and pressed the button on the device for about ~10s (according to [this German guide](https://manuals.smabit.eu/lde/av2010_25.html#c)). (Can I add this information to the device info page somehow? PR?)

I've also rebooted Hassio 2x, restarted the HassIO addon, entered/left pairing mode a few times, unplugged/re-plugged the device, entered pairing mode again and again but all without success. I also `/remove`d and `/force_remove`d the device from zigbee2mqtt (since my attemps seem to have worked once / a bit ...? However, I was not able to talk to the device and also did not get any response) but that did not help, too. I've also reset the device (keep button pressed while plugging it in, then keep it pressed for 5-6s, according to [this german post](https://telekomhilft.telekom.de/t5/Smart-Home/Bitron-Zwischenstecker-quot-smart-plug-quot-anlernen-klappt/td-p/4042689/page/2)) but it did not help as well.

zigbee2mqtt version: 1.7.1 (not dev), log_level=debug
CC2531 firmware version: 20190619 / zStack12

Side note: after upgrading to zigbee-herdsman, pairing of Xiaomi devices is a breeze. So thanks for that great huge update!

Thank you very much! I you need any more information, just let me know.

Maybe related (but also probably old / outdated): #770, #348

Koenkk, 2019-12-07:
Please provide the herdsman debug logging when pairing

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

CodeFinder2, 2019-12-07:
Here's the log: https://pastebin.com/QHTY6FJf

(I think it contains the same information as the log I provided in my first post or do I miss something?)

Thanks!

Koenkk, 2019-12-08:
The problem is that the plug does not advertise it's modelID. I don't know why but you can workaround it by opening `data/database.db`, look for the line containing `0x00124b0006227235 `, find the `modelID` in that line and change `undefined` to `902010/25`. After that restart zigbee2mqtt and it should work.

CodeFinder2, 2019-12-09:
Thanks for your reply.

Are you sure that this is sufficient? My `database.db` entry for that device currently looks like this:
```json
{"id":27,"type":"Router","ieeeAddr":"0x00124b0006227235","nwkAddr":23246,"manufId":0,"epList":[4,1],"endpoints":{"1":{"profId":260,"epId":1,"devId":9,"inClusterList":[0,3,4,6],"outClusterList":[0],"clusters":{},"binds":[]},"4":{"profId":260,"epId":4,"devId":9,"inClusterList":[],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{}}},"binds":[]}},"interviewCompleted":true,"meta":{}}
````

There is no "modelID". I also created a "modelID": "902010/25" but it still doesn't work. Is there anything I can do about it? Is there specific information you need to debug this?

Koenkk, 2019-12-10:
Sorry, it should be `modelId` with non capital d at the end.

CodeFinder2, 2019-12-10:
Okay fixed the typo, deleted ALL entries (in database.db, devices.yaml), `/remove`'d and `/force_remove`'d (via MQTT) the device (just to be sure), rebooted HassIO host (!) completely, enabled joining and the device joined. I then added the modelId stuff and I **can now talk to the device and receive updates** when pushing the button.

However, there seems to be **no data at 0x00124b0006227235_power** (current power consumption). Do you have an idea, why? Is another patch of the database.db-entry necessary? May there are some attributes missing or so? 🙈

Koenkk, 2019-12-12:
There is a bug in zigbee2mqtt 1.7.1 which can affect this. Try switching to the latest dev branch and try to configure it manually: https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/information/mqtt_topics_and_message_structure.md#zigbee2mqttbridgeconfigure

CodeFinder2, 2019-12-13:
Can I use the stable and dev version as Hassio addons side-by-side?

More specifically, I mean: right now, I have installed the stable version of the addon. Can I stop that addon, install the `-edge` version of it additionally, copy the entire config of the stable addon to the `-edge` version and simply start the edge version? Is there a potential risk of destroying anything (e.g., having to repair) or overwritten any configurations?

If no, I would try what you have suggested. 😊

Koenkk, 2019-12-14:
You can just remove the stable one and install the edge one, data should automatically be migrated.

CodeFinder2, 2019-12-14:
Did that but it's not working. Here's the log: https://pastebin.com/nrimgPMf (with the -edge version).

Anything more I can do about it to debug the issue?

Thanks again for your super support!!! :-)

Koenkk, 2019-12-15:
Looks good, does it provide the power values now when attaching a load?

CodeFinder2, 2019-12-15:
Unfortunately, no. No power values 🤔

I've plugged in a hair dryer (2k watts) and waited a minute or so 🙄

Koenkk, 2019-12-15:
Maybe you need to factory reset it again, can you try this?

CodeFinder2, 2019-12-15:
Will test right now and report back ASAP.

Do you think we can get this working today or at least "at some time"? :-D The problem is: my deadline for sending the device back to the seller ends tomorrow. I bought it for the power consumption measurement capability (to send a notification when my washing machine finishes). I would keep it so we can further investigate (taking the risk that it finally won't work) if you have more ideas to try and assist me in debugging this. (I try to provide whatever you need!)

But if you do not have more ideas to try as well (apart from the factory reset which I'll try in a second), it may still be a device issue. Anyway, the device would just be an "expensive smart plug" for me and I have plenty of them around here (the cheaper but perfectly working OSRAMs). So acutally no need for it w/o the power consumption thing...

CodeFinder2, 2019-12-15:
Still no success. I used your latest dev version (without Hassio /addon) with a CC2530 + CC2591 at /dev/ttyUSB0 and also reconfigured via MQTT. I (hopefully...) factory-reset the device and also restarted zigbee2mqtt (see logs).

Version:
```
$ git lg
0b9997b - (HEAD, origin/dev) Update converters. (vor 58 Minuten) <Koen Kanters>
```

Here are the logs:
- [FIRST-2019-12-15.16-34-36](https://pastebin.com/Xiz3aSgZ)
- [SECOND-2019-12-15.16-37-32](https://pastebin.com/ERA5MMZB)
- [THIRD-2019-12-15.16-38-38](https://pastebin.com/hs4nWuuR)

What's important:
- between 16:35:38 to 16:36:40 I started my hair dryer for the first time (note that ON -> OFF transition)
- at 16:39:07, I issued a reconfiguration using: `$ mosquitto_pub -h localhost -p 1883 -u XXX -P XXX -m "0x00124b0006227235" -t "zigbee2mqtt/bridge/configure"
`
- between 16:39:17 to 16:39:50 I started my hair dryer again

I would have expected that there are log entries reporting some sort of a "power consumption" (or a similar keyword) but I could not find ANYTHING that looks like that... would love to help reading the logs but I cannot find anything special apart from those missing lines, reporting power consumption (and I don't know why they are not there...)

More I can do?

Thanks again! :-)

Koenkk, 2019-12-15:
Can you try modifying the entry in `devices.js` to

```js
{
    zigbeeModel: ['902010/25'],
    model: 'AV2010/25',
    vendor: 'Bitron',
    description: 'Video wireless socket',
    supports: 'on/off, power measurement',
    fromZigbee: [fz.on_off, fz.bitron_power],
    toZigbee: [tz.on_off],
    meta: {configureKey: 2},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering']);
        await configureReporting.instantaneousDemand(endpoint);
    },
},
```

After this make sure that the device configured succesfully.

CodeFinder2, 2019-12-15:
Sure! Given the above commit as a base, I modified `/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices.js` accordingly.

Then did a `npm install`, a `npm start`, and issued a reconfigure via MQTT.

Still no power consumption. Log: https://pastebin.com/4EmzgEEc  :-(

Koenkk, 2019-12-16:
I doubt if your device really supports the power metering, according to the clusters announced by the device itself, it does not support the power metering cluster. I would advise you return this device as I don't know what to do next.

CodeFinder2, 2019-12-17:
Okay, will do so. Nonetheless, thank you so much for your great assistance!

closed 2019-12-17 by Koenkk

CodeFinder2, 2019-12-22:
You are a genius! **TL;DR: its working now with a new plug and including your mods above!**

Sorry to reply to this once more but I've bought a new plug which is explicitly labeled as "902010/25" and I was able to pair it on *first* try using the latest dev branch (yeah :-) ).

However, then there were still no power consumption values being reported. Here's the log (but not actually required to take a look, just for completeness, see below): https://pastebin.com/BNV8GCgr

And this is how my database.db entry looked like:
```
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b001b7d9065","nwkAddr":0,"manufId":0,"epList":[12,110,11,8,6,5,4,3,2,1],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{},"binds":[]},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]}},"interviewCompleted":false,"meta":{},"lastSeen":null}

{"id":2,"type":"Router","ieeeAddr":"0x00124b00088b7b99","nwkAddr":37630,"manufId":0,"manufName":"Bitron Home","powerSource":"Mains (single phase)","modelId":"902010/25","epList":[1,4],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,4,5,6,1794],"outClusterList":[0],"clusters":{"genBasic":{"attributes":{"modelId":"902010/25","manufacturerName":"Bitron Home","powerSource":1,"zclVersion":1,"appVersion":15,"hwVersion":0,"dateCode":"20150629"}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001b7d9065","endpointID":1}]},"4":{"profId":260,"epId":4,"devId":9,"inClusterList":[],"outClusterList":[25],"clusters":{},"binds":[]}},"appVersion":15,"hwVersion":0,"dateCode":"20150629","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1577014386135}
```

I've then tried [your modification](https://github.com/Koenkk/zigbee2mqtt/issues/2482#issuecomment-565831884) a few comments above and voila: now it's working! :-)

Here's the log (note that I didn't pair the device again after applying the modifications to `devices.js`, just a `npm install` and `npm start`): https://pastebin.com/qwRqtNYS (sufficient to take a look at the end -> `power`)

**Shall I create a PR for these changes or will you do so? ;-)**

One last question (hopefully :-D ): I also tried pairing with my v1.8.0 stable via the HassIO plugin (also w nearly 40 devices) and that did not work. I got it working by manually editing the database.db file (although still w/o power metering). Is there still a difference between dev and stable (1.8) that explains this problem? If so, I would really appreciate a minor release to also have these fixes in the stable branch. (For the,  above tests, I used the latest dev branch.)

Once again, THANK YOU very much and Merry Christmas! :-)