Koenkk/zigbee2mqtt #2484: 1.7.0 upgrade docker startup issues

thenakuls, 2019-12-08:
# Bug Report
## Upgrade after 1.6.0 will not start with docker. The log present at: https://pastebin.com/Vf75qyeY

 #719 is probably related however, I cannot run this command:
`rm -rf node_modules && npm install`
because the image fails to start because of the error

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

thenakuls, 2019-12-08:
I tried adding this to my docker-compose, but didn't help. 
- ./run.sh:/bin/sh -c "rm -rf node_modules && npm install && /run.sh"

Koenkk, 2019-12-08:
Do you have a docker volume mapping `homeassistant.js` or `zigbee-shepherd-converters`? If so, remove this.

thenakuls, 2019-12-10:
I did have a volume mapping to homeassistant.js
Thank you, removing it solved the issue

closed 2019-12-10 by thenakuls