Koenkk/zigbee2mqtt #2485: Devices unavailable, were discovered before.

Kjronning, 2019-12-08:
I get this when trying to do npm start. It worked well a few hours ago. 

```
zigbee2mqtt:info  2019-12-08T09:05:14: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-12-08.09-05-13'
(node:3649) UnhandledPromiseRejectionWarning: Error: EACCES: permission denied, unlink '/opt/zigbee2mqtt/data/log/2019-12-08.06-02-09/log.txt'
    at Object.unlinkSync (fs.js:956:3)
    at rimrafSync (/opt/zigbee2mqtt/node_modules/rimraf/rimraf.js:309:17)
    at options.readdirSync.forEach.f (/opt/zigbee2mqtt/node_modules/rimraf/rimraf.js:344:39)
    at Array.forEach (<anonymous>)
    at rmkidsSync (/opt/zigbee2mqtt/node_modules/rimraf/rimraf.js:344:26)
    at rmdirSync (/opt/zigbee2mqtt/node_modules/rimraf/rimraf.js:337:7)
    at Function.rimrafSync [as sync] (/opt/zigbee2mqtt/node_modules/rimraf/rimraf.js:307:9)
    at directories.forEach (/opt/zigbee2mqtt/lib/util/logger.js:77:20)
    at Array.forEach (<anonymous>)
    at DerivedLogger.cleanup (/opt/zigbee2mqtt/lib/util/logger.js:75:21)
(node:3649) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:3649) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

```

Kjronning, 2019-12-08:
Deleted the logs and it seems to be fixed. will probably break again due to logs interaction.

Koenkk, 2019-12-08:
You need to fix the permission on your file system, (e.g. `sudo chown -R pi:pi /opt/zigbee2mqtt`). Closing this as it is not a zigbee2mqtt issue.

closed 2019-12-08 by Koenkk

TheOlfer, 2021-01-07:
Hi,

Had a similar issue. npm start works without errors, but starting through service threw exactly this rimraf error. 

In the logs one folder was owned by root all others by pi. 
I could fix it with chown.

My question, as I also experienced other errors caused by  root beeing the owner instead of pi , how it comes that sometimes root creates the logs and sometimes pi?

Would it be possible to change that behaviour as it is creating some issues?

Cheers!

michaelkrog, 2021-08-18:
I also ran into this today after power outage. When rebooting my raspberry, the log file was locked somehow and hence zigbee2mqtt could not start.

Maybe if it could try to roll to a new file to see if it at least has access to creaete a new log file. 

```
Aug 18 03:17:36 raspberrypi systemd[1]: Started zigbee2mqtt.
Aug 18 03:17:39 raspberrypi npm[447]: > zigbee2mqtt@1.21.0 start /opt/zigbee2mqtt
Aug 18 03:17:39 raspberrypi npm[447]: > node index.js
Aug 18 03:17:46 raspberrypi npm[447]: Zigbee2MQTT:info  2021-08-18 03:17:46: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2021-08-18.03-17-43' filename: log.txt
Aug 18 03:17:46 raspberrypi npm[447]: (node:502) UnhandledPromiseRejectionWarning: Error: EACCES: permission denied, unlink '/opt/zigbee2mqtt/data/log/2021-08-13.10-52-32/log.txt'
Aug 18 03:17:46 raspberrypi npm[447]:     at Object.unlinkSync (fs.js:1251:3)
Aug 18 03:17:46 raspberrypi npm[447]:     at rimrafSync (/opt/zigbee2mqtt/node_modules/rimraf/rimraf.js:303:17)
Aug 18 03:17:46 raspberrypi npm[447]:     at /opt/zigbee2mqtt/node_modules/rimraf/rimraf.js:336:39
Aug 18 03:17:46 raspberrypi npm[447]:     at Array.forEach (<anonymous>)
Aug 18 03:17:46 raspberrypi npm[447]:     at rmkidsSync (/opt/zigbee2mqtt/node_modules/rimraf/rimraf.js:336:26)
Aug 18 03:17:46 raspberrypi npm[447]:     at rmdirSync (/opt/zigbee2mqtt/node_modules/rimraf/rimraf.js:329:7)
Aug 18 03:17:46 raspberrypi npm[447]:     at Function.rimrafSync [as sync] (/opt/zigbee2mqtt/node_modules/rimraf/rimraf.js:301:9)
Aug 18 03:17:46 raspberrypi npm[447]:     at /opt/zigbee2mqtt/lib/util/logger.js:126:20
Aug 18 03:17:46 raspberrypi npm[447]:     at Array.forEach (<anonymous>)
Aug 18 03:17:46 raspberrypi npm[447]:     at DerivedLogger.cleanup (/opt/zigbee2mqtt/lib/util/logger.js:124:21)
Aug 18 03:17:46 raspberrypi npm[447]: (Use `node --trace-warnings ...` to show where the warning was created)
Aug 18 03:17:46 raspberrypi npm[447]: (node:502) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch bloc
Aug 18 03:17:46 raspberrypi npm[447]: (node:502) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Nod
```