Koenkk/zigbee2mqtt #2487: Wireshark - unrecognized libpcap format or not libpcap data

FlyingPersian, 2019-12-08:
# Bug Report
## What happened
I installed the firmware for the CC2531 as described on the website. I then installed Wireshark and whsniff and then ran it as described on the website as well. When Starting to log, I get this error:

Unrecognized libpcap format or not libpcap data. 

I wasn't sure whether the debugger has to be connected or not, so I tried both with connected and disconnected debugger. Same result. The CC2531 USB Stick is hooked up to my Laptop running Ubuntu 18.04. 

## What did you expect to happen
I expected Wireshark to log everything. I'm not sure whether it's capturing ZigBee traffic. Here a screenshot of my Wireshark after I click "okay" on the error: https://imgur.com/a/3eyBxq2

## How to reproduce it (minimal and precise)

Flash the firmware:
#Install tools for cc-tool
sudo apt install dh-autoreconf libusb-1.0 libboost-all-dev

#install cc-tool
git clone https://github.com/dashesy/cc-tool.git
cd cc-tool
./bootstrap
./configure
make

#Flash firmware (tried both the firmware from your github (CC2531_DEFAULT_20190608) and the firmware from the PACKET-SNIFFER (not V2)).
sudo ./cc-tool -e -w CC2531ZNP-Prod.hex

#Install further tools. Using curl didn't work. I downloaded v1.1.tar.gz using wget and extracted it with tar zxf
cd /opt
sudo apt-get install -y libusb-1.0-0-dev wireshark
curl -L https://github.com/homewsn/whsniff/archive/v1.1.tar.gz | tar zx
cd whsniff-1.1
make
sudo make install

#Run Wireshark
sudo whsniff -c ZIGBEE_CHANNEL_NUMBER | wireshark -k -i -

## Debug Info
zigbee2mqtt version: not sure. 
CC253X firmware version: CC2531_DEFAULT_20190608 and the firmware from PACKET-SNIFFER (not V2)

FlyingPersian, 2019-12-08:
Okay so i'm pretty sure that CC2531_DEFAULT_20190608 is not the correct firmware for sniffing. Since I'm running Ubuntu I tried the following firmwares:

sniffer_fw_cc2531.hex from PACKET-SNIFFER (not V2)
zboss_sniffer.hex from ZBOSS (both Windows and Ubuntu version)

I also tried this on Windows since I have a Windows PC. The CC2531 Dongle is recognized, but no traffic appears in Wireshark. Since I want to mainly use this on Ubuntu, I'd like to solve the "unrecognized libpcap format or not libpcap data" error. It refers to the CC2531 interface (since it's not listed in the interface list). 

I had the idea to use ZBOSS for this since it's available on Ubuntu. It fails to connect to ttyACM0 though: https://imgur.com/a/rLWpzKl

wtf-42, 2019-12-26:
Hello Persian, I have the same problem, did you find any solution?

FlyingPersian, 2019-12-26:
> Hello Persian, I have the same problem, did you find any solution?

I think I downloaded zboss and just started that with sudo.

closed 2020-03-03 by stale[bot]